openhab/org.openhab.binding.zigbee #2: IO exception on dongle initialization

kaikreuzer, 2017-05-09:
Trying my TI dongle, I get the following exception on startup (on my Mac):
```
2017-05-09 15:46:15.559 [DEBUG] [ZigBeeCoordinatorCC2531Handler:143  ] - Opening ZigBee CC2531 serial port
2017-05-09 15:46:15.559 [DEBUG] [ZigBeeCoordinatorCC2531Handler:95   ] - Connecting to serial port [/dev/tty.usbmodem1421]
2017-05-09 15:46:15.593 [INFO ] [ZigBeeCoordinatorCC2531Handler:110  ] - Serial port [/dev/tty.usbmodem1421] is initialized.
2017-05-09 15:46:15.605 [DEBUG] [b.z.h.ZigBeeCoordinatorHandler:301  ] - initResponse is JOINED
2017-05-09 15:46:15.605 [DEBUG] [b.z.h.ZigBeeCoordinatorHandler:308  ] - initializeNetwork is true
2017-05-09 15:46:15.889 [ERROR] [z.d.c.n.i.CommandInterfaceImpl:136  ] - IO exception in packet parsing.java.io.IOException: Device not configured in readByte
	at gnu.io.RXTXPort.readByte(Native Method)
	at gnu.io.RXTXPort$SerialInputStream.read(RXTXPort.java:1286)
	at com.zsmartsystems.zigbee.dongle.cc2531.zigbee.util.MarkableInputStream.read(MarkableInputStream.java:100)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:102)
	at java.lang.Thread.run(Thread.java:745)
```

What does that want to tell me?

kaikreuzer, 2017-05-14:
@cdjackson Any hint for me? I would like to join the group of testers and I only have the Pine dongle, which is supposed to be the "standard" TI chip.

cdjackson, 2017-05-14:
Sorry Kai - I missed this.

I’ve also seen this on the Pine dongle - I don’t know why since it seems to be something low level and it doesn’t happen with the other TI dongles I have.  If I run in the debug environment I have which uses jscc instead of nrjavaserial, it works ok :(

I’ll take a look over the next couple of days.

cdjackson, 2017-05-16:
A few links for reference - just trying to get some ideas what's causing this issue...

The following implies a hardware issue (although it's not exactly the same error - it's a write error where here we see a read error, although in my system with the Pin stick I see both read and write errors!).
http://forum.arduino.cc/index.php?topic=65761.0

Another issue in OH regarding this and USB serial ports -:
https://github.com/openhab/openhab1-addons/issues/3899

for completeness - this is what I see -@kaikeeuzer is it the same for you -:

```
2017-05-16 18:03:38.971 [INFO ] [ZigBeeCoordinatorCC2531Handler:111  ] - Serial port [/dev/tty.usbmodem1411] is initialized.
2017-05-16 18:03:38.975 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:177  ] - ->  SYS_RESET (Packet: subsystem=null, length=1, apiId=41 00, data=FE 01 41 00 01 41, checksum=41, error=false) 
2017-05-16 18:03:39.583 [INFO ] [ome.event.ThingStatusInfoEvent:43   ] - 'yahooweather:weather:berlin' updated: ONLINE
2017-05-16 18:03:40.323 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:153  ] - <-- SYS_RESET_RESPONSE (FE 06 41 80 00 02 00 02 06 01 C0)
2017-05-16 18:03:40.327 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:378  ] - Received Async Cmd: Packet: subsystem=null, length=6, apiId=41 80, data=FE 06 41 80 00 02 00 02 06 01 C0, checksum=C0, error=false
2017-05-16 18:03:40.328 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:177  ] - ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false) 
2017-05-16 18:03:40.601 [ERROR] [z.d.c.n.i.CommandInterfaceImpl:136  ] - IO exception in packet parsing: java.io.IOException: Device not configured in readByte
	at gnu.io.RXTXPort.readByte(Native Method)
	at gnu.io.RXTXPort$SerialInputStream.read(RXTXPort.java:1286)
	at com.zsmartsystems.zigbee.dongle.cc2531.zigbee.util.MarkableInputStream.read(MarkableInputStream.java:100)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:102)
	at java.lang.Thread.run(Thread.java:745)

2017-05-16 18:03:40.602 [DEBUG] [d.c.n.packet.ZToolPacketParser:127  ] - ZToolPacketParser parserThread exited.
2017-05-16 18:03:41.581 [DEBUG] [c.n.i.ZigBeeNetworkManagerImpl:1116 ] - SYS_VERSION executed and timed out while waiting for response.
2017-05-16 18:03:41.582 [DEBUG] [c.n.i.ZigBeeNetworkManagerImpl:1122 ] - Failed to send SYS_VERSION [attempt 1]
2017-05-16 18:03:42.586 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:177  ] - ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false) 
Write action failed! Device not configured2017-05-16 18:03:42.588 [ERROR] [c.n.i.ZigBeeNetworkManagerImpl:1090 ] - Synchronous command send failed due to IO exception. java.io.IOException: Device not configured in writeByte
	at gnu.io.RXTXPort.writeByte(Native Method)
	at gnu.io.RXTXPort$SerialOutputStream.write(RXTXPort.java:1145)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.sendRaw(CommandInterfaceImpl.java:298)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.sendPacket(CommandInterfaceImpl.java:179)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.sendSynchronousCommand(CommandInterfaceImpl.java:264)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.ZigBeeNetworkManagerImpl.sendSynchrouns(ZigBeeNetworkManagerImpl.java:1088)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.ZigBeeNetworkManagerImpl.sendSynchrouns(ZigBeeNetworkManagerImpl.java:1062)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.ZigBeeNetworkManagerImpl.getStackVersion(ZigBeeNetworkManagerImpl.java:909)
	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.ZigBeeNetworkManagerImpl.startup(ZigBeeNetworkManagerImpl.java:287)
	at com.zsmartsystems.zigbee.dongle.cc2531.ZigBeeDongleTiCc2531.initialize(ZigBeeDongleTiCc2531.java:88)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:265)

2017-05-16 18:03:42.589 [DEBUG] [c.n.i.ZigBeeNetworkManagerImpl:289  ] - Failed to get CC2531 version
```

This shows the initial communications working for the reset command and response, then it stops. From what I've found on the web (quite minimal) it indicates that this is a hardware issue. 

Sometimes I don't see the exceptions although the result is similar -:
```
2017-05-16 18:11:33.643 [INFO ] [ZigBeeCoordinatorCC2531Handler:111  ] - Serial port [/dev/tty.usbmodem1411] is initialized.
2017-05-16 18:11:33.650 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:177  ] - ->  SYS_RESET (Packet: subsystem=null, length=1, apiId=41 00, data=FE 01 41 00 01 41, checksum=41, error=false) 
2017-05-16 18:11:35.213 [INFO ] [ome.event.ThingStatusInfoEvent:43   ] - 'yahooweather:weather:berlin' updated: OFFLINE (COMMUNICATION_ERROR): The Yahoo Weather API did not return any data.
2017-05-16 18:11:35.213 [INFO ] [nt.ThingStatusInfoChangedEvent:43   ] - 'yahooweather:weather:berlin' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR): The Yahoo Weather API did not return any data.
2017-05-16 18:11:35.226 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:153  ] - <-- SYS_RESET_RESPONSE (FE 06 41 80 00 02 00 02 06 01 C0)
2017-05-16 18:11:35.227 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:378  ] - Received Async Cmd: Packet: subsystem=null, length=6, apiId=41 80, data=FE 06 41 80 00 02 00 02 06 01 C0, checksum=C0, error=false
[... cut ...]
2017-05-16 18:11:37.244 [DEBUG] [z.d.c.n.i.CommandInterfaceImpl:177  ] - ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false) 
```

I also note that once the LED goes out on the Pine stick it won't communicate - this seems to happen quite quickly (maybe after a minute or so).

It does seem to work with jscc library instead of nrvjavaserial though, so I don't know what's up. I don't see this with the standard TI stick and firmware - I might try to burn a new firmware when I get home to see if that solves it.

kaikreuzer, 2017-06-01:
I just retested it with the same result. I have two such sticks, both behave in the same way.

> It does seem to work with jscc library

Do you actively use jSSC? Do you have good experience with that? If so, maybe it is worth to consider this as an alternative to RXTX in general? How much work would it be to switch your implementation over?

kaikreuzer, 2017-06-01:
@maggu2810, I just noticed that you did some PRs 3 years ago on jSSC - do you have any feedback on it? Unfortunately, the project looks equally dead as RXTX by now...

maggu2810, 2017-06-01:
No, I haven't.

cdjackson, 2017-06-01:
I use jSCC on some projects - it seems ok but I don't have strong feelings either way. The conversion between them is pretty simple I think - they use the same interfaces, just some configuration options (ie enums) are different if I remember correctly.

It's unfortunate (and strange even) that there seems to be no good / active serial library for Java.

maggu2810, 2017-06-01:
IIRC I found some memory leaks (or has it been only the one I reported? -- I don't know):
https://github.com/scream3r/java-simple-serial-connector/issues/56

kaikreuzer, 2017-06-01:
> It's unfortunate (and strange even) that there seems to be no good / active serial library for Java.

That's indeed very annoying. Even Oracle's new attempt with Device IO API seems to be inactive...

wsowa, 2017-11-23:
**tl;rd;**
It is a hardware problem with the dongle. In order to fix/workaround it you need to de-solder R91 resistor (0 ohm resistor - a jumper) and solder it in place of R92 which is not mounted normally. See dongle's reference design on TI's site for location of the jumper (schematics and PCB layouts): http://www.ti.com/lit/sw/swrr067/swrr067.zip

See the elements marked on the photo of my dongle 
[image]

These are quite small elements which require a bit of experience with soldering. You may try to remove R91 either with a solder iron or with sharp tweezers or a knife. Just don't damage anything around it. Instead of soldering it in place of R92 you may just short R92 terminals my placing a blob of solder instead of a jumper.

**Full version:**
For a reference this is the stack we're running:
1. TI' CC2531 Chip - a zigbee radio + microcontroller with USB support on a single chip
2. ZNP (Zigbee Network Processor) - a firmware running on the chip that generally expose zigbee over serial port API. Serial port is implemented over USB
3. com.zsmartsystems.zigbee - a java library running on a PC/raspberry that implements ZNP API and talks to ZNP over the serial port
4. zigbee openhab binding
5. openhab.
First three are where the problem is...

The ZNP API allows for requesting chip to reset in two ways. A hard reset basically resets the chip. A soft reset resets the ZNP application while the chip is running. After the hard reset the chip starts in a bootloader. In order to leave the bootloader and start the ZNP we need to send it a "magic byte" 0xEF. After the soft reset the ZNP should run immediately. There's one more difference. The hard reset resets the USB as well while the soft one don't. In practice the difference is that the hard reset causes the serial port to vanish and reappear immediately. Interestingly the com.zsmartsystems.zigbee library requests a soft reset as it should. Yet I could observe it on my Ubuntu dmesg output that the dongle has disappeared from /dev/ttyACM0 port an re-appeared on /dev/ttyACM1 every time I've been issuing the reset command. That made the hardware problem my suspicion. Obviously if the com.zsmartsystems.zigbee is talking to the ZNP over /dev/ttyACM0 and after the reset the device moves to /dev/ttyACM1 then there's nothing that the library can talk to  /dev/ttyACM1 and that's why there are failures in communication just after the reset.

Why does it happen to some dongles?
I don't know. I can suspect that chip clones of the original TI dongle from Aliexpress (like the one I have) have lower quality or underrated capacitors and when the soft reset happens the voltage on the chip drops a bit causing hard reset. Or maybe the design itself is unreliable and the original TI dongles are affected too and it is more a function of the PC usb hub rather than dongle itself... I'm just guessing here yet the fact is that the CC2531 hard resets regardless of what reset type was requested on some dongles.

How does removing R91 and adding R92 jumper helps?
The USB host detects the presence of the device by seeing the D+ data line at other state than D- data line. On the dongle the D+ is pulled up by a resistor and R91 jumper to one of the CC2531's pins. CC2531 then controls the D+ pullup with this pin. I suspect that hard reset causes this pull up to deactivate for brief moment but long enough for USB host to "lost" the device. by removing the R91 and adding R92 we stop controlling the D+ pullup with the CC2531 and connect it to +3.3V which means effectively that the device is always present. The dongle still seems to hard reset regardless of the reset type requested but the change of the jumpers makes this reset invisible to USB host. The hard reset means that the chip starts in a bootloader so after every reset we need to send a "magic byte" which current code does if there's no response to reset within 5 seconds.

You may say, "Wait, wait, wait. In the logs above there is a reset request, a reset response and then the communication breaks at getting the ZNP version. So it can't be the wrong reset problem as the chip confirms it had reset" Well, the logs are misleading here and what you see is a race condition... When I use current master version of com.zsmartsystem.zigbee I can see quite reasonable behaviour that the first reset after plugging in the dongle works fine (but takes 5 sec) and the following reset breaks the communication. This is because initially the dongle is in boodloader. The reset code sends a reset request and if there's no response in 5 seconds it sends the "magic byte" assuming the chip is in bootloader starting ZNP in effect. Second reset does the hard reset which breaks the serial port so no more communication can happen. With the version used by the binding it works a bit differently though... It took me quite a lot to realize that the code that is supposed to just open the port actually opens the port AND sends the magic byte 8-O. Besides the fact that it implies an assumption that the chip is a bootloader when the binding starts up it creates the funny effect that the ZNP starts (and thus sends reset respons) exactly at the same time when we sent the first reset to it. It is still the reset that breaks the port while the reset confirmation you see in the logs is an effect of magic byte sent just before the reset with no mention in the logs...

One more comment for @cdjackson. Please, don't put JARs into binding source code but use maven dependency instead. And if you really have to put them in the lib directory then please, please, please put them in released version rather than in 1.0.0-SNAPSHOT. If I want to debug, contribute, fork, I have no idea what exactly version of the lib you're using as there are tens of commits to 1.0.0-SNAPSHOT. I hope it will get better with the next release.

cdjackson, 2017-11-23:
I will change to using maven for dependencies at some point soon. But there are other limitations that also need to be worked around and while there is so many updates going on it's easier for now.

cdjackson, 2018-01-13:
Closing as there's a good explanation above of the likely hardware issues causing this.

closed 2018-01-13 by cdjackson