openhab/org.openhab.binding.zigbee #7: NumberFormatException when adding ZigBee things

giannello, 2017-05-13:
Openhab2 nightly v2.1.0~20170513100611-1
ZigBee Binding v2.0.0.201705102217

When adding a new thing, my Osram lightbulbs are identified and the pairing process is successful, but the things in openhab stay in `UNINITIALIZED` state because of the following error:

```
2017-05-13 16:42:08.938 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while initializing handler of thing 'zigbee:device:ea6edccc:1234567812ca12e1': java.lang.NumberFormatException: For input string: "1234567812CA12E1"
java.util.concurrent.ExecutionException: java.lang.NumberFormatException: For input string: "1234567812CA12E1"
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_65]
	at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_65]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9.run(ThingManager.java:710)[106:org.eclipse.smarthome.core.thing:0.9.0.201705120951]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_65]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_65]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_65]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_65]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_65]
	at java.lang.Thread.run(Thread.java:745)[:1.8.0_65]
Caused by: java.lang.NumberFormatException: For input string: "1234567812CA12E1"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)[:1.8.0_65]
	at java.lang.Long.parseLong(Long.java:592)[:1.8.0_65]
	at com.zsmartsystems.zigbee.IeeeAddress.<init>(IeeeAddress.java:29)[183:org.openhab.binding.zigbee:2.0.0.201705102217]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.initialize(ZigBeeThingHandler.java:76)[183:org.openhab.binding.zigbee:2.0.0.201705102217]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:713)[106:org.eclipse.smarthome.core.thing:0.9.0.201705120951]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:1)[106:org.eclipse.smarthome.core.thing:0.9.0.201705120951]
	at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:181)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
	... 3 more
```

The problem was verified with multiple zigbee bulbs from the same vendor.

cdjackson, 2017-05-13:
This should already be fixed - I’ll create a new snapshot tonight.

giannello, 2017-05-21:
I finally managed to compile the JAR myself, and the issue is still present on master
```
2017-05-21 20:20:05.673 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while initializing handler of thing 'zigbee:device:d83baf2f:1234567812ca12e1': java.lang.NumberFormatException: For input string: "1234567812ca12e1"
java.util.concurrent.ExecutionException: java.lang.NumberFormatException: For input string: "1234567812ca12e1"
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_65]
	at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_65]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9.run(ThingManager.java:710)[106:org.eclipse.smarthome.core.thing:0.9.0.201705120951]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_65]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_65]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_65]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_65]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_65]
	at java.lang.Thread.run(Thread.java:745)[:1.8.0_65]
Caused by: java.lang.NumberFormatException: For input string: "1234567812ca12e1"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)[:1.8.0_65]
	at java.lang.Long.parseLong(Long.java:592)[:1.8.0_65]
	at com.zsmartsystems.zigbee.IeeeAddress.<init>(IeeeAddress.java:29)[188:org.openhab.binding.zigbee:2.1.0.201705211921]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.initialize(ZigBeeThingHandler.java:77)[188:org.openhab.binding.zigbee:2.1.0.201705211921]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:713)[106:org.eclipse.smarthome.core.thing:0.9.0.201705120951]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:1)[106:org.eclipse.smarthome.core.thing:0.9.0.201705120951]
	at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:181)[99:org.eclipse.smarthome.core:0.9.0.201705120951]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
	... 3 more
```

cdjackson, 2017-05-21:
Please use the version provided on my download site - not here as the library isn’t up to date and this has completely changed.

giannello, 2017-05-22:
Confirmed working with the latest libs from https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/unsigned64bit

closed 2017-05-22 by cdjackson