openhab/org.openhab.binding.zigbee #9: Ieee Address for XXXX returned null

bguidinger, 2017-05-17:
ZigBee Binding v2.1.0.201705152139

When a new device joins the controller, the IEEE Address Request never returns, and the device is not added to OpenHAB.  This is happening with both a GE Link bulk and a Philips Hue White and Color bulb.

You can see that the end device is in the neighbor list of the controller (with the IEEE address), but it still tries to do an IEEE Address Request on it which subsequently fails.  Couldn't it just use the IEEE Address from the neighbor table instead?

```
2017-05-17 07:05:56.571 [DEBUG] [ee.internal.ZigBeeNetworkMeshMonitor] - 0: Starting mesh update
2017-05-17 07:05:56.573 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - Ieee Address for 8866 returned null
2017-05-17 07:05:56.574 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementLqiRequest [0/0 -> 0/0, cluster=0031, TID=06, startIndex=0]
2017-05-17 07:05:56.574 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 8866: Discovery request IEEE_ADDRESS failed. Wait before retry.
2017-05-17 07:05:56.575 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=12, apiId=24 01, data=FE 0C 24 01 00 00 00 00 31 00 06 30 1F 02 00 00 33, checksum=33, error=false)
2017-05-17 07:05:56.593 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
2017-05-17 07:05:56.604 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZToolPacket (FE 23 45 FF 00 00 00 31 80 00 00 00 00 00 01 00 01 00 B0 47 45 50 C0 D9 7E 57 AD 26 10 01 88 17 00 A2 22 15 02 01 AA C3)
2017-05-17 07:05:56.606 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=35, apiId=45 FF, data=FE 23 45 FF 00 00 00 31 80 00 00 00 00 00 01 00 01 00 B0 47 45 50 C0 D9 7E 57 AD 26 10 01 88 17 00 A2 22 15 02 01 AA C3, checksum=C3, error=false
2017-05-17 07:05:56.626 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementLqiResponse [0/0 -> 0/0, cluster=8031, TID=NULL, status=SUCCESS, neighborTableEntries=1, startIndex=0, neighborTableListCount=1, neighborTableList=[NeighborTable [extendedPanId=7ED9C0504547B000, extendedAddress=001788011026AD57, networkAddress=8866, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=CHILD, permitJoining=UNKNOWN, depth=1, lqi=170]]]
2017-05-17 07:05:56.634 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: IeeeAddressRequest [0/0 -> 0/0, cluster=0001, TID=07, nwkAddrOfInterest=0, requestType=1, startIndex=0]
2017-05-17 07:05:56.636 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=15, apiId=24 01, data=FE 0F 24 01 00 00 00 00 01 00 07 30 1F 05 00 00 00 01 00 07, checksum=07, error=false)
2017-05-17 07:05:56.639 [DEBUG] [ee.internal.ZigBeeNetworkMeshMonitor] - 8866: Starting mesh update
2017-05-17 07:05:56.640 [DEBUG] [ee.internal.ZigBeeNetworkMeshMonitor] - 8866: ZigBee node not found during mesh update
2017-05-17 07:05:56.660 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
2017-05-17 07:05:56.670 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZToolPacket (FE 18 45 FF 00 00 00 01 80 00 00 00 00 00 7C 04 EB 09 00 4B 12 00 00 00 01 00 A2 22 61)
2017-05-17 07:05:56.672 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=24, apiId=45 FF, data=FE 18 45 FF 00 00 00 01 80 00 00 00 00 00 7C 04 EB 09 00 4B 12 00 00 00 01 00 A2 22 61, checksum=61, error=false
2017-05-17 07:05:56.675 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: IeeeAddressResponse [0/0 -> 0/0, cluster=8001, TID=NULL, status=SUCCESS, ieeeAddrRemoteDev=00124B0009EB047C, nwkAddrRemoteDev=0, numAssocDev=1, startIndex=0, nwkAddrAssocDevList=[8866]]
2017-05-17 07:05:56.682 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementRoutingRequest [0/0 -> 0/0, cluster=0032, TID=08, startIndex=0]
2017-05-17 07:05:56.684 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=12, apiId=24 01, data=FE 0C 24 01 00 00 00 00 32 00 08 30 1F 02 00 00 3E, checksum=3E, error=false)
2017-05-17 07:05:56.706 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
2017-05-17 07:05:56.713 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZToolPacket (FE 0D 45 FF 00 00 00 32 80 00 00 00 00 00 00 00 00 05)
2017-05-17 07:05:56.715 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=13, apiId=45 FF, data=FE 0D 45 FF 00 00 00 32 80 00 00 00 00 00 00 00 00 05, checksum=05, error=false
2017-05-17 07:05:56.720 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementRoutingResponse [0/0 -> 0/0, cluster=8032, TID=NULL, status=SUCCESS, routingTableEntries=0, startIndex=0, routingTableListCount=0, routingTableList=[]]
2017-05-17 07:05:56.739 [DEBUG] [ee.internal.ZigBeeNetworkMeshMonitor] - 0: Ending mesh update
```

cdjackson, 2017-05-17:
Please provide the log before this that actually shows what is happening. This works fine here - what controller are you using? What versions of the binding are you using?
We can’t just use the neighbour table request as the ieee address request provides more information.

bguidinger, 2017-05-17:
ZigBee Binding v2.1.0.201705152139 (latest from your site)

CC2531 - Z-Stack Home 1.2.2a.44539 - CC2531ZNP-Pro-Secure_Standard.hex

```
2017-05-17 07:05:45.732 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false)
2017-05-17 07:05:45.744 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  SYS_VERSION_RESPONSE (FE 05 61 02 02 00 02 06 03 63)
2017-05-17 07:05:45.748 [DEBUG] [etwork.impl.ZigBeeNetworkManagerImpl] - CC2531 version is Software=2.6 Product=0 Hardware=3 Transport=2
2017-05-17 07:05:45.754 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - initResponse is JOINED
2017-05-17 07:05:45.755 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - initializeNetwork is false
2017-05-17 07:05:45.758 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 05 24, checksum=24, error=false)
2017-05-17 07:05:45.770 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 05 0B 00 00 00 00 00 00 00 67)
2017-05-17 07:05:45.773 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 06 27, checksum=27, error=false)
2017-05-17 07:05:45.784 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 06 ED 69 00 00 00 00 00 00 EB)
2017-05-17 07:05:45.790 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 07 26, checksum=26, error=false)
2017-05-17 07:05:45.803 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 07 00 B0 47 45 50 C0 D9 7E EB)
2017-05-17 07:05:45.805 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - CC2531 transport startup
2017-05-17 07:05:45.808 [DEBUG] [etwork.impl.ZigBeeNetworkManagerImpl] - Creating network as Coordinator
2017-05-17 07:05:45.810 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZDO_MSG_CB_REGISTER (Packet: subsystem=null, length=2, apiId=25 3E, data=FE 02 25 3E FF FF 19, checksum=19, error=false)
2017-05-17 07:05:45.820 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZDO_MSG_CB_REGISTER_SRSP (FE 01 65 3E 00 5A)
2017-05-17 07:05:45.823 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZDO_STARTUP_FROM_APP (Packet: subsystem=null, length=2, apiId=25 40, data=FE 02 25 40 00 00 67, checksum=67, error=false)
2017-05-17 07:05:46.469 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZDO_STARTUP_FROM_APP_SRSP (FE 01 65 40 00 24)
2017-05-17 07:05:46.472 [INFO ] [etwork.impl.ZigBeeNetworkManagerImpl] - Initialized ZigBee network with existing network state.
2017-05-17 07:05:46.485 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZDO_STATE_CHANGE_IND (FE 01 45 C0 09 8D)
2017-05-17 07:05:46.488 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: ZDO_STATE_CHANGE_IND{State=DEV_COORD_STARTED}
2017-05-17 07:05:46.494 [DEBUG] [etwork.impl.ZigBeeNetworkManagerImpl] - Started as ZigBee Coordinator
2017-05-17 07:05:46.505 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x45c0
2017-05-17 07:05:46.507 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - ZToolPacket packet not processed
2017-05-17 07:05:46.526 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_REGISTER (Packet: subsystem=null, length=9, apiId=24 00, data=FE 09 24 00 01 04 01 00 00 00 00 00 00 29, checksum=29, error=false)
2017-05-17 07:05:46.543 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_REGISTER_SRSP (FE 01 64 00 00 65)
2017-05-17 07:05:46.546 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - Registered endpoint 1 with profile: 260
2017-05-17 07:05:46.549 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 0: Scheduling node discovery
2017-05-17 07:05:46.559 [DEBUG] [ee.internal.ZigBeeNetworkMeshMonitor] - Starting mesh update task with interval of 60 seconds
2017-05-17 07:05:46.559 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 0: Starting node discovery
2017-05-17 07:05:46.573 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 05 24, checksum=24, error=false)
2017-05-17 07:05:46.584 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 05 0B 06 00 FF FE 01 64 00 05)
2017-05-17 07:05:46.592 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: IeeeAddressRequest [0/0 -> 0/0, cluster=0001, TID=00, nwkAddrOfInterest=0, requestType=1, startIndex=0]
2017-05-17 07:05:46.594 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 07 26, checksum=26, error=false)
2017-05-17 07:05:46.606 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 07 00 B0 47 45 50 C0 D9 7E EB)
2017-05-17 07:05:46.609 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialise done....... 11  27117  9140548369717374976
2017-05-17 07:05:46.655 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=15, apiId=24 01, data=FE 0F 24 01 00 00 00 00 01 00 00 30 1F 05 00 00 00 01 00 00, checksum=00, error=false)
2017-05-17 07:05:46.677 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
[... cut ...]
2017-05-17 07:06:28.465 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x45ca
2017-05-17 07:06:28.465 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - ZToolPacket packet not processed
2017-05-17 07:06:31.140 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - Ieee Address for 8866 returned null
2017-05-17 07:06:31.141 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 8866: Discovery request IEEE_ADDRESS failed. Wait before retry.
2017-05-17 07:06:32.665 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 8866: Ending node discovery
```

cdjackson, 2017-05-17:
I’ve made an update - let me know if it’s any better (or different even!). If not, then please provide the same log as earlier.

bguidinger, 2017-05-18:
Thanks, @cdjackson.  Still getting the same results.  Here's a full log starting with the controller initialization.  Let me know if I can get you any other info/logs.

```
2017-05-17 20:11:31.075 [DEBUG] [zigbee.internal.ZigBeeHandlerFactory] - Creating coordinator handler for org.eclipse.smarthome.core.thing.internal.BridgeImpl@d77f87f3
2017-05-17 20:11:31.197 [DEBUG] [ndler.ZigBeeCoordinatorCC2531Handler] - Initializing ZigBee CC2531EMK serial bridge handler.
2017-05-17 20:11:31.200 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initializing ZigBee network [zigbee:coordinator_cc2531:5d11adf0].
2017-05-17 20:11:31.205 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Channel 11
2017-05-17 20:11:31.229 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - PANID 0
2017-05-17 20:11:31.232 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - EPANID 0000000000000000
2017-05-17 20:11:31.235 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2017-05-17 20:11:31.238 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialising network
2017-05-17 20:11:31.244 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Created random ZigBee PAN ID [C7C5].
2017-05-17 20:11:31.311 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Created random ZigBee extended PAN ID [03A1B0AC61F8CC00].
2017-05-17 20:11:31.335 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key String 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2017-05-17 20:11:31.338 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key array [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2017-05-17 20:11:31.341 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key final array [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2017-05-17 20:11:31.417 [DEBUG] [ndler.ZigBeeCoordinatorCC2531Handler] - ZigBee Coordinator CC2531 opening Port:'/dev/ttyACM0' PAN:c7c5, Channel:11
2017-05-17 20:11:31.429 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Scheduling ZigBee start
2017-05-17 20:11:32.433 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee network starting
2017-05-17 20:11:32.434 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialising ZigBee coordinator
2017-05-17 20:11:32.440 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Creating ZigBee discovery service for zigbee:coordinator_cc2531:5d11adf0
2017-05-17 20:11:32.442 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Activating ZigBee discovery service for zigbee:coordinator_cc2531:5d11adf0
2017-05-17 20:11:32.461 [DEBUG] [org.openhab.binding.zigbee          ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.discovery.DiscoveryService}={service.id=325, service.bundleid=231, service.scope=singleton} - org.openhab.binding.zigbee
2017-05-17 20:11:32.530 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key initialise [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2017-05-17 20:11:32.532 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - CC2531 transport initialize
2017-05-17 20:11:32.554 [DEBUG] [ndler.ZigBeeCoordinatorCC2531Handler] - Opening ZigBee CC2531 serial port
2017-05-17 20:11:32.556 [DEBUG] [ndler.ZigBeeCoordinatorCC2531Handler] - Connecting to serial port [/dev/ttyACM0]
2017-05-17 20:11:32.666 [INFO ] [ndler.ZigBeeCoordinatorCC2531Handler] - Serial port [/dev/ttyACM0] is initialized.
2017-05-17 20:11:32.691 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  SYS_RESET (Packet: subsystem=null, length=1, apiId=41 00, data=FE 01 41 00 01 41, checksum=41, error=false)
2017-05-17 20:11:33.798 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- SYS_RESET_RESPONSE (FE 06 41 80 00 02 00 02 06 03 C2)
2017-05-17 20:11:33.802 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=6, apiId=41 80, data=FE 06 41 80 00 02 00 02 06 03 C2, checksum=C2, error=false
2017-05-17 20:11:33.810 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false)
2017-05-17 20:11:33.823 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  SYS_VERSION_RESPONSE (FE 05 61 02 02 00 02 06 03 63)
2017-05-17 20:11:33.826 [DEBUG] [etwork.impl.ZigBeeNetworkManagerImpl] - CC2531 version is Software=2.6 Product=0 Hardware=3 Transport=2
2017-05-17 20:11:33.832 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - initResponse is JOINED
2017-05-17 20:11:33.835 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - initializeNetwork is true
2017-05-17 20:11:33.839 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 05 24, checksum=24, error=false)
2017-05-17 20:11:33.851 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 05 0B 00 00 00 00 00 00 00 67)
2017-05-17 20:11:33.854 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 06 27, checksum=27, error=false)
2017-05-17 20:11:33.866 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 06 2E 64 00 00 00 00 00 00 25)
2017-05-17 20:11:33.871 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 07 26, checksum=26, error=false)
2017-05-17 20:11:33.883 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  ZB_GET_DEVICE_INFO_RSP (FE 09 66 06 07 00 48 36 86 E5 38 BA 40 B1)
[... cut ...]
2017-05-17 20:13:39.430 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - ZToolPacket packet not processed: Packet: subsystem=ZDO, length=12, apiId=45 CA, data=FE 0C 45 CA 2C BB 57 AD 26 10 01 88 17 00 00 00 46, checksum=46, error=false
2017-05-17 20:13:40.244 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - Ieee Address for 47916 returned null
2017-05-17 20:13:40.245 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 47916: Discovery request IEEE_ADDRESS failed. Wait before retry.
2017-05-17 20:13:41.746 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 47916: Ending node discovery
```

cdjackson, 2017-05-18:
As a matter of interest, have you completely reinitialised the stick, or was this initialised with another software?

bguidinger, 2017-05-19:
I was playing around with the Z-Stack 3.0 for a little bit (with some custom code).  However, I flashed the Z-Stack Home 1.2.2a right before using it with OpenHAB.  Is there another way to re-initialize the stick?  I figured flashing would take care of it.

cdjackson, 2017-05-19:
I would have expected reflashing would reinitialise the system, but maybe not as the network configuration might be in a separate flash so that firmware updates can be made without the network having to be reconfigured. In the binding, you could try setting the initialise configuration to true in the controller. This should reset the stick and reinitialise the network.

In the log I see messages showing that the devices are joining (there are some trust centre callbacks). This indicates to me that it seems to find the devices, but the fact that they don't then seem to respond to basic ZDO messages such as the IeeeAddress call suggests something fundamental is wrong (maybe wrong keys). If you have a sniffer, it would be useful to see what's happening over the air.

Have you completely reset the hue device or has it previously been included in a network? I don't know the Osram bulbs so can't comment on them, but again, it might need resetting?

bguidinger, 2017-05-19:
These bulbs have been paired with a few controllers, but I've been resetting them before joining each time.  I don't have a sniffer right now, but I may end up getting another CC2531 stick to use as one.

I did manage to have some success through the Z-Tool.  Here's a log of the controller initialization and a ZDO Device Announce message.  I'm wondering if the issue is just with it not setting the ZCD_NV_ZDO_DIRECT_CB variable to true?

Thanks for all your help with this so far!

```
Start Time: 5/19/2017 1:24:06 PM

<TX>01:24:14.08 COM4 ZB_SYSTEM_RESET (0x4609)

<TX>01:24:14.13 COM4 ZB_SYSTEM_RESET (0x4609)

<TX>01:24:15.55 COM4 ZB_SYSTEM_RESET (0x4609)

<RX>01:24:15.57 COM4 SYS_RESET_RESPONSE (0x4180)
    Reason: 0x00
    TransportRev: 0x02
    Product: 0x00
    MajorRel: 0x02
    MinorRel: 0x06
    HwRev: 0x03

<TX>01:24:15.6 COM4 ZB_WRITE_CONFIGURATION (0x2605)
    ConfigId: 0x03
    Len: 0x01
    Value: . (0x03)

<TX>01:24:16.98 COM4 ZB_WRITE_CONFIGURATION (0x2605)
    ConfigId: 0x03
    Len: 0x01
    Value: . (0x03)

<RX>01:24:17.01 COM4 SYS_RESET_RESPONSE (0x4180)
    Reason: 0x00
    TransportRev: 0x02
    Product: 0x00
    MajorRel: 0x02
    MinorRel: 0x06
    HwRev: 0x03

<RX>01:24:17.03 COM4 ZB_WRITE_CONFIGURATION_RSP (0x6605)
    Status: SUCCESS (0x0)

<RX>01:24:17.03 COM4 ZB_WRITE_CONFIGURATION_RSP (0x6605)
    Status: SUCCESS (0x0)

<TX>01:24:17.18 COM4 ZB_SYSTEM_RESET (0x4609)

<TX>01:24:17.23 COM4 ZB_SYSTEM_RESET (0x4609)

<TX>01:24:19.34 COM4 ZB_SYSTEM_RESET (0x4609)

<RX>01:24:19.37 COM4 SYS_RESET_RESPONSE (0x4180)
    Reason: 0x00
    TransportRev: 0x02
    Product: 0x00
    MajorRel: 0x02
    MinorRel: 0x06
    HwRev: 0x03

<TX>01:24:19.39 COM4 ZB_WRITE_CONFIGURATION (0x2605)
    ConfigId: 0x87
    Len: 0x01
    Value: . (0x00)

<TX>01:24:21.68 COM4 ZB_WRITE_CONFIGURATION (0x2605)
    ConfigId: 0x87
    Len: 0x01
    Value: . (0x00)

<RX>01:24:21.68 COM4 SYS_RESET_RESPONSE (0x4180)
    Reason: 0x00
    TransportRev: 0x02
    Product: 0x00
    MajorRel: 0x02
    MinorRel: 0x06
    HwRev: 0x03

<RX>01:24:21.7 COM4 ZB_WRITE_CONFIGURATION_RSP (0x6605)
    Status: SUCCESS (0x0)

<RX>01:24:21.7 COM4 ZB_WRITE_CONFIGURATION_RSP (0x6605)
    Status: SUCCESS (0x0)

<TX>01:24:21.88 COM4 ZB_WRITE_CONFIGURATION (0x2605)
    ConfigId: 0x83
    Len: 0x02
    Value: ". (0x22, 0x06)

<RX>01:24:21.89 COM4 ZB_WRITE_CONFIGURATION_RSP (0x6605)
    Status: SUCCESS (0x0)

<TX>01:24:22.08 COM4 ZB_WRITE_CONFIGURATION (0x2605)
    ConfigId: 0x84
    Len: 0x04
    Value: .... (0x00, 0x00, 0x08, 0x00)

<RX>01:24:22.09 COM4 ZB_WRITE_CONFIGURATION_RSP (0x6605)
    Status: SUCCESS (0x0)

<TX>01:24:22.28 COM4 ZB_START_REQUEST (0x2600)

<TX>01:24:23.28 COM4 ZB_START_REQUEST (0x2600)

<RX>01:24:23.78 COM4 ZB_START_REQUEST_RSP (0x6600)

<RX>01:24:24.19 COM4 ZB_START_REQUEST_RSP (0x6600)

<RX>01:24:24.19 COM4 ZDO_STATE_CHANGE_IND (0x45C0)
    State: 8 (0x8)

<RX>01:24:24.81 COM4 ZDO_STATE_CHANGE_IND (0x45C0)
    State: 9 (0x9)

<RX>01:24:25.19 COM4 ZDO_STATE_CHANGE_IND (0x45C0)
    State: 8 (0x8)

<RX>01:24:26.22 COM4 ZDO_STATE_CHANGE_IND (0x45C0)
    State: 9 (0x9)

*** At this point, I reset the light ***
<RX>01:24:55.79 COM4 ZDO_LEAVE_IND (0x45C9)
    srcAddr: 0xC72A
    extAddr: 0x7CE52400000BCC39
    request: 0x00
    removeChildren: 0x00
    rejoin: 0x00

*** Write the ZCD_NV_ZDO_DIRECT_CB variable to true ***
*** This was necessary to get the ZDO device announce ***

<TX>01:25:36.23 COM4 SYS_OSAL_NV_WRITE (0x2109)
    Id: 0x008F
    Offset: 0x00
    Len: 0x01
    Value: . (0x01)

<RX>01:25:36.24 COM4 SYS_OSAL_NV_WRITE_SRSP (0x6109)
    Status: SUCCESS (0x0)

*** At this point, I turned the light on ***
<RX>01:25:42.35 COM4 ZDO_END_DEVICE_ANNCE_IND (0x45C1)
    SrcAddr: 0x058E
    NwkAddr: 0x058E
    IEEEAddr: 0x7CE52400000BCC39
    Capabilities: 0x0E
```

cdjackson, 2017-05-19:
Thanks. I’ll take a look over this later. It’s interesting that I don’t see the TC announce messages here so maybe there’s something different in the security configuration.

cdjackson, 2017-05-20:
So hopefully you’re right and it’s just this setting. I thought I had previously set this to true, but I might have deleted that when I updated the initialisation of the 2531. Anyway, I’ve now added it so let’s see. What seems strange to me is that this works with other firmware - even if it’s reporting the same version!

bguidinger, 2017-05-21:
That did it!  I can see my Hue bulb get added as a Thing.  Thanks for getting this working!

closed 2017-05-21 by bguidinger

cdjackson, 2017-05-21:
Great stuff - thanks for the feedback :)