openhab/org.openhab.binding.zigbee #12: Representing extended PAN ids

DuncanvR, 2017-05-27:
Going through the code yesterday, I noticed the extended PAN id is stored using a `Long` (in `ZigBeeCoordinatorHandler.java`). As I understand (for example from [here](http://community.silabs.com/t5/Mesh-Knowledge-Base/What-is-an-Extended-PAN-ID-and-how-is-it-used/ta-p/113403)), the extended PAN id is a 64-bit identifier used in conjunction with the regular PAN id.

64 bits give `2^64 = 18446744073709551616 = 16^16` possible values; hence the `0xFFFFFFFFFFFFFFFF` mask at `ZigBeeCoordinatorHandler.java:178`. Java's `Long` can only represent half that range, however. While it is represented using 64 bits, it's also singed; so it has a range of [`-2^63 to 2^63 - 1`](https://docs.oracle.com/javase/7/docs/api/constant-values.html#java.lang.Long.MAX_VALUE). So in order to support all possible values, we need to change something.

I can see a couple of options:
1. Change the backing type to something with a larger precision, e.g. `BigInteger`. This is probably the easiest;
1. As per [this answer](https://stackoverflow.com/questions/508630/java-equivalent-of-unsigned-long-long/25248688#25248688), Java 8 supports [parsing](https://docs.oracle.com/javase/8/docs/api/java/lang/Long.html#parseUnsignedLong-java.lang.String) and [printing](https://docs.oracle.com/javase/8/docs/api/java/lang/Long.html#toUnsignedString-long) unsigned long values to/from the long primitive;
1. Do not store it as a number at all, but keep it as a string and convert to/from numbers where necessary.

The main issue is probably not with our code though. In the `initialiseZigBee()` method, the extended PAN id is passed through to the network manager, which is imported from `com.zsmartsystems.zigbee.ZigBeeNetworkManager`. (I wasn't able to quickly find the source of that; can I assume it was based on [`org.bubblecloud.zigbee.network.ZigBeeNetworkManager`](https://github.com/tlaukkan/zigbee4java/blob/master/zigbee-dongle-cc2531/src/main/java/org/bubblecloud/zigbee/network/ZigBeeNetworkManager.java) from [zigbee4java](https://github.com/tlaukkan/zigbee4java)?) That class stores the value as a long as well; so if we want to change it, we'll have to do it on both sides.

What do you think; is this worth chasing?

cdjackson, 2017-05-27:
Yes - I’m aware of this and the latest code has been changed to resolve the underlying problem with unsigned longs. There is now a new class called ExtendedPanId which is backed by an int array. I probably haven’t updated the binding with this change yet.

Note that zigbee4java is not used any more. I made a lot of changes to the library and tried to merge them but there was no response from the maintainer for many months so I created a new library -:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee <https://github.com/zsmartsystems/com.zsmartsystems.zigbee>

DuncanvR, 2017-05-27:
I figured as much. An int array sounds like a good solution as well. Anything I can do to help with the update?

cdjackson, 2017-05-27:
If you want to update your existing PR to use the ExtendedPanId class rather than the long in the coordinator handler, then I think that would be great :)

DuncanvR, 2017-05-30:
Sure, I will have a look. Not sure I'll find time for that this week though; I wouldn't mind creating a separate pull request either.

DuncanvR, 2017-06-14:
I don't see this `ExtendedPanId` class you mentioned used in [`ZigBeeTransportTransmit`](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/transport/ZigBeeTransportTransmit.java#L123) yet, so this'll be a bigger change than just updating the binding.

cdjackson, 2017-06-20:
Seems I created the class, but didn't update the API - I'll do so soon.

cdjackson, 2017-06-20:
It might pay to hold off on this - I might end up updating this to sort out the pan id while integrating the new library into the binding.

Stay tuned ;)

cdjackson, 2017-07-04:
This should now be resolved.

closed 2017-07-04 by cdjackson