openhab/org.openhab.binding.zigbee #22: startDeviceDiscovery() not fully implemented

saukijan, 2017-07-13:
Hey @cdjackson, I am trying to add a custom lamp, I had it working back in 2016 release, but after you refactored the code I can not seem to get it discovered. I poked around in the code and found that inside **startDeviceDiscovery()** method `networkManager.permitJoin(60);`  does not get called for some strange reason. 'permitJoin()' method is declared like this: 
```
public void permitJoin(final int duration) {
        logger.debug("Permit join for {} seconds.", duration);
        permitJoin(new ZigBeeDeviceAddress(ZigBeeBroadcastDestination.BROADCAST_ROUTERS_AND_COORD.getKey()), duration);
    }
```
 So it should log a message on DEBUG level, but it doesn't and if I add a few more log messages like this: 
```
        logger.debug("Allowing devices to join.");
        networkManager.permitJoin(60);
        logger.debug("Did networkManager log something?")
```
I only get this: 
```
14:00:54.121 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Allowing devices to join.
14:00:54.156 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Did networkManager log something?
```
Also as the title claims **startDeviceDiscovery()** method hasn't been fully implemented. I can see that you wanted to add devices from ZigBeeDevice list, but never implemented it there. Have you done so somewhere else in the code? Furthermore what did you want to move to discovery handler from this method? The whole method or just the **permitJoin()** part? What did you want to do with **ZigBeeDiscoveryManager**?

cdjackson, 2017-07-13:
I think it’s impossible that you are not getting more logged since the next line in the permit join is a log statement - are you sure you have enabled all logging including namespace com.zsmartsystems.zigbee? I think probably not. Since the joining is working for others I think it should work.

I suggest to ensure that debug is properly enabled so you can see what is happening.

cdjackson, 2017-12-30:
Closing as this is an old issue. If it still exists then please reopen and provide an updated log since the binding has undergone major refactoring since this was raised.

closed 2017-12-30 by cdjackson