openhab/org.openhab.binding.zigbee #36: [Feature Request] Add support for transition time for ZLL

openhab-5iver, 2017-11-29:
I'm using the binding for some GE Link bulbs, which I had previously configured through SmartThings to slow fade when turned off and when changing the dim setting (IIRC, these were different settings). So, I know these bulbs support it... and it would be very nice to be able to change the setting after now moving the bulbs around :).

cdjackson, 2017-12-01:
I will take a look at this soonish as I'll use this as an example of adding configuration into this binding...

cdjackson, 2018-01-13:
This is implemented, but doesn't fully work due to the following issue (https://github.com/eclipse/smarthome/issues/4910). This appears to cause the wrong type to be passed into the handler which causes ClassCast exceptions.

openhab-5iver, 2018-01-13:
Thank you! I had seen in available for my bulbs but did not test it. I'll wait now for the fix :). I could be remembering incorrectly, and haven't looked at the ST DTH to confirm, but I thought there was a separate OFF and ON transition time available in ST...

cdjackson, 2018-01-13:
Yes - I think both are implemented. There are from memory 4 or 5 settings. The binding polls the device to see which ones it supports and then makes them available for configuration. Not all will probably work at the moment, but you should get all settings available if the bulb supports them - let me know, but be careful as the linked bug can render the thing unusable if you save the configuration!

openhab-5iver, 2018-01-13:
Hmmm... then I was probably remembering correctly. I only see one...
[image]

... so maybe there are more that these bulbs support but aren't showing up. I'm not going to touch it until there's a PR to resolve that issue (thanks for the warning)!

cdjackson, 2018-01-13:
> 
> Hmmm... then I was probably remembering correctly. I only see one…
> 
Yes - you need the newest HABmin, or a future PaperUI to see the others. This one is defined in the channel-type as it’s a function of the binding - the others are defined in the channel (i.e. by the binding) as they are functions of the device and the binding will only show the ones the binding supports. This consolidation of channel configuration is new in the past few days.

openhab-5iver, 2018-01-13:
> you need the newest HABmin
O sweet... you're working on Habmin?! Is this available in a recent snapshot?

cdjackson, 2018-01-13:
Yes - it should be available now. It just fixes channel configuration issues and adds this change to request the channel config, rather than the channel-type configuration...

openhab-5iver, 2018-01-13:
Very cool! Habmin 2.3.0.201801131507  from 2.3 snapshot 1181 shows this...
[image]

Looks like some are duplicated.

openhab-5iver, 2018-02-02:
Chris, should there be duplicates here? I see two On/Off Transition Time and Off Transition Time (although this looks like the label may just need to be changed to Default On Level). I still see some in version 2.3.0.201802012251 ...
[image]

Also, is there any documentation on how/where the Default Move Rate is applied? I'll go look at the code, but others may wonder too. 😉

cdjackson, 2018-02-02:
> should there be duplicates here?
No. I've not looked at this at the moment as until the normalisation issue is resolved, changing these configurations breaks the system if they get saved as the wrong type. I don't want to mess around with this until the normalisation is resolved.

> is there any documentation on how/where the Default Move Rate is applied?
It's only used for move commands, not set commands. Currently I don't think there is any support in the binding for this item type.

> I'll go look at the code
You'd need to read the ZigBee docs I think - not the binding code.

weakfl, 2018-02-11:
Chris, did you have a chance to verify that eclipse/smarthome#4910 has been fixed by eclipse/smarthome#5018 yet?

If not, could you please update this issue when the fix is available? I'd love to try this 😉

cdjackson, 2018-02-11:
No - I didn’t get the chance as some other issues came up that I spent a lot of time looking at :( I will try and look at this today if I can.

weakfl, 2018-02-11:
> I will try and look at this today if I can.

Great, thanks!

weakfl, 2018-03-02:
Chris, since [#5018](https://github.com/eclipse/smarthome/pull/5018) has been merged meanwhile, is it now safe to change the transition time?

cdjackson, 2018-03-03:
No - #160 seems to indicate the normalisation issue is still possible.

weakfl, 2018-04-04:
I can now set *"On/Off Transition Time"* through PaperUI (dimmer channel), but it doesn't seem to have an impact. 

I cannot set the *"Default transition time"*, the setting doesn't get saved.

[image]

openhab-5iver, 2018-04-04:
I've set the Off Transition Time and do not see any effect (and some of the labels are inaccurate too)...
[image]

cdjackson, 2018-04-04:
I will look at this in a month or so. There are some labelling issues and this probably explains the apparent duplication.

cdjackson, 2018-04-20:
Note that the latest binding should improve this issue - any feedback welcome.

openhab-5iver, 2018-04-20:
I do not see an updated build yet on Cloudbees, but I'll test it when it gets there.

cdjackson, 2018-04-20:
Sorry - I assumed it had built in the last 6 days, but it seems not. I’ve just kicked off a build now...

openhab-5iver, 2018-04-21:
Np... updated now and appears to be working well. I only see one issue, or maybe it's just my understanding of the settings. The first two parameters appear to be linked (changing one changes the other).

[image]

There is also a UI oddity where some parameters have dropdowns and some having up/down arrows. Also, it would be helpful to have the range of the settings in the label or description.

cdjackson, 2018-04-21:
Glad it’s mostly working.

There should only be 6 parameters - I’m not sure why the first one is duplicated.  What is the actual device/channel that you’re showing?

The UI difference is that some have an option, and therefore show the dropdown, while others don’t have options, so just show as an integer input.

openhab-5iver, 2018-04-21:
This is the dimmer channel for a GE Link bulb.

cdjackson, 2018-04-21:
Ok, I see where the duplication is coming from - I’ll create a PR to fix this once I work out the best option...

cdjackson, 2018-04-21:
I've created #189 which should remove the duplication and then hopefully we can close this (finally ;) ).

weakfl, 2018-04-22:
Works with my BJ relays, too. Thanks Chris.

weakfl, 2018-04-22:
There seems to be an issue: after setting the 'Default Transition Time', the dimmer channel is now intermittently reporting a level of 100, although the relay is off.

cdjackson, 2018-04-22:
If the dimmer is reporting this without any binding request etc, then it sounds like an issue with the dimmer? Something I think I’ve seen elsewhere is the dimmer can report a level even if the device is off - there is a separate issue about that I think, but I don’t think it’s related to this which is about configuring the device.

weakfl, 2018-04-22:
> Something I think I’ve seen elsewhere is the dimmer can report a level even if the device is off - there is a separate issue about that I think

Yes, the BJ relays are indeed reporting a level of 1 at times. This hasn't been fixed in the BJ firmware yet, but I've been able to resolve it by using a proxy item and a filtering rule.

The level 100 reporting is definitely new though. It seems only the one relay for which I've changed the transition time is affected.
I've also updated OH today, so I can't be 100% sure. I'll keep an eye on it to see if it happens with the other relay too...

weakfl, 2018-04-23:
FWIW: some observations regarding the level issue:
- Only the one relay with transition time set to 0 is affected
- Setting the transition time to 10 resolves the reporting issue, but doesn't really work. The transition takes one second instead of 10 ms.

> I've created #189 which should remove the duplication [...]

Duplication is fixed with the latest build.

cdjackson, 2018-04-23:
> Only the one relay with transition time set to 0 is affected

As mentioned earlier, I think this is a device issue. Maybe there's something wrong somewhere, but without more information I can't say what. At the moment I consider that it's not related to this issue as clearly the setting is working.

> Setting the transition time to 10 resolves the reporting issue, but doesn't really work. The transition takes one second instead of 10 ms.

Note that the value is in 10ms steps - not 1ms steps. The default probably needs to be disabled - the issue is probably that all these settings are inter-related. I'll add an option to disable the default.

cdjackson, 2018-04-23:
> Setting the transition time to 10 resolves the reporting issue, but doesn't really work. The transition takes one second instead of 10 ms.

In fact, this is 100% correct - the time is specified in 1/10th of a second, so a value of 10 is 1 second.

cdjackson, 2018-04-23:
The update of the config descriptions is now merged and I've just kicked off a build. Hopefully this can be closed now @openhab-5iver ?

openhab-5iver, 2018-04-23:
> The update of the config descriptions is now merged and I've just kicked off a build. Hopefully this can be closed now @openhab-5iver ?

I'm trying to test now, but ran into a snag after updating OH to build 1263. It seems some of the dependencies for zwave and zigbee are not being installed.

    Unresolved requirement: Import-Package: gnu.io

I'll let you know when I get this figured out.... or rollback.

cdjackson, 2018-04-23:
gnu.io is just the serial driver nrjavaserial… Should be installed, but note that there is now a new dependency which is the USB serial bundle - this is more likely the problem and it only comes with the latest version of ESH/OH so you might need to update everything.

weakfl, 2018-04-23:
> Before you conclude on this, please look at #82 and #83. I think it's similar. In theory, I believe it's a device problem as I think that the ZigBee standards state that level and on/off state should be synchronised. My understanding of that is that a device shouldn't report OFF with a level != 0, but that said, if we see this happening on a number of devices, then we probably ought to solve it.

Understood. I'll take a look at these issues tomorrow.

The level issue definitely seems related to the transition setting somehow. With the new setting of 1(00) ms the relay is reporting a level of 10 although it's off.

cdjackson, 2018-04-24:
> Including the allowed range would be a start

I don't like to duplicate the data, so I don't really agree with this.

> currently the up/down arrows allow for negative numbers, and a max of 60,000

That's a UI issue - I'm not sure what you propose.

> What is the Default Transition Time? I could not find any reference to it in the spec.

It's the transition time that is used in the commands that are sent unless you disable it. It looks like you have not disabled it, so this will override everything and probably explains why the other settings dont work.

I'm not sure what else I can add here. If you have any ideas, then please be very specific.

cdjackson, 2018-05-17:
As above, this seems to be fully working now, so closing. If there's any further issues, please open another specific request.

closed 2018-05-17 by cdjackson