openhab/org.openhab.binding.zigbee #37: Support for ZigBee remotes

DuncanvR, 2017-11-29:
Hi @cdjackson,

I was thinking of getting a ZigBee remote, to make switching my lights as easy as it used to be with physical switches. For example the [Hue dimmer switch](http://www2.meethue.com/en-us/p/hue-dimmer-switch/046677458140), the [Lightify smart switch](https://www.sylvania.com/en-us/products/new-products/Pages/LIGHTIFY-Smart-Switch.aspx) or the [Trust ZigBee remote control](https://www.trust.com/en/product/71171-zigbee-remote-control-zyct-202).

Are there any plans to add support for remotes to this binding? I only see dongles and bulbs listed in the README at the moment, so I'm assuming they aren't already. What would be needed to implement such a feature?

Cheers!

cdjackson, 2017-12-01:
I've purchased the Trust remote so will look at this in the coming week or four.

rbi, 2018-05-31:
@cdjackson does the Trust remote work well?

I've got an Osram SMART+ Switch Mini partial working with an Qivicon USB stick and openHAB 2.3. When pressing the big arrow key it always sends ON. When pressing the small arrow key it always sends OFF. I receive no events when pressing the circle key in the middle.

I've also tried an IKEA TRÅDFRI remote. Pairing and reading the battery status works but I do not revive any events from it.

darkblackside, 2018-06-20:
A support of remotes would be great!

JochenPfeifferHD, 2018-08-08:
Any news for the remote support?

choller, 2018-09-24:
I would also love to see this happen. I have a TRADFRI switch here and as @rbi pointed out before, pairing works but none of the buttons seem to generate any events.

It would be absolutely fabulous if the buttons on this remote could be used to trigger events/commands in OpenHAB. @cdjackson could we receive the different button events on the switch channel somehow?

Willing to help debugging/testing this on my installation of course :)

hsudbrock, 2018-10-23:
I'd also like to have support for remotes in the binding :) I have looked at three types of remotes, and found that each of them models key presses differently:

| Device | Handling of key presses |
| -------- | --------------------------------|
| Bitron 902010/23 (key-chain remote control with four buttons) | Two of the buttons trigger an ON and an OFF command from the ON/OFF cluster, respectively. The other two buttons trigger a Step-with-on-off command from the LevelControl cluster, respectively (with different direction depending on which button is pressed) |
| Bitron 902010/34 (Wall-mounted device with four buttons) | Each of the four buttons triggers a Recall-Scene command from the Scenes cluster for a different endpoint for each button |
| Osram Lightify Switch (wall-mounted switch with four buttons) | Depending on the button, and depending on how the button is pressed (long press, short press), ON/OFF commands from the ON/OFF cluster, Move-with-on-off or Move commands from  the LevelControl cluster, or MoveHue and StepColorTemperature commands from the ColorControl cluster are triggered.

That said, I'd like to add support for remotes to the binding - but have to think about how to do this first, in light of that all three remotes that I checked model button presses so differently (the only common thing being that some command is triggered from some client cluster on some endpoint for each  button).

I think that the system channel types for buttons from Eclipse SmartHome should be sufficient, so that no new channel types need to be introduced here (those channel types are for trigger channels, emitting  different events depending on the type of button press).

I currently doubt that one can easily implement this using a channel converter that is applied to all devices implementing a particular client cluster, and that one maybe needs to use static thing type definitions, seeing how diverse the implementations of remotes with respect to implemented clusters appear to be.

I will think about this, and probably suggest a solution here after that.

hsudbrock, 2018-10-29:
I do not see a simple solution that automatically generates channels for a ZigBee device based on the clusters that the device announces in its simple descriptor. The reason is that the device does not tell us which commands are generated for its button. Example: The bitron remote 902010/23 generates Step-with-on-off commands, where the step mode indicates which button was pressed; another remote might generate simple Step commands (without on-off). One cannot decide for which commands trigger channels should be generated?

My suggestion is, hence, to configure remote controls using static thing types, where the commands that are triggered when pressing a button on the remote control are configured as channel properties. A converter for such channels then configures itself based on the channel properties. I will create a PR for discussing this approach based on a first implementation.

hsudbrock, 2018-10-29:
Here's the PR mentioned in the previous comment: https://github.com/openhab/org.openhab.binding.zigbee/pull/284

hsudbrock, 2019-03-06:
@cdjackson Can we close this issue, as there is now support for some remotes? I think that not all remotes that have been mentioned in this issue have been integrated, but I also think that people could open dedicated issues for single remotes that they think should be integrated. WDYT?

cdjackson, 2019-03-06:
Thanks Henning - we can close.

closed 2019-03-06 by cdjackson