openhab/org.openhab.binding.zigbee #38: Add documentation for textual config

Orfait, 2017-11-29:
Hi,

In most of openhab configs I've seen until now, the config rely only on .things and .items files.
The documentation does not mention how to write a correct config file for this binding.
I could be very helpful to give, at least, a sample config.

Regards,
Albin

cdjackson, 2017-11-29:
Hi,
For ZigBee things, the only configuration required is to set the mac address with the zigbee_macaddress=“” configuration. For the coordinator, the configuration settings are in the coordinator XML files.

It shouldn’t be too hard from this to work out the configuration as the format etc is obviously the same as every binding. It would be really great if you would be able to help document this? :)

Chris

Orfait, 2017-11-29:
Of course, it should not be very hard.

In my situation, I tried something but I'm getting errors, so I don't know where is the problem.
Thing file is almost simple

```
Bridge zigbee:coordinator_cc2531:gw "Zigbee Gateway" [ zigbee_port="/dev/ttyACM0", zigbee_networkkey="5A 69 67 42 65 65 41 6C 6C 69 61 6E 63 65 30 39" ] {
  Thing zigbee:device:gw:00124b000e8969bd "Zigbee thing" [ zigbee_macaddress="00124B000E8969BD" ]
}
```

I got this error with only the bridge :
`2017-11-29 22:54:31.075 [INFO ] [org.quartz.core.QuartzScheduler     ] - Scheduler openHAB-job-scheduler_$_NON_CLUSTERED started.
2017-11-29 22:54:31.116 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while calling thing updated at ThingHandler 'org.openhab.binding.zigbee.handler.ZigBeeCoordinatorCC2531Handler@e85eac': null
java.util.concurrent.TimeoutException: null
	at java.util.concurrent.FutureTask.get(FutureTask.java:205) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:199) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:88) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:72) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingManager.thingUpdated(ThingManager.java:429) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyTrackers(ThingRegistryImpl.java:222) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:145) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:1) ~[?:?]
	at org.eclipse.smarthome.core.common.registry.AbstractRegistry.updated(AbstractRegistry.java:188) ~[?:?]
	at org.eclipse.smarthome.core.common.registry.AbstractRegistry.updated(AbstractRegistry.java:1) ~[?:?]
	at org.eclipse.smarthome.core.common.registry.AbstractProvider.notifyListeners(AbstractProvider.java:62) ~[?:?]
	at org.eclipse.smarthome.core.common.registry.AbstractProvider.notifyListenersAboutUpdatedElement(AbstractProvider.java:86) ~[?:?]
	at org.eclipse.smarthome.model.thing.internal.GenericThingProvider.lambda$21(GenericThingProvider.java:899) ~[?:?]
	at java.util.ArrayList.forEach(ArrayList.java:1255) ~[?:?]
	at org.eclipse.smarthome.model.thing.internal.GenericThingProvider.createThingsFromModelForThingHandlerFactory(GenericThingProvider.java:906) ~[?:?]
	at org.eclipse.smarthome.model.thing.internal.GenericThingProvider.lambda$3(GenericThingProvider.java:299) ~[?:?]
	at java.lang.Iterable.forEach(Iterable.java:75) [?:?]
	at org.eclipse.smarthome.model.thing.internal.GenericThingProvider.createThingsFromModel(GenericThingProvider.java:301) [152:org.eclipse.smarthome.model.thing:0.9.0.201711242229]
	at org.eclipse.smarthome.model.thing.internal.GenericThingProvider.modelChanged(GenericThingProvider.java:741) [152:org.eclipse.smarthome.model.thing:0.9.0.201711242229]
	at org.eclipse.smarthome.model.core.internal.ModelRepositoryImpl.notifyListeners(ModelRepositoryImpl.java:314) [135:org.eclipse.smarthome.model.core:0.9.0.201711242229]
	at org.eclipse.smarthome.model.core.internal.ModelRepositoryImpl.addOrRefreshModel(ModelRepositoryImpl.java:143) [135:org.eclipse.smarthome.model.core:0.9.0.201711242229]
	at org.eclipse.smarthome.model.core.internal.folder.FolderObserver.checkFile(FolderObserver.java:247) [135:org.eclipse.smarthome.model.core:0.9.0.201711242229]
	at org.eclipse.smarthome.model.core.internal.folder.FolderObserver.processWatchEvent(FolderObserver.java:311) [135:org.eclipse.smarthome.model.core:0.9.0.201711242229]
	at org.eclipse.smarthome.core.service.WatchQueueReader.run(WatchQueueReader.java:209) [110:org.eclipse.smarthome.core:0.9.0.201711242229]
	at java.lang.Thread.run(Thread.java:748) [?:?]`

and the thing generate a warning :
`2017-11-29 22:59:56.142 [INFO ] [el.core.internal.ModelRepositoryImpl] - Validation issues found in configuration model 'zigbee.things', using it anyway:
Provide a thing type ID and a thing ID in this format:
 <thingTypeId> <thingId>`

And an error :
`2017-11-29 23:00:01.238 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occurred while initializing handler of thing 'zigbee:device:gw:00124b000e8969bd': java.lang.NumberFormatException: For input string: " 12 4B "
java.util.concurrent.ExecutionException: java.lang.NumberFormatException: For input string: " 12 4B "
	at java.util.concurrent.FutureTask.report(FutureTask.java:122) ~[?:?]
	at java.util.concurrent.FutureTask.get(FutureTask.java:206) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:199) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:88) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:72) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingManager.doInitializeHandler(ThingManager.java:639) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingManager.initializeHandler(ThingManager.java:574) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingManager.thingUpdated(ThingManager.java:446) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyTrackers(ThingRegistryImpl.java:222) ~[?:?]
	at org.eclipse.smarthome.core.thing.internal.ThingRegistryImpl.notifyListenersAboutUpdatedElement(ThingRegistryImpl.java:145) ~[?:?]
[... cut ...]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$6.call(ThingManager.java:1) ~[?:?]
	at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:186) ~[?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) ~[?:?]
	... 1 more`

Could you tell me if the error is from the config file ?

cdjackson, 2017-11-29:
I'm not sure - it's not getting to the binding, but it is likely something to do with the configuration...

> NumberFormatException: For input string: " 12 4B "

This seems to be a strange error - I don't see this string in this format in your configuration. It's part of the Mac address and the error is converting this into the structure required in the binding, but I'm not sure why it would be split here with spaces. The binding doesn't split this - it just passes it straight through to create the address.

I will add an extra debug to the binding startup to see what is actually being received before it makes the call that is causing the exception.

Thanks.

cdjackson, 2017-12-06:
@Orfait I realise I didn't explicitly say, but I added the extra debugging when the thing is initialised. You should see -:

```
Initializing ZigBee thing handler {UID}, Ieee="{IEEE}"
```

It would be interesting to see what this shows...

cdjackson, 2019-08-10:
This is nearly 2 years old and documentation has since been updated. If there are further updates suggested, then please provide a PR.

closed 2019-08-10 by cdjackson