openhab/org.openhab.binding.zigbee #55: Telegesis Stick Configuration

jn-se, 2017-12-09:
Does anyone establish a successful connection to the Telegesis ETRX3USB stick in openHab?

I can start the binding but no devices are found during scan.

My system is a Raspberry PI with Ubuntu.

My system settings:
export EXTRA_JAVA_OPTS=-Dgnu.io.rxtx.SerialPorts=/dev/ttyUSB0
sudo chmod 666 /dev/ttyUSB0

I installed the the Zigbee binding snapshot: org.openhab.binding.zigbee-2.2.0-SNAPSHOT.jar

openHab logfile:

`[DEBUG] [zigbee.internal.ZigBeeHandlerFactory] - Creating coordinator handler for org.eclipse.smarthome.core.thing.internal.BridgeImpl@d035b982
[DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Creating ZigBee discovery service for zigbee:coordinator_telegesis:9a1e58bd
[DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Activating ZigBee discovery service for zigbee:coordinator_telegesis:9a1e58bd
[DEBUG] [org.openhab.binding.zigbee          ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.discovery.DiscoveryService}={service.id=347, service.bundleid=224, service.scope=singleton} - org.openhab.binding.zigbee
[DEBUG] [org.openhab.binding.zigbee          ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.core.thing.binding.firmware.FirmwareUpdateHandler}={service.id=348, service.bundleid=224, service.scope=singleton} - org.openhab.binding.zigbee
[DEBUG] [er.ZigBeeCoordinatorTelegesisHandler] - Initializing ZigBee Telegesis serial bridge handler.
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initializing ZigBee network [zigbee:coordinator_telegesis:9a1e58bd].
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Channel -1
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - PANID 0
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - EPANID 0000000000000000
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key null
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialising network
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Created random ZigBee PAN ID [3165].
[ome.core.thing.internal.ThingManager] - Initializing handler for thing 'zigbee:coordinator_telegesis:9a1e58bd' takes more than 5000ms.
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Created random ZigBee extended PAN ID [A74769766C2B9E92].
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key String 
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key initialised 7070CCB0ECE8D086AACB01EBF9099995
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key final array 7070CCB0ECE8D086AACB01EBF9099995
[DEBUG] [er.ZigBeeCoordinatorTelegesisHandler] - ZigBee Coordinator Telegesis opening Port:'/dev/ttyUSB0' PAN:3165, EPAN:A74769766C2B9E92, Channel:-1
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Scheduling ZigBee start
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee network starting
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialising ZigBee coordinator
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Key initialise 7070CCB0ECE8D086AACB01EBF9099995
[DEBUG] [ing.zigbee.internal.ZigBeeSerialPort] - Connecting to serial port [/dev/ttyUSB0] at 19200 baud, flow control FLOWCONTROL_OUT_NONE.
[INFO ] [ing.zigbee.internal.ZigBeeSerialPort] - Serial port [/dev/ttyUSB0] is initialized.
[DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: done.
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - initResponse is JOINED
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - initializeNetwork is true
[DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialise done....... 25  12645  A74769766C2B9E92
[DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Starting ZigBee scan for zigbee:coordinator_telegesis:9a1e58bd`

How can I verify that the Telegesis stick works and where the problem might originate?

cdjackson, 2017-12-09:
> Does anyone establish a successful connection to the Telegesis ETRX3USB stick in openHab?

Yes - it works fine.

Enable full debugging with ```log:set debug com.zsmartsystems.zigbee```

Please use code fences when posting logs - it's very difficult to read without this.

I suggest to discuss the issue on the forum so will close this here. If we find a bug, then we can open something specific to resolve it, but this should work ok.

cdjackson, 2017-12-09:
Just to add - I edited your post to add the fences, and everything in the log looks fine.

closed 2017-12-09 by cdjackson

jn-se, 2017-12-09:
I started a thread and posted new log messages.

Nevertheless I would prefer more documentation. There are several configuration options that are not explained.