openhab/org.openhab.binding.zigbee #60: Innr (RS 125 GU10) bulb support

iomode, 2017-12-14:
Hi,
I am using a TI CC2531EMK dongle with openHab2 to control my bulbs, a Philips Hue and an Innr GU10.
The Hue can be connected successfully while the Innr does not show up in the UI or the org.openhab.binding.zigbee logs.

If I enable the debug logs for com.zsmartsystems.zigbee I can see the device address (00158D0001C8A692) showing up but I am not able to interpret the log.

This is a log snippet shortly after I started a search in the UI (all other ZigBee devices are offline):
[debug.log](https://github.com/openhab/org.openhab.binding.zigbee/files/1560917/debug.log)

I have also a "ConBee" dongle from dresden elektronik which is not compatible with openHab but has its own software "DeCONZ". Using this software I am able to connect both bulbs without any problems so i guess it should also work with openHab.

Anyway, I am not sure if this is a bulb related issue or a general bug.

cdjackson, 2017-12-14:
I will take a look at this in a couple of weeks (after the 27th Dec). I do actually have an Innr bulb in my box of test devices so can probably test this.

cdjackson, 2017-12-28:
It seems that these devices are communicating via ZLL, not HA. I've tried sending commands via ZLL profile, but it's not having it. ZDO works, but not ZCL, so the initial join is ok, but all the more detailed stuff doesn't respond.

I note it also doesn't work with SmartThings (https://community.smartthings.com/t/innr-gu10-compatability/56904).

I'll keep playing, but thought I'd add this update anyway...

cdjackson, 2018-01-13:
Earlier this month I emailed Innr support and they replied to say that they had sent the request to their technical department but they might not reply for a long time as they were very busy...

cdjackson, 2018-01-15:
I've just tested this with an Ember stick and it seems to work -:

[image]

[image]

@iomode Looking at your log I don't really see a reason for it not to work - it failed getting the IEEE address which I think it should respond to ok no matter what since it is not related to the profile (being ZDP).

I need to go back and try the Telegesis stick to see if it works there. I also note that having tried again with the Ember it seems to be using ZHA rather than ZLL so maybe the device is changing modes based on something...

iomode, 2018-01-16:
@cdjackson Thank you for looking into this issue.
I have done a fresh install of openHab 2.2 ( used 2.1 before) but I am still not able to find the innr bulb using the TI dongle.
Did you also test it yourself with the TI CC2531EMK ?

cdjackson, 2018-01-16:
No - I have not tried this with a TI dongle - just the Ember. I will see if I can also test with the TI dongle in the coming days.

cdjackson, 2018-01-17:
I confirm it works ok here with a TI dongle as well.

iomode, 2018-01-17:
Strange, you have a different model but they should have a similar firmware.
Maybe the "deCONZ" software changed some settings which makes the bulb unresponsive.
I can control the bulb using deCONZ software without problems. Before I try to add it to openHAB I reset the device with the deCONZ software which makes the bulb blink and removes it from the network.

Anyway, I will try to get a second innr bulb to test it.

iomode, 2018-01-20:
I got a new RS125 bulb and successfully connected it to openHab.

From the visual point of view the only difference is the manufacturing code on the case:  
1726-3 vs 1731-6 (new)
I guess 17 is for 2017 and 26/31 is for the week.

Anyway, I think the problem was the "deCONZ" software which changed something persistently.
I think we can close this issue now.

By the way, I don't now if I should open a issue for that but typically when I add a new device the device is always found as "Unknown ZigBee Device ...." while the name is correctly displayed in the properties.
But sometimes it is displayed correctly with the name. Any ideas why?

zigbee_logicaltype | ROUTER
-- | --
zigbee_powerlevel | FULL
modelId | RS 125
zigbee_networkaddress | 603
zigbee_powersource | MAINS
zigbee_stkversion | 2
zigbee_datecode | 20161202-140
zigbee_zclversion | 3
vendor | innr
zigbee_appversion | 1
zigbee_powermode | RECEIVER_ON_IDLE
zigbee_powersources | [MAINS]
hardwareVersion | 1
Current firmware version: 11060002

closed 2018-01-20 by iomode

cdjackson, 2018-01-20:
Glad to hear it's working :)

> By the way, I don't now if I should open a issue for that but typically when I add a new device the device is always found as "Unknown ZigBee Device ...." while the name is correctly displayed in the properties.

If you use PaperUI, there is currently a bug that doesn't update the inbox. The binding does a 2 stage inbox entry - first with the unknown name, then with the correct name. The reason is that if it's a battery device, it may take 10 or 20 seconds to discover the properties to give it a proper name (since battery devices only wake every 7 seconds). I didn't want to wait this time before adding something into the inbox, otherwise people have no feedback and will think nothing is happening.

The inbox should update, but current doesn't - see https://github.com/eclipse/smarthome/issues/4921 - it's being worked on.

iomode, 2018-01-20:
Yes I am using PaperUI, thank your for help.