openhab/org.openhab.binding.zigbee #67: Exception on SwitchOnoff

zalke, 2017-12-31:
When i toggle the switch from OH the following exception occurs

 ` Exception in thread "pool-52-thread-25" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
 #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:127)
#011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:777)
 #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
 #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
 #011at java.lang.Thread.run(Thread.java:748)`

Looks like the SwitchOnoff Converter expects a Bool but it is receiving a Integer, may I suggest that the value should be checked for the type, the same way its done in the handleCommand function in the same file?

Thanks.  

Solomon M

cdjackson, 2017-12-31:
Please provide a debug log showing the response.

The attribute type should be deterministic - it’s defined by ZCL so this is not the same as for the handleCommand method which can receive commands of different types.

Thanks
Chris

zalke, 2018-01-01:
hope this is enough of  log.
this happens on device discovery

```09:50:05.471 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=3]
09:50:05.471 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting input clusters [0, 5, 4, 6]
09:50:05.489 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - Sent queue larger than window [1 > 1].
09:50:05.543 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting cluster BASIC as server
09:50:05.564 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameAck [ackNum=5]
09:50:05.563 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=5, ackNum=3, reTx=true, data=F8 00 34 00 00 00 00 00 32 00 00 00 40 11 00 00 F8 F8 02 00 00]
09:50:05.575 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting cluster SCENES as server
09:50:05.628 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - Sent queue larger than window [1 > 1].
09:50:05.644 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting cluster GROUPS as server
09:50:05.679 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting cluster ON_OFF as server
09:50:05.698 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting output clusters [0]
09:50:05.717 [DEBUG] [m.zsmartsystems.zigbee.ZigBeeEndpoint] - 7287/1: Setting cluster BASIC as client
09:50:05.739 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - 00124B000880C3D4: Node 7287 is added to the network
09:50:05.773 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000880C3D4: Starting ZigBee device discovery
09:50:05.787 [DEBUG] [rnal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: done.
09:50:05.774 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameNak [ackNum=5]
09:50:05.797 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000880C3D4: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_ember:9c034c43
09:50:05.833 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=5, ackNum=3, reTx=true, data=F8 00 34 00 00 00 00 00 32 00 00 00 40 11 00 00 F8 F8 02 00 00]
09:50:05.867 [DEBUG] [.zigbee.internal.ZigBeeHandlerFactory] - Creating coordinator handler for org.eclipse.smarthome.core.thing.internal.ThingImpl@7f0684df
09:50:05.866 [INFO ] [smarthome.event.InboxRemovedEvent    ] - Discovery Result with UID 'zigbee:device:9c034c43:00124b000880c3d4' has been removed.
09:50:05.943 [DEBUG] [org.openhab.binding.zigbee           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.core.thing.binding.firmware.FirmwareUpdateHandler}={service.id=453, service.bundleid=242, service.scope=singleton} - org.openhab.binding.zigbee
09:50:05.890 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - Sent queue larger than window [1 > 1].
09:50:06.024 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:device:9c034c43:00124b000880c3d4' changed from UNINITIALIZED to INITIALIZING
09:50:06.022 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - Initializing ZigBee thing handler zigbee:device:9c034c43:00124b000880c3d4, Ieee="00124B000880C3D4".
09:50:06.042 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameAck [ackNum=6]
09:50:06.095 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:device:9c034c43:00124b000880c3d4' changed from INITIALIZING to UNKNOWN
09:50:06.094 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Coordinator status changed to ONLINE.
09:50:06.112 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=3, ackNum=6, reTx=false, data=F8 80 34 00 65]
09:50:06.162 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Coordinator is ONLINE. Starting device initialisation.
09:50:06.185 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - RX EZSP: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=101]
09:50:06.250 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Initialising node
09:50:06.247 [INFO ] [smarthome.event.InboxAddedEvent      ] - Discovery Result with UID 'zigbee:device:9c034c43:00124b000880c3d4' has been added.
09:50:06.242 [INFO ] [ig.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:9c034c43:00124b000880c3d4' to inbox.
09:50:06.240 [DEBUG] [gbee.internal.ZigBeeNetworkDiscoverer] - Starting node rediscovery of 00124B000880C3D4
09:50:06.350 [DEBUG] [iscovery.ZigBeeNodePropertyDiscoverer] - 00124B000880C3D4: ZigBee node property discovery start
09:50:06.312 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Created 1 channels
09:50:06.282 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=101]
09:50:06.453 [DEBUG] [.converter.ZigBeeConverterSwitchOnoff] - 00124B000880C3D4: Initialising device on/off cluster
09:50:06.450 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Command for channel zigbee:device:9c034c43:00124b000880c3d4:00124B000880C3D4_1_switch_onoff --> REFRESH
09:50:06.438 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:device:9c034c43:00124b000880c3d4' has been updated.
09:50:06.390 [DEBUG] [iscovery.ZigBeeNodePropertyDiscoverer] - 00124B000880C3D4: ZigBee node property discovery using 7287/1
[... cut ...] RY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=254], messageTag=254, messageContents=00 FE 00 05 00]
09:50:13.661 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=107]
09:50:13.729 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=4, reTx=false, data=FE 00 34 00 77 1C 04 01 00 00 01 01 40 11 00 00 FE FE 05 00 FE 00 05 00]
```

cdjackson, 2018-01-01:
Thanks - I’ll take a look shortly. What device(s) does this happen with?

zalke, 2018-01-01:
Its a in wall us switch made by orvibo

cdjackson, 2018-01-01:
I confirmed that ZCL defines the type as a BOOLEAN and not as an integer as the device is reporting -:

09:50:12.811 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [On/Off: 7287/1 -> 0/1, cluster=0006, TID=FC, records=[Read Attribute Status Record: attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, status=0, attributeValue=1]]
[image]

I've opened an [issue](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/160) to look at how to handle this.

zalke, 2018-01-01:
thanks much

cdjackson, 2018-01-04:
I've added a system to normalise data types - hopefully this should now be fixed with the latest binding.

zalke, 2018-01-04:
@cdjackson 
thanks so much for the update, the debug does not report errors anymore, but in the syslog I'm getting this every 10 seconds

```Jan  4 17:24:08 raspberrypi karaf[29812]: Exception in thread "pool-50-thread-89" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
Jan  4 17:24:08 raspberrypi karaf[29812]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 17:24:08 raspberrypi karaf[29812]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 17:24:08 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 17:24:08 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 17:24:08 raspberrypi karaf[29812]: #011at java.lang.Thread.run(Thread.java:748)
Jan  4 17:24:18 raspberrypi karaf[29812]: Exception in thread "pool-50-thread-92" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
Jan  4 17:24:18 raspberrypi karaf[29812]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 17:24:18 raspberrypi karaf[29812]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 17:24:18 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 17:24:18 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 17:24:18 raspberrypi karaf[29812]: #011at java.lang.Thread.run(Thread.java:748)
Jan  4 17:24:28 raspberrypi karaf[29812]: Exception in thread "pool-50-thread-91" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
Jan  4 17:24:28 raspberrypi karaf[29812]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 17:24:28 raspberrypi karaf[29812]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 17:24:28 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 17:24:28 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 17:24:28 raspberrypi karaf[29812]: #011at java.lang.Thread.run(Thread.java:748)
Jan  4 17:24:38 raspberrypi karaf[29812]: Exception in thread "pool-50-thread-86" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
Jan  4 17:24:38 raspberrypi karaf[29812]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 17:24:38 raspberrypi karaf[29812]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 17:24:38 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 17:24:38 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 17:24:38 raspberrypi karaf[29812]: #011at java.lang.Thread.run(Thread.java:748)
Jan  4 17:24:48 raspberrypi karaf[29812]: Exception in thread "pool-50-thread-96" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
Jan  4 17:24:48 raspberrypi karaf[29812]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 17:24:48 raspberrypi karaf[29812]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 17:24:48 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 17:24:48 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 17:24:48 raspberrypi karaf[29812]: #011at java.lang.Thread.run(Thread.java:748)
Jan  4 17:24:58 raspberrypi karaf[29812]: Exception in thread "pool-50-thread-95" java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Boolean
Jan  4 17:24:58 raspberrypi karaf[29812]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 17:24:58 raspberrypi karaf[29812]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 17:24:58 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 17:24:58 raspberrypi karaf[29812]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 17:24:58 raspberrypi karaf[29812]: #011at java.lang.Thread.run(Thread.java:748)
```

cdjackson, 2018-01-04:
Thanks for testing. I’ve run some more static tests within the library since I can’t replicate with a real device and within the library the conversion seems to work fine. The error is basically the same as before - can you provide a short debug log as well please.

cdjackson, 2018-01-04:
Please also double check that you have really the latest version - the binding was actually updated until about an hour after I posted the message to say it was updated.

zalke, 2018-01-04:
Version
```
242 │ Active   │  80 │ 2.3.0.201801042145     │ ZigBee Binding
```

OH debug:
```
18:40:58.153 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Polling...
18:40:58.172 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000880C3D4: Polling zigbee:device:9c034c43:00124b000880c3d4:00124B000880C3D4_1_switch_onoff
18:40:58.198 [DEBUG] [m.zsmartsystems.zigbee.zcl.ZclCluster] - readSync request: ZclAttribute [cluster=ON_OFF, id=0, name=OnOff, dataType=BOOLEAN, lastValue=0]
18:40:58.221 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [On/Off: 0/0 -> 7287/1, cluster=0006, TID=53, identifiers=[0]]
18:40:58.236 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=83, commandId=0]
18:40:58.258 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=7287/1, profile=0104, cluster=6, addressMode=DEVICE, radius=31, sequence=83, payload=00 53 00 00 00]
18:40:58.278 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - EzspSendUnicastRequest [type=EMBER_OUTGOING_DIRECT, indexOrDestination=7287, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=83], messageTag=83, messageContents=00 53 00 00 00]
18:40:58.312 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - TX EZSP queue: 1
18:40:58.322 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - TX EZSP: EzspSendUnicastRequest [type=EMBER_OUTGOING_DIRECT, indexOrDestination=7287, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=83], messageTag=83, messageContents=00 53 00 00 00]
18:40:58.357 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=1, ackNum=5, reTx=false, data=53 00 34 00 77 1C 04 01 06 00 01 01 40 11 00 00 53 53 05 00 53 00 00 00]
18:40:58.523 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=5, ackNum=2, reTx=false, data=53 80 34 00 CF]
18:40:58.535 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - RX EZSP: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=207]
18:40:58.545 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=207]
18:40:58.556 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=207]
18:40:58.567 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=6]
18:40:58.581 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=2, reTx=false, data=53 90 3F 00 77 1C 04 01 06 00 01 01 40 11 00 00 CF 53 00 00]
18:40:58.595 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - RX EZSP: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=7287, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=207], messageTag=83, status=EMBER_SUCCESS, messageContents=]
18:40:58.626 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=7287, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=207], messageTag=83, status=EMBER_SUCCESS, messageContents=]
18:40:58.654 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=7287, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=207], messageTag=83, status=EMBER_SUCCESS, messageContents=]
18:40:58.682 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=7]
18:40:58.693 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=7, ackNum=2, reTx=false, data=53 90 45 00 04 01 06 00 01 01 40 01 00 00 C6 FE C0 77 1C FF FF 08 18 53 01 00 00 00 20 00]
18:40:58.709 [DEBUG] [gbee.dongle.ember.ash.AshFrameHandler] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=198], lastHopLqi=254, lastHopRssi=-64, sender=7287, bindingIndex=255, addressIndex=255, messageContents=18 53 01 00 00 00 20 00]
18:40:58.736 [DEBUG] [.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCO [... cut ...]
Jan  4 18:40:58 raspberrypi karaf[5575]: #011at org.openhab.binding.zigbee.converter.ZigBeeConverterSwitchOnoff.attributeUpdated(ZigBeeConverterSwitchOnoff.java:115)
Jan  4 18:40:58 raspberrypi karaf[5575]: #011at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:774)
Jan  4 18:40:58 raspberrypi karaf[5575]: #011at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
Jan  4 18:40:58 raspberrypi karaf[5575]: #011at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
Jan  4 18:40:58 raspberrypi karaf[5575]: #011at java.lang.Thread.run(Thread.java:748)
```

cdjackson, 2018-01-04:
Thanks. I think I’ve found the issue - I’ll do another PR tomorrow as it’s getting a bit late here.

zalke, 2018-01-05:
Thanks so much, atleast its working, I can now control the switch from OH, looks like its just the pulling thats not working.

On a side note looking at the logs it seems to me like there are duplicate requests at the same time, am i correct? does is suppose to be this way?

cdjackson, 2018-01-05:
@zalke the latest version once built will hopefully resolve the exception related to the polling.

zalke, 2018-01-07:
@cdjackson 
I’m so excited to report that everything looks good now.

I do see another log that might be intresting to look at but i’m not sure it related.

```
Jan  6 19:50:51 raspberrypi kernel: [181077.973620] cp210x ttyUSB1: failed set request 0x7 status: -32
Jan  6 19:53:25 raspberrypi kernel: [181231.820025] cp210x ttyUSB1: failed set request 0x7 status: -32
Jan  6 19:53:25 raspberrypi kernel: [181231.823648] cp210x ttyUSB1: failed set request 0x7 status: -32
Jan  6 19:55:37 raspberrypi kernel: [181363.353639] cp210x ttyUSB1: failed set request 0x7 status: -32
Jan  6 19:55:37 raspberrypi kernel: [181363.366495] cp210x ttyUSB1: failed set request 0x7 status: -32
```

cdjackson, 2018-01-07:
> I do see another log that might be intresting to look at but i’m not sure it related.

I have no idea what this is - it's something in the linux driver so is not something that the binding has any control over. Maybe there was a communication issue with the stick - I don't know at all.

Given that this is now working, I'll close - thanks.

closed 2018-01-07 by cdjackson