openhab/org.openhab.binding.zigbee #81: Consolidated channels reporting conflicting attributes

weakfl, 2018-01-09:
This might have to do with channel consolidation, the reported attributes are inverted depending on the channel.

The On/Off channel seems to be correct, the Level channel is always reporting the same value.

```
14:17:50.513 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [On/Off: 45692/18 -> 0/1, cluster=0006, TID=D7, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=false]]
14:17:50.531 [INFO ] [smarthome.event.ItemStateChangedEvent] - LivingroomLights1 changed from ON to OFF

14:17:50.568 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Level Control: 45692/18 -> 0/1, cluster=0008, TID=D8, reports=[Attribute Report: attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, attributeValue=254]]
14:17:50.581 [INFO ] [smarthome.event.ItemStateChangedEvent] - LivingroomLights1 changed from OFF to ON

14:47:52.224 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [On/Off: 45692/18 -> 0/1, cluster=0006, TID=DD, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=true]]

14:47:52.265 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Level Control: 45692/18 -> 0/1, cluster=0008, TID=DE, reports=[Attribute Report: attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, attributeValue=254]]
```

[complete log including startup](https://github.com/openhab/org.openhab.binding.zigbee/files/1615246/attribute_reporting_20180109.log.zip)

cdjackson, 2018-01-09:
The channel consolidation just removes channels - that's all. It doesn't combine channels or anything like that. It just says "if there's a dimmer, and a switch, then delete the switch and only use the dimmer".

Here, the device is reporting a dimmer value of 254 at all times - that's not really a problem with the binding I think. I might be wrong, but I think the device is meant to report the current value, and it's meant to be linked to the on/off channel.

I'll take a look at how better to handle this - it probably requires some logic in the converter.

weakfl, 2018-01-09:
When discussing PR [#66](https://github.com/openhab/org.openhab.binding.zigbee/pull/66#issuecomment-354654972) you mentioned that a switch widget would use the on/off cluster, even if the associated item was linked to a level channel.
It looks a bit like the level value doesn't get set if the switch widget uses the on/off cluster.

cdjackson, 2018-01-09:
I have changed this back to only use the level cluster - it should solve this.

weakfl, 2018-01-09:
I'm sure this will most likely resolve the issue, but it will also cause a delay when switching off relays. This is really noticeable when the level cluster is used.
Can't you just revert channel consolidation? I know this is quite a specific issue, but I'd prefer options over simplicity in this case ;)

cdjackson, 2018-01-09:
> Can't you just revert channel consolidation?

This isn't about simplicity - this is required by ESH concepts - there is not meant to be an OnOff channel AND a Dimmer channel.

> I'm sure this will most likely resolve the issue, but it will also cause a delay when switching off relays. 

I suspect that's just a configuration issue? If we set the delay to 0, then it should be instantaneous I expect?

weakfl, 2018-01-09:
> I suspect that's just a configuration issue? If we set the delay to 0, then it should be instantaneous I expect?

I'm afraid I'm not following, sorry. Do you mean setting a delay in the binding, when it configures devices?

cdjackson, 2018-01-09:
The delay for turning a device on is configurable. The binding uses 1 second but default, but it can be changed. Unfortunately, right now the configuration doesn’t work due to a problem in ESH. I just tested this, and it works with a Hue lamp - if I set it to 0, and set the level, it turns on instantly.

weakfl, 2018-01-09:
Sounds good, thanks.

weakfl, 2018-01-10:
I can confirm this is fixed in the latest version.

I'm just wondering why I only get a delay when switching the relay off. If it's set to 1 second in general, shouldn't there be a delay when turning the relay on too?

closed 2018-01-10 by weakfl

cdjackson, 2018-01-10:
I don't know - I guess it's down to the device itself. All commands are sent with the 1 second time and I note my Hue devices fade up and down in that time. If I set the time to 0, then the Hue bulbs switch on/off instantly.