openhab/org.openhab.binding.zigbee #82: Switch does not set HSB and Brightness Channels to zero

BClark09, 2018-01-09:
### Outline

When trying to use a switch item on a Colour or Brightness channel, I notice that the 'B' of the HSB, and the Brightness linked items never go to zero, only to very low values 0.39370078. Since they're non-zero, this seems to flick the switch back on after a very short time after it polls. See the event log for a logical summary.

Could be related to #81 but since I've got a slightly different result I thought it be best to make a new issue.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  Telegisis                                                                              |
| openHAB version  | 2.3.0-SNAPSHOT (#1176) [2.3.0~20180109005608]  |
| Hardware         |  RPi 3                                                                            |
| Memory           |  ?                                                                              |
| Java version     |   Zulu Embedded 8.25.0.76-linux-aarch32hf (build 1.8.0_152-b76) |
| Devices          |  3 Hue Colour Lights, 3 Hue Brightness Lights |

### Logs

#### Event Log

```
2018-01-09 15:49:10.194 [ome.event.ItemCommandEvent] - Item 'Bedroom_Light_Switch' received command OFF
2018-01-09 15:49:10.205 [vent.ItemStateChangedEvent] - Bedroom_Light_Switch changed from ON to OFF
2018-01-09 15:49:12.479 [vent.ItemStateChangedEvent] - Bedroom_Light_Colour changed from 82.20473,55.51181,100.0 to 82.20473,55.51181,0.39370078
2018-01-09 15:49:12.483 [vent.ItemStateChangedEvent] - Bedroom_Light_Dimmer changed from 100.0 to 0.39370078
2018-01-09 15:49:12.514 [vent.ItemStateChangedEvent] - Bedroom_Light_Switch changed from OFF to ON
```

#### Zigbee Log

[switch_conflict.log](https://github.com/openhab/org.openhab.binding.zigbee/files/1616263/switch_conflict.log)

cdjackson, 2018-01-09:
The problem is the device doesn't appear to be reporting 0, so the binding isn't setting the level to 0 -:

```
2018-01-09 15:49:12.457 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [Level Control: 2203/11 -> 0/0, cluster=0008, TID=9F, records=[ReadAttributeStatusRecord [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, status=SUCCESS, attributeValue=1]]]
```

It's reporting a value of 1, so 1/254 = 0.3% which looks correct?

BClark09, 2018-01-09:
That's right, I'm guessing this is the same as #83 then?

cdjackson, 2018-01-09:
Probably related, yes. What I found while testing #81 is that sending commands without the ```WithOnOff``` methods doesn't actually result in an off - so setting the level to 0 still leaves a Hue bulb on. This is probably then reported back and this is the issue you see.

BClark09, 2018-01-21:
I might have been wrong (or you may have fixed) when I mentioned brightness only bulbs did the same thing. These are working fine in the latest version of openHAB, it's just the colour bulbs that are misbehaving with switch items.

cdjackson, 2018-01-21:
Can you provide an updated log when you get a chance.  Thanks.

BClark09, 2018-01-26:
Finally had some free time to get some logs, but happy to say that this is no longer an issue! Thanks for solving @cdjackson !

closed 2019-02-01 by BClark09