openhab/org.openhab.binding.zigbee #83: Color converter potential OnOff / LevelControl issue

cdjackson, 2018-01-09:
As seen in #81 use of OnOffCluster to turn a device off can result in a discrepancy between OnOff and LevelControl reports.

Checking the ZCL spec, it seems that the ```ZclCluster.xxxWithOnOff()``` methods should be used to ensure synchronisation between the two clusters.

@puzzle-star do you have any thoughts on this?

puzzle-star, 2018-01-09:
Hi @cdjackson. Indeed, for the color cluster changes I did to support XY color, I already did it like this. If the device supports Level Control and Switch Control, it uses the xxxWithOnOff commands even for OnIff. Also, it should be taking care of the reports either from Switch or Level clusters. I will have a look at it, and I could apply the same to the dimmer channel, but I won't be able to till the weekend

cdjackson, 2018-01-09:
What is seen in #81 is that a device is reporting both on/off, and level. So a device can seemingly report OFF, but level=254. If these reports coming in the wrong way around, then it's a potential problem...

Maybe it's not an issue if LevelWithOnOff is always used - I've not checked this myself, but #82 seems to indicate this issue -:

```
2018-01-09 14:27:35.906 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 37 39 36 43 2C 43 30 35 45 2C 30 31 2C 31 32 2C 30 30 30 36 2C 30 37 3A 18 ED 0A 00 00 10 00 2C 2D 34 32 2C 46 46
2018-01-09 14:27:35.909 [DEBUG] [gesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=31084, profileId=49246, destinationEp=1, sourceEp=18, clusterId=6, messageData=18 ED 0A 00 00 10 00, rssi=-66, lqi=255]
2018-01-09 14:27:35.914 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=31084, profileId=49246, destinationEp=1, sourceEp=18, clusterId=6, messageData=18 ED 0A 00 00 10 00, rssi=-66, lqi=255]
2018-01-09 14:27:35.919 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=31084/18, destinationAddress=0/1, profile=C05E, cluster=6, addressMode=null, radius=0, sequence=0, payload=18 ED 0A 00 00 10 00]
2018-01-09 14:27:35.922 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=237, commandId=10]
2018-01-09 14:27:35.926 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [On/Off: 31084/18 -> 0/1, cluster=0006, TID=ED, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=false]]
2018-01-09 14:27:35.952 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 37 39 36 43 2C 43 30 35 45 2C 30 31 2C 31 32 2C 30 30 30 38 2C 30 37 3A 18 EE 0A 00 00 20 02 2C 2D 34 32 2C 46 46
2018-01-09 14:27:35.955 [DEBUG] [gesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=31084, profileId=49246, destinationEp=1, sourceEp=18, clusterId=8, messageData=18 EE 0A 00 00 20 02, rssi=-66, lqi=255]
2018-01-09 14:27:35.959 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=31084, profileId=49246, destinationEp=1, sourceEp=18, clusterId=8, messageData=18 EE 0A 00 00 20 02, rssi=-66, lqi=255]
2018-01-09 14:27:35.962 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=31084/18, destinationAddress=0/1, profile=C05E, cluster=8, addressMode=null, radius=0, sequence=0, payload=18 EE 0A 00 00 20 02]
2018-01-09 14:27:35.965 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=238, commandId=10]
2018-01-09 14:27:35.968 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Level Control: 31084/18 -> 0/1, cluster=0008, TID=EE, reports=[Attribute Report: attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, attributeValue=2]]
2018-01-09 14:27:50.535 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 42 32 37 43 2C 43 30 35 45 2C 30 31 2C 31 32 2C 30 30 30 36 2C 30 37 3A 18 D9 0A 00 00 10 00 2C 2D 35 38 2C 46 46
2018-01-09 14:27:50.539 [DEBUG] [gesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=45692, profileId=49246, destinationEp=1, sourceEp=18, clusterId=6, messageData=18 D9 0A 00 00 10 00, rssi=-88, lqi=255]
2018-01-09 14:27:50.543 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=45692, profileId=49246, destinationEp=1, sourceEp=18, clusterId=6, messageData=18 D9 0A 00 00 10 00, rssi=-88, lqi=255]
2018-01-09 14:27:50.546 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=45692/18, destinationAddress=0/1, profile=C05E, cluster=6, addressMode=null, radius=0, sequence=0, payload=18 D9 0A 00 00 10 00]
2018-01-09 14:27:50.549 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=217, commandId=10]
2018-01-09 14:27:50.551 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [On/Off: 45692/18 -> 0/1, cluster=0006, TID=D9, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=false]]
2018-01-09 14:27:50.565 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 42 32 37 43 2C 43 30 35 45 2C 30 31 2C 31 32 2C 30 30 30 38 2C 30 37 3A 18 DA 0A 00 00 20 FE 2C 2D 35 38 2C 46 46
2018-01-09 14:27:50.568 [DEBUG] [gesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=45692, profileId=49246, destinationEp=1, sourceEp=18, clusterId=8, messageData=18 DA 0A 00 00 20 FE, rssi=-88, lqi=255]
[... cut ...] andId=10]
2018-01-09 14:27:50.582 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Level Control: 45692/18 -> 0/1, cluster=0008, TID=DA, reports=[Attribute Report: attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=0, attributeValue=254]]
```

I believe this is from a Hue bulb, and it's showing OnOff=false, but Level=254 and I think the color converter will show the device is on, when it's probably off.

I'm wondering if OnOff cluster is needed at all?

puzzle-star, 2018-01-09:
xxxWithOnOff should solve this (I do not have any Hue bulb to test anyway)

But OnOff cluster is still needed because if the light is turned on/off via a remote controller it will (at least both for the IKEA lights and flspp led controllers I tested) report an on off change, not a level change.

BTW, I just saw in the updateOnOff method I did in the color channel that the brightness should be updated conditionally (zero vs last brightness). This responds to a strange behaviour in the reported value I was seeing on my lights when turning off (not on) using the remote controller

cdjackson, 2018-01-10:
> xxxWithOnOff should solve this (I do not have any Hue bulb to test anyway)

I don't understand your reasoning here. If the device sends an OnOff=Off followed by Level=254, then the state will be On even when it's off. This is what we are seeing already and I think this will also occur in the color converter won't it? Or is there something to protect against this?

Also, in the ```handleRefresh``` method, we are requesting both the LevelControl and OnOff clusters, so again if the device is reporting OFF with Level=254, then we will show on if they appear in that order.

I'm not really sure why devices are reporting in this way, but it doesn't seem uncommon based on the logs provided in the open issues, and I don't think the binding will handle these cases correctly. Even my latest change with the LevelControl converter which now only uses the LevelControl cluster which will report incorrectly if the device is going to report a level of 254 when off :(

puzzle-star, 2018-01-10:
In my understanding, using moveToLevelWithOnOff commands, should make the light report 0 on level, but I do not have a Hue light to test. Ikea lights and FLSPP controllers (the only dimmables I have), seem to behave correctly here.

Anyway, if it does not work with Hue this should be solvable by adding the OnOff cluster to the level cluster, and not passing level updates to the channel (even if we may remember the value for later) when the light is off (update as zero regardless of the received value)

Also we should take into acount that even if onoff is not mandatory for level control in Zigbee, in the model of ESH, dimmers should support switching, and color should support dimming, so channel consolidation makes total sense. But then we need to check if we have onoff in the level converter and use withOnOff commands if it is the case to keep also consistency within Zigbee, which complicates a bit the converter, but should work.

This is what I did (of course, first version migrh have some bugs) for the color channel changes.

I can change this in the Level converter to implement it the same way (and review a couple of things in the color converter), but I will not  be able to test Hue lights. Not possible for me till this weekend, anyway.

cdjackson, 2018-01-10:
> In my understanding, using moveToLevelWithOnOff commands, should make the light report 0 on level, 

Agreed, but it seems that there are cases when this isn't the case - as per the above.  Also, as said, we can't be sure that someone won't use a different command using some sort of remote control that might cause the reporting to be done different.

> Also we should take into acount...

Agreed. This is why I initially added this to the level converter - I've currently removed it, but as I said above this has other problems so I will rethink this.

It's a little messy I know :(

I think that we need to do something along the same lines as you did in the color converter - add a timer so that if there is an update to either level or onoff, that we wait (say) 100ms and if there's no update use the report. If both reports are received in this period, then if onoff=off, show off, if onoff=on, show level.

What do you think?

puzzle-star, 2018-01-10:
For color control, this delay before processing attribute reports is needed for consistency, as we may receive Hue and Sat at different moments (not too bad) or separately X Y attributes, which would definitely provide an inconsistent color update to the channel.

For level control, it may be simpler, and anyway I am not sure this delay would solve the issue.

For lights that report onoff status change I would just not push a level attribute report to the channel when we know the light is off, and for those that don't report (some hue lights?), when polling we should poll level always just after onoff, and only if on.

Would this make sense? Or do we already know of some lights that WON'T report onoff changes but WILL report level change in some cases?

cdjackson, 2018-01-10:
> For level control, it may be simpler, and anyway I am not sure this delay would solve the issue.

Th delay could be short, but the issue is that as we don't know if a report might be any of the following (onoff / level / onoff then level / level then onoff) there needs to be some way to cope with this and I think the "only" option is a short timer. Delay can be short, but I think something needs to keep track of this without sending "spurious" state updates into the system.

> For lights that report onoff status change I would just not push a level attribute report to the channel when we know the light is off

This is effectively what I also suggested above - don't use level if the device is off, so this is fine. I was trying to think if we can avoid the timer, but I don't think so. Eg, we could have the situation where the device is off, and we receive an update because someone resent the off (maybe from a remote control, or maybe from the binding). If this causes a level=254 followed by an onoff=off, then we have a transient ON/OFF being sent on the event bus and for sure, users will complain.

> Or do we already know of some lights that WON'T report onoff changes but WILL report level change in some cases?

I don't think we can be 100% sure of the reporting configuration.... For example, my Hue light only has a reporting table with 1 entry, so we aren't able to be sure of reporting configuration - or at least we can't have the reporting configuration set for all clusters, and we could potentially end up with different reporting in different installations as currently the reporting config is done in the converter, and not in the thing level where we might be able to handle this sort of thing. I think as soon as we make assumptions like this, a bulb will turn up that breaks the assumptions, or someone will use a controller that causes some other report to be sent.

cdjackson, 2018-01-25:
#82

weakfl, 2018-07-17:
@cdjackson since my dimmers are only changing the on/off state (but not the level) when the **wall-mounted switch** is used to turn them off, I'm wondering how reporting is configured for devices which have both, a level and an on/off channel.
Do you even configure the on/off channel to report state changes if channels are consolidated?

If reporting is still active for the on/off channel it should be possible to setup a rule that updates the level accordingly (as a temporary workaround for this issue).

cdjackson, 2018-07-17:
> since my dimmers are only changing the on/off state (but not the level)

I'm not sure I understand the problem - there should only be one channel (a dimmer channel) and in the event that it is set to off, the dimmer should go to 0%. It sounds like you have multiple channels which would be strange?

> Do you even configure the on/off channel to report state changes if channels are consolidated?

Yes, reporting is configured for all clusters used in the color converter (so, color, level and on/off).  All reports are used to update the state.

Do you have a log of the reports being received in this condition.

weakfl, 2018-07-17:
> I'm not sure I understand the problem - there should only be one channel (a dimmer channel) and in the event that it is set to off, the dimmer should go to 0%.

As I said, if I use the **wall-mounted control element** and **NOT** openhab to turn the dimmable relay off.

It actually makes sense. If I turn the dimmable relay back on through the **wall-mounted control element** the last level value is used, so the level is persistent.

> It sounds like you have multiple channels which would be strange?

I do have multiple channels and I think it does make sense as explained above:

[image]

**NOTE**: The 3rd channel is for the second rocker on the wall-mounted control-element, the first two are for the dimmable relay.

> Do you have a log of the reports being received in this condition.

Not atm, but I can of course provide one if you want?

puzzle-star, 2018-07-17:
Hi @cdjackson,

"Yes, reporting is configured for all clusters used in the color converter (so, color, level and on/off). All reports are used to update the state."

I recall putting this in the color converter, but was this also implemented in the switchlevel converter?

I will have a look at the code and, if OK for you, I could add the same to the switchlevel converter.

cdjackson, 2018-07-17:
> I recall putting this in the color converter, but was this also implemented in the switchlevel converter?

How is this relevant? Reporting is used in the switch converter for sure, but if the Color converter is used, then other converters are not. Ie the channel consolidation eliminates channels that are already available through other channels. ie if color is available, then it doesn't show the level control, and if level control is available, it doesn't show the switch - you will only get the color converter.

> I will have a look at the code and, if OK for you, I could add the same to the switchlevel converter.

I'm confused - sorry. What do you want to add to the switchlevel converter? This issue is about color control (sorry, but it's a little confusing - please be clear).

cdjackson, 2018-07-17:
> As I said, if I use the wall-mounted control element and NOT openhab to turn the dimmable relay off.

I'm not sure how that matters? If you change the control on the device, it should report back the level. If I remember correctly, the standard states that the level control, and on/off should be linked and consistent. If they aren't consistent, then we get into another situation where the order of the reports becomes important - ie if it sends the level report immediately after the onoff report, then it still will report the level control value.

> I do have multiple channels and I think it does make sense as explained above:

No - that doesn't make sense. There should only be one channel - a dimmer if this is a level control, or a color control if this is related to color control.

If this issue is not related to the color control, please clarify

puzzle-star, 2018-07-17:
Hi @cdjackson,

Some lights (i.e. ikea) do not report level change when switched on off. Probably they would if they receive the "moveToLevelWithOnOff" command , but they do not when they get the OnOff command. They do report onoff then, but not level.

The result is that when these lights arr sent on/off commands from a remote or not directly from openhab, they will report the status change on onoff cluster, but not to level cluster.

The switchlevel converter will filter out the reported attribute: the code checks and only takes care of the level cluster updates (just checked). So the channel is not updated when this happens, even if the light is sending the report (on onoff cluster). I do not know if this is 100% compliant: I could not find a right answer on the Zigbee spec, but for sure we have some lights acting like this...

Hope this clarifies.

I see the original mail speaks of color channel, but the examples he gives are about switch level.

When I brought XY color to the color channel I recall we discussed on this, and a fix for this was added. But it was never for the switch level channel, so I was offering to do it there also.

Hope it clarifies also this :)

weakfl, 2018-07-17:
> I'm not sure how that matters? If you change the control on the device, it should report back the level. If I remember correctly, the standard states that the level control, and on/off should be linked and consistent.

If that's the case the BJ devices do not conform to the standard. Sounds strange though, imho it makes perfect sense to have a persistent dimmer level and still be able to change the on/off state.

> No - that doesn't make sense. There should only be one channel - a dimmer if this is a level control, or a color control if this is related to color control.

Well, let me explain how the control element works:
- Short press left side -> relay Off
- Short press right side -> relay On
- Long press left/ride side -> adjust dimmer level

> If this issue is not related to the color control, please clarify

Well, not really related to color control as @puzzle-star explained. **Sorry for the confusion**, but from what I read I thought this issues are related 😟

cdjackson, 2018-07-20:
Ok, that's fine, but I've said above, the problem I have is that if the device sends different things for both clusters, which one do we use? 

If it sends "OFF", then "level 10" - is it off, or on? 

If it sends "level 10" then "OFF" - is this the same? 

If they are sent right next to each other, then maybe (just maybe!) we can make an assumption, but if we receive an OFF, then 1 second later receive "level 10" - is it on or off? This is quite likely to happen for a number of reasons. For starters, if we use reporting on the level control cluster (as we need to), then it will periodically report level=10 even if it's off (right?). How do we resolve this - we are in the position where two conflicting reports can be received at any time to indicate different states of the same channel.

I'm not really sure how best to handle these two issues - I've not found anything that clearly distinguishes how to handle these. I'm open to suggestions of course.

weakfl, 2018-07-20:
Some thoughts...

> the problem I have is that if the device sends different things for both clusters, which one do we use?

**Imho**: if the On/Off cluster is OFF, all level reports should be ignored and not propagated to OH. The order doesn't matter, because:

_"When the level is reduced to its minimum the OnOff attribute is automatically turned to Off, and when the level is increased above its minimum the OnOff attribute is automatically turned to On."_

This is well defined, the level is irrelevant if the device is OFF. If the level is changed, the On/Off state should change accordingly.

> If they are sent right next to each other, then maybe (just maybe!) we can make an assumption, but if we receive an OFF, then 1 second later receive "level 10" - is it on or off?

How does this matter? Off is off, On is on. If On is reported, we have two cases:

1. _OnLevel_ is defined, level will be set accordingly, consecutive reports are fine and should be propagated (_"The attribute OnLevel (see 3.10.2.3.4) determines whether commands of the On/Off cluster have a permanent effect on the CurrentLevel attribute or not."_).
2. ON received, _CurrentLevel_ is what it is, consecutive level reports are fine and should be propagated.

**BUT**: of course there can be devices like a volume control, where the On/Off state represents unmuted/muted and where level (volume) reports should probably be propagated to OH, even if the device is muted (OFF).
Tbh, I have no idea how to identify these kind of devices. I haven't found anything in the specs telling us how to distinguish the different use cases. I didn't have much time to read the docs, maybe there is a way. Should be an edge case anyway...

As I said, just my two cents after taking a brief look at the docs and probably nothing you don't know already...

cdjackson, 2018-08-02:
This sounds fine - I've mostly implemented this and will get a PR in in the next day or so. My only concern is if there is any problem with getting the different reports (ie if a device only configures reporting for level, or on/off - not both) then it might get confused. Some devices (Hue bulbs!) have very limited reporting table entries (some Hue bulbs only have 1 reporting table entry, so it would not be possible to configure the two clusters - I know this doesn't matter to Hue bulbs, but maybe there will be problems with some devices).

Let's see...

weakfl, 2018-08-03:
> This sounds fine - I've mostly implemented this and will get a PR in in the next day or so.

Very nice, thanks.

> My only concern is if there is any problem with getting the different reports (ie if a device only configures reporting for level, or on/off - not both) then it might get confused.

Good thought, that might indeed be a problem. I've seen you've already pushed a PR, looking forward to test this when it becomes available.

I'm also wondering if all devices will indeed change the on/off state when the level is lowered to 0 or vice versa (as they should).
Maybe the controller should send *LevelWithOnOff* commands if a device has both clusters? On the other hand this might be wrong for devices that don't change the on/off state on purpose.

cdjackson, 2018-08-04:
This same issue will also need to be fixed in the color converter. I also see a bug in the color converter where it is completely ignoring the onoff state.

cdjackson, 2018-08-11:
Thanks for testing @weakfl - I assume that you pulled #231 and compiled your own binding to test? If so, I'll merge this thanks.

> One question: will I be able to change the reporting interval once static device definitions become available?

I don't think it's linked (or did I say in the past it was ;) )? I suggest we discuss the need for this on a separate topic so it doesn't get lost (or again, I suspect we discussed it in the frame of your switches - if so, let's keep the conversation in one place and restart there, or we'll repeat everything).

Basic answer though is no - it won't immediately support the ability to change reporting periods. The static things will provide the ability to tune a thing definition, but it doesn't really add extra features like changing the reporting period.

Cheers
Chris

weakfl, 2018-08-11:
>  I assume that you pulled #231 and compiled your own binding to test? If so, I'll merge this thanks.

No I didn't, sorry for the confusion. I was referring to the on/off/level issue which has been fixed with #229.

> I don't think it's linked (or did I say in the past it was ;) )? I suggest we discuss the need for this on a separate topic so it doesn't get lost (or again, I suspect we discussed it in the frame [...]

No you didn't, I just wasn't sure what can be configured in the static definitions. But as you said, lets discuss this in the appropriate thread :)

cdjackson, 2020-01-03:
Closing this as there doesn't seem to be any issue now.

closed 2020-01-03 by cdjackson