openhab/org.openhab.binding.zigbee #93: Request timeouts

weakfl, 2018-01-15:
Chris, in #57 you said:
> If you start to see timeouts in the log, please let me know - I'm not 100% sure that 500ms will cover every transaction, but I think it should ;)

I do get some timeouts:
```
09:52:36.987 [DEBUG] [iscovery.ZigBeeNodePropertyDiscoverer] - D85DEF11A1002A12: Manufacturer request timeout
09:52:47.050 [DEBUG] [iscovery.ZigBeeNodePropertyDiscoverer] - D85DEF11A1002A12: Model request timeout
09:52:57.119 [DEBUG] [iscovery.ZigBeeNodePropertyDiscoverer] - D85DEF11A1002A12: Hardware version request timeout
```

The device is a BJ battery powered switch, so you should be able to reproduce it with your switch. Maybe the device just doesn't respond to the requests, but I figured I'd report it and let you decide if it's relevant ;)

cdjackson, 2018-01-15:
> I do get some timeouts:

These aren't the timeouts I meant in the other thread ;) The timeouts I was referring to are the low level transactions with the Telegesis stick - every time the binding sends a message, there is a low level exchange between the driver and the stick which results in a couple of messages sent at that level - this is the timeout I referred to.

The timeouts here are high level - they have a 10 second timeout to cope with battery devices which don't respond for 8 seconds. This means that the device is not responding to the request - if you have a more detailed log it would be interesting to take a look. 

I guess this is happening during startup and I think I will try and improve the logic here so it doesn't do some of these requests if it already has the data as startup is probably swamping the network and that might be at least part of the reason...

weakfl, 2018-01-15:
> I guess this is happening during startup and I think I will try and improve the logic here so it doesn't do some of these requests if it already has the data as startup is probably swamping the network and that might be at least part of the reason...

It happened during a scan for new devices. So I'm afraid there isn't much more in the log, but here you go: 
[zigbee_issue93_20180115095236.log.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/1631174/zigbee_issue93_20180115095236.log.zip)

I was trying to add a Tradfri bulb (ieeeAddress `000B57FFFEB1BAB5`), which didn't work btw. But that's probably a topic for another issue :)

cdjackson, 2018-01-15:
> It happened during a scan for new devices. 

Was this a device that is already part of the network?

> I was trying to add a Tradfri bulb (ieeeAddress 000B57FFFEB1BAB5), which didn't work btw

Did it join at all (I guess so or you probably wouldn't have the address).  What type of bulb - I have a white bulb arriving today for testing - color bulbs should work ok already as I know people are using them, but feel free to open another issue...

weakfl, 2018-01-15:
> Was this a device that is already part of the network?

No, I've never been able to add it.

> Did it join at all (I guess so or you probably wouldn't have the address).

It never turned up in the Inbox. I do have the address because it turns up in the log and doesn't match any of my other devices.

There isn't much in the log though:

```
09:52:36.505 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisDeviceJoinedNetworkEvent [networkAddress=12129, ieeeAddress=000B57FFFEB1BAB5, parentAddress=45692]
09:52:36.510 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisDeviceJoinedNetworkEvent [networkAddress=12129, ieeeAddress=000B57FFFEB1BAB5, parentAddress=45692]
09:52:36.516 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisDeviceJoinedNetworkEvent [networkAddress=12129, ieeeAddress=000B57FFFEB1BAB5, parentAddress=45692]

...

09:52:58.525 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementLqiResponse [45692/0 -> 0/0, cluster=8031, TID=NULL, status=SUCCESS, neighborTableEntries=4, startIndex=2, neighborTableList=[NeighborTable [extendedPanId=FC32931ECD920928, extendedAddress=D85DEF11A1002A12, networkAddress=51223, deviceType=END_DEVICE, rxOnWhenIdle=RX_OFF, relationship=CHILD, permitJoining=DISABLED, depth=2, lqi=252], NeighborTable [extendedPanId=FC32931ECD920928, extendedAddress=000B57FFFEB1BAB5, networkAddress=12129, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=CHILD, permitJoining=DISABLED, depth=2, lqi=253]]]
```

> What type of bulb - I have a white bulb arriving today for testing

It is a white bulb, E27 980 lm dimmable

weakfl, 2018-01-15:
Since I've managed to derail this issue once again and the original issue actually isn't one, I've at least changed the title to reflect the new topic 😁

cdjackson, 2018-01-15:
> > Was this a device that is already part of the network?
>
> No, I've never been able to add it.

To be clear, with this question I was referring to the timeouts you first reported ->

> D85DEF11A1002A12: Manufacturer request timeout

I think this is not the Tradfri bulb - you said BJ...

So my question is that this BJ switch is already included, then you press the include button again and you get these timeouts?

> It is a white bulb, E27 980 lm dimmable

Ok, I think this is the same one I have arriving today so maybe I will try this tonight.

weakfl, 2018-01-15:
> So my question is that this BJ switch is already included, then you press the include button again and you get these timeouts?

Exactly, switch is already included and I get the timeouts when starting a scan for new devices.

cdjackson, 2018-01-15:
Ok, thanks. Let's leave this issue with the 980LM bulb. I will also look at the binding startup and discovery over the coming days to try and reduce the load which might solve this timeout...

cdjackson, 2018-01-15:
For me the bulb works 'ok' - it included ok, and is controllable, but does show a color control as well as CT. There's a couple of things to look at, but it does seem to work here.....

cdjackson, 2018-01-15:
For my info, the reason it shows a color control as well is because it indicates that it supports XY color control, which is probably true, but only for CT setting. Will likely need to work this out at some point, but it's not a major problem at the moment.

weakfl, 2018-01-16:
> For me the bulb works 'ok' - it included ok, and is controllable

That's interesting. My bulb never completes discovery:

```
12129: Node discovery request IEEE_ADDRESS failed. Wait before retry.
```

I did reset it before starting discovery and that should have worked (bulb flashes when turned on to indicate a reset). Maybe there's something wrong with the bulb.

If you don't have any additional advice I think we can close here as it's obviously not a problem with the binding.

cdjackson, 2018-01-16:
Have you tried the latest version? It would be interesting to try it as it should reduce some network "noise" during discovery - maybe this helps.
The IeeeAddress request is a pretty standard request so I'm not sure why it would systematically fail. I was also testing with Ember coordinator rather than Telegesis, but again that shouldn't matter for this request - it works with other devices with the Telegesis dongle so I don't see why it would be a problem with this.

Please try the latest version. Also if you can provide a log I'll take a look just to make sure that I don't think there's a binding problem.

weakfl, 2018-01-16:
Thanks for looking into it Chris. I have tried the latest version, but it doesn't make a difference.

Here's the log: [zigbee_issue93_20180116142731.log.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/1635499/zigbee_issue93_20180116142731.log.zip)

cdjackson, 2018-01-16:
The device is joining via another router (one of your BJ switches). Can I suggest as an experiment to try and move it closer to the coordinator.

Another thing to check (although I don't think this is the issue) - in the coordinator settings there is an option called trust centre join mode - make sure this isn't set to TC Deny. It should default to Secure, but you could also try insecure. This setting sets what the trust centre does when a device joins - so a router might let it join, but if the trust centre doesn't allow it then it doesn't get the key.

However, I also see evidence that the IeeeAddress command is being received. In the logs, if I decode the data coming from the dongle, I see SEQ:90 followed a bit later by ACK:90 (this is all in hex so you won't see it like this in the log). The SEQ90 is the frame number for this frame from the stick, and the ACK90 means the remote device acked the packet!  It "just" doesn't respond!

So, I think two main possibilities - your bulb doesn't like the IeeeAddress request (unlikely, but...), or there is some issue with the routing.

I'm thinking routing - I just noticed that the routing table for this device doesn't have the bulb in it (9217) -:
```
14:28:36.314 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementRoutingResponse [0/0 -> 0/0, cluster=8032, TID=NULL, status=SUCCESS, routingTableEntries=64, startIndex=0, routingTableList=[RoutingTable [destinationAddress=45692, status=ACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=45692], RoutingTable [destinationAddress=51223, status=ACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=45692], RoutingTable [destinationAddress=31084, status=ACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=31084], RoutingTable [destinationAddress=0, status=DISCOVERY_UNDERWAY, memoryConstrained=true, manyToOne=true, routeRecordRequired=true, nextHopAddress=45692], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0]]]
```

So, I'd suggest trying to move it close to the coordinator and let's see what that does. I don't think there's any way to influence the routing with the Telegesis (at least not on a per command basis) but it would be good to know if it can work at all.

weakfl, 2018-01-17:
> The device is joining via another router (one of your BJ switches).

That was my impression looking at the log, but I thought I must be wrong. My understanding was that a relay would become a repeater if it is not controlling the network.

> but you could also try insecure

I did, but it doesn't work either. I've noticed a typo in PaperUI, the setting is labeled _"All all joins"_ instead of _"**Allow** all joins"_.

> So, I'd suggest trying to move it close to the coordinator and let's see what that does.

I'll have to find a lamp with an E27 socket to do that, all my lights with E27 are wall mounted. Alternatively I could try to remove current from the relays and see if that changes anything.

I'll report back ;)

cdjackson, 2018-01-17:
> My understanding was that a relay would become a repeater if it is not controlling the network.

Yes - this is what is happening. The relay is a router. This should work fine, but it does mean an indirect connection to the trust centre and coordinator so there is more going on. I don't think this is the issue, but...

cdjackson, 2018-01-19:
Last night I did a test where I put a hue bulb half way down the house, then the Tradfri bulb at the far end of the house, and the Tradfri still works ok here. The coordinator can still directly see the Tradfri, but it's very weak and the routing table shows it's routed via another device, so I don't think this is an issue.

weakfl, 2018-01-19:
Thanks for testing Chris. I'm pretty sure commands to my second relay are routed through relay number one and that's working ok too.

Is it possible that routing is only a problem when scanning for new devices? If I remember correctly I wasn't able to add the Tradfri even before I had the relays installed though, so I think something else is going on.

I still have to find an E27 lamp to do some proper testing. I'll also get a hue bulb for comparison.

cdjackson, 2018-01-19:
> Is it possible that routing is only a problem when scanning for new devices?

In theory, no. The way ZigBee works is a device should join the router with the best signal strength. The network as a whole is put into join mode, and the device then does a scan to find all the routers it can hear, and it will join the network via the "best" one.  I joined the Tradfri bulb with it at the far end of the house - according to the routing tables there is no direct route between them (although the coordinator can still be heard by the Tradfri, but not the other way).

Anyway, it will be interesting to check if it works when close to the coordinator - I don't think it will, but let's see. I could potentially remove the failing check from the discovery as it's arguably not needed, but that said, it is a mandatory command that should be supported, so this shouldn't be needed.

weakfl, 2018-01-21:
I've have now been able to move the bulb closer to the coordinator and there's no hop between the bulb and the coordinator when scanning for new devices (parentAddress is 0).

I've tried secure and insecure joining, both without success. Insecure seems a bit chattier, but i think that's because the bulb did reset shortly after I turned it on (networkAddress did change).

Here's the relevant part of the log:
```
16:26:14.650 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisDeviceJoinedNetworkEvent [networkAddress=1519, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:14.655 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisDeviceJoinedNetworkEvent [networkAddress=1519, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:14.661 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisDeviceJoinedNetworkEvent [networkAddress=1519, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:22.408 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 4E 45 57 4E 4F 44 45 3A 34 33 43 37 2C 30 30 30 42 35 37 46 46 46 45 42 31 42 41 42 35 2C 30 30 30 30
16:26:22.414 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisDeviceJoinedNetworkEvent [networkAddress=17351, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:22.420 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisDeviceJoinedNetworkEvent [networkAddress=17351, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:22.426 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisDeviceJoinedNetworkEvent [networkAddress=17351, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:30.103 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 4E 45 57 4E 4F 44 45 3A 34 44 38 36 2C 30 30 30 42 35 37 46 46 46 45 42 31 42 41 42 35 2C 30 30 30 30
16:26:30.109 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisDeviceJoinedNetworkEvent [networkAddress=19846, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:30.114 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisDeviceJoinedNetworkEvent [networkAddress=19846, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:30.120 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisDeviceJoinedNetworkEvent [networkAddress=19846, ieeeAddress=000B57FFFEB1BAB5, parentAddress=0]
16:26:44.713 [INFO ] [smarthome.event.ItemStateChangedEvent] - TerraceCurrentTemperature changed from 5.300000000000001 to 5.2
16:26:55.264 [DEBUG] [stems.zigbee.ZigBeeNetworkMeshMonitor] - 0: Starting mesh update
16:26:55.272 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementLqiRequest [0/0 -> 0/0, cluster=0031, TID=DC, startIndex=0]
16:26:55.279 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0000, cluster=49, addressMode=DEVICE, radius=31, sequence=220, payload=00 00]
16:26:55.285 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis send: TelegesisSendUnicastCommand [address=0, sourceEp=0, destEp=0, profileId=0, clusterId=49, messageData=00 00, messageId=null]
16:26:55.291 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
16:26:55.298 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=0, sourceEp=0, destEp=0, profileId=0, clusterId=49, messageData=00 00, messageId=null]
16:26:55.308 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS TX: Data 41 54 2B 53 45 4E 44 55 43 41 53 54 42 3A 30 32 2C 30 30 30 30 2C 30 30 2C 30 30 2C 30 30 30 30 2C 30 30 33 31 0D 00 00 0D 0A
16:26:55.507 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 53 45 51 3A 38 32
16:26:55.514 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 4F 4B
16:26:55.521 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis command complete: TelegesisSendUnicastCommand [address=0, sourceEp=0, destEp=0, profileId=0, clusterId=49, messageData=00 00, messageId=130, status=SUCCESS]
16:26:55.540 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 52 58 3A 30 30 30 30 2C 30 30 30 30 2C 30 30 2C 30 30 2C 38 30 33 31 2C 34 37 3A 00 00 03 00 03 28 09 92 CD 1E 93 32 FC B5 BA B1 FE FF 57 0B 00 86 4D 35 02 0F FF 28 09 92 CD 1E 93 32 FC 73 19 00 A1 11 EF 5D D8 6C 79 35 02 0F FF 28 09 92 CD 1E 93 32 FC 26 28 00 A1 11 EF 5D D8 7C B2 35 02 0F FF 2C 30 30 2C 46 46
16:26:55.557 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis event received: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=0, profileId=0, destinationEp=0, sourceEp=0, clusterId=32817, messageData=00 00 03 00 03 28 09 92 CD 1E 93 32 FC B5 BA B1 FE FF 57 0B 00 86 4D 35 02 0F FF 28 09 92 CD 1E 93 32 FC 73 19 00 A1 11 EF 5D D8 6C 79 35 02 0F FF 28 09 92 CD 1E 93 32 FC 26 28 00 A1 11 EF 5D D8 7C B2 35 02 0F FF, rssi=0, lqi=255]
16:26:55.565 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=0, profileId=0, destinationEp=0, sourceEp=0, clusterId=32817, messageData=00 00 03 00 03 28 09 92 CD 1E 93 32 FC B5 BA B1 FE FF 57 0B 00 86 4D 35 02 0F FF 28 09 92 CD 1E 93 32 FC 73 19 00 A1 11 EF 5D D8 6C 79 35 02 0F FF 28 09 92 CD 1E 93 32 FC 26 28 00 A1 11 EF 5D D8 7C B2 35 02 0F FF, rssi=0, lqi=255]
[... cut ...] ighborTable [extendedPanId=FC32931ECD920928, extendedAddress=D85DEF11A1001973, networkAddress=31084, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=15, lqi=255], NeighborTable [extendedPanId=FC32931ECD920928, extendedAddress=D85DEF11A1002826, networkAddress=45692, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=15, lqi=255]]]
```

I should receive a hue bulb on Tuesday, might be interesting to see if that works.

cdjackson, 2018-01-21:
This bit of the log probably isn’t the discovery handler - the IeeeAddressRequest that was timing out should probably come later. Can you provide a longer log?

Thanks.

weakfl, 2018-01-21:
Sure, [zigbee_issue93_20180121162614.log.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/1649883/zigbee_issue93_20180121162614.log.zip)

Let me know if it contains what you're looking for, otherwise I'll capture a new log.

cdjackson, 2018-01-21:
Interesting - there's a message from the coordinator I've not handled and that will help, but I'm not sure it will necessarily solve the issue since the original log the discovery was starting, but the device wasn't responding.

Anyway, it should be implemented so I will add this.

cdjackson, 2018-01-21:
See https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/189

weakfl, 2018-01-25:
Working now, see [details here](https://github.com/openhab/org.openhab.binding.zigbee/issues/49#issuecomment-360390742).

closed 2018-01-25 by weakfl

cdjackson, 2018-01-28:
> For me the bulb works 'ok' - it included ok, and is controllable, but does show a color control as well as CT.

This should now be resolved by #121

weakfl, 2018-01-29:
Thanks Chris, I can confirm mine has now a level and a CT channel instead of color and CT.