openhab/org.openhab.binding.zigbee #100: HUE motion sensor: request LED indicator configuration

curlyel, 2018-01-21:
When paired to a HUE bridge, the LED on the sensor is configured to just blink in case of battery is low or empty (orange==low, red==almost empty, steady red==no connection to coordinator).

When connected to the Zigbee binding, the LED flashes every time when motion is detected.
This is somehow distracting/annoying.

It seems, that Philips has some attribute in the basic cluster
`0x0000/0x0033* config.ledindication`
which likely is intended to set the “LED mode”.

```
From your sniffer, you should see that the Hue bridge sets up bindings to the bridge for the following clusters on endpoint 02:

- 0x0000 (Basic)

- 0x0001 (Power)

- 0x0400 (Illuminance Measurement)

- 0x0402 (Temperature Measurement)

- 0x0406 (Occupancy Sensing)

Then, it sets up attribute reporting for the following attributes:

- 0x0000/0x0032* config.usertest

- 0x0000/0x0033* config.ledindication

- 0x0001/0x0021 config.battery

- 0x0400/0x0000 state.lightlevel

- 0x0402/0x0000 state.temperature

- 0x0406/0x0000 state.presence

- 0x0406/0x0030* config.sensitivity

*) these are manufacturer specific attributes.`
```
So would it be possible to make the LED configurable, or switch it off completely, or initialize it just similar to Philips (just blink if battery is weak)?

See: https://developers.meethue.com/content/philips-hue-motion-sensor-and-zigbee-attributes
See also: https://community.openhab.org/t/zigbee-binding/15763/735

cdjackson, 2018-01-25:
Can be looked at after #101

cdjackson, 2018-03-11:
@curlyel can you provide information on exactly what sensor you have. The motion sensor I have here does not support these features.

curlyel, 2018-03-11:
@cdjackson this is the type of sensors I have:
[https://www2.meethue.com/de-de/p/hue-bewegungsmelder/8718696743171](url)
On the back of the sensors is a model number printed: 9290012607

The mentioned trace in the initial post was done from someone at developers.meethue.com:
https://developers.meethue.com/content/philips-hue-motion-sensor-and-zigbee-attributes
Seems, that this guy has observed/sniffed the initialization of the sensor when connected to a Hue bridge. 

I cannot check myself, if my Hue motion sensors have the same attributes. But what I can say is, if my sensors are paired to the Hue bridge, then **the LED is steady off during normal operation**.

According to the manual, the LED will flash/lit in orange or red to indicate battery low or loss of connection to the bridge.

The Hue bridge also has the option to adjust the sensitivity of the sensors. Which is also shown in the captured packets. But the user there said:

> sets up attribute reporting for the following attributes

Not sure, if this is the way how Philips is controlling the LED/sensitivity level or if there is another way to achieve this which is not yet revealed...

cdjackson, 2018-03-11:
So this is the same model I have here and the stated attributes are not supported by my device so I don't think there's much we can do. I can look in other clusters to see if there's something else available, but it's a bit hit and miss...

cdjackson, 2018-03-23:
Closing this as device doesn't support these features.

closed 2018-03-23 by cdjackson