openhab/org.openhab.binding.zigbee #108: Dimmer items not displaying correct values

openhab-5iver, 2018-01-21:
OH 2.3.0 build 1191
Zigbee 2.3.0.201801200902 

I have rules that set the lights to a specific level, and don't adjust them if they are not at that specific level. This way, I can manually turn the light on (which sets it to 100), and the light won't turn off when motion stops or be adjusted by lux changes. I noticed this wasn't working for my Zigbee bulbs, and that when I send 99 through the rule, the lights are reporting 98 in the logs and UI. It looks like there may be a rounding issue in ZigBeeConverterSwitchLevel.java.

Here is a light being set to 99 through a rule, which is sent to the device, but then the bulb reports back 98...
```
2018-01-21 17:14:59.361 [DEBUG] [.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 7CE524000011F4D1: Command for channel zigbee:device:9b51c72d:7ce524000011f4d1:7CE524000011F4D1_1_switch_level --> 99
2018-01-21 17:14:59.788 [DEBUG] [inding.zigbee.converter.ZigBeeConverterSwitchLevel] - 7CE524000011F4D1: ZigBee attribute reports ZclAttribute [cluster=LEVEL_CONTROL, id=0, name=CurrentLevel, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=1]
2018-01-21 17:14:59.789 [DEBUG] [inding.zigbee.converter.ZigBeeConverterSwitchLevel] - 7CE524000011F4D1: ZigBee attribute reports ZclAttribute [cluster=LEVEL_CONTROL, id=0, name=CurrentLevel, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=58]
2018-01-21 17:14:59.790 [DEBUG] [inding.zigbee.converter.ZigBeeBaseChannelConverter] - 7CE524000011F4D1: Channel zigbee:device:9b51c72d:7ce524000011f4d1:7CE524000011F4D1_1_switch_level updated to 22
2018-01-21 17:14:59.789 [DEBUG] [inding.zigbee.converter.ZigBeeBaseChannelConverter] - 7CE524000011F4D1: Channel zigbee:device:9b51c72d:7ce524000011f4d1:7CE524000011F4D1_1_switch_level updated to 22
2018-01-21 17:15:00.756 [DEBUG] [inding.zigbee.converter.ZigBeeConverterSwitchLevel] - 7CE524000011F4D1: ZigBee attribute reports ZclAttribute [cluster=LEVEL_CONTROL, id=0, name=CurrentLevel, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=251]
2018-01-21 17:15:00.757 [DEBUG] [inding.zigbee.converter.ZigBeeBaseChannelConverter] - 7CE524000011F4D1: Channel zigbee:device:9b51c72d:7ce524000011f4d1:7CE524000011F4D1_1_switch_level updated to 98
```

Since lastValue=251 (98.8), it looks like this...
https://github.com/openhab/org.openhab.binding.zigbee/blob/25dfb3835c2b836a9845e2f04831c474cb6b647b/src/main/java/org/openhab/binding/zigbee/converter/ZigBeeConverterSwitchLevel.java#L128

... is truncating to an integer when converting to PercentType or because of integer division. Either way, adding 0.5 to the value (like you have in the outbound command at line 100) should fix this? In looking at the scripts that I had used with a Wink hub, this is basically how I handled refreshing values from the hub.

closed 2018-01-25 by cdjackson

cdjackson, 2018-01-25:
Let me know if this is working now...

openhab-5iver, 2018-01-26:
This looks to be working... thank you! The states jump around a bit, but they seem to finally all get there!

cdjackson, 2018-01-26:
Thanks. 

I think the jumping states is caused by the update that set. The UI sets to (say) 100% when the device is at 0%, then there's a number of updates come in as the bulb changes. So, you will see 0% 100% 20% 40% 60% 80% 100% (sort of thing). Short of changing the update rate I don't think there's a way around this at the moment - I know other bindings have the same issue and I've seen other discussions about how to solve this sort of thing.

weakfl, 2018-01-26:
One of my relays is now randomly shown as ON in OH although it is off. Is it possible the behaviour was introduced with this commit?
Strangely only one of two relays seems to be affected. But it's quite random, the state is ok for an hour or two and suddenly jumps to ON.

cdjackson, 2018-01-26:
It shouldn't be related to this but if the device is returning dimmer levels that aren't 0, then it might be the case that a value of 1 would have been interpreted as off in the past. 

In any case if the device is returning a non 0 value then it's a device issue. 

I'd suggest to open a new issue and provide a debug log and I'll take a look.

PR #117 (2018-01-25): Provide standard methods for converting level to percentage
Fixes #108