openhab/org.openhab.binding.zigbee #110: ST Motion sensors only show temperature channel

AfromD, 2018-01-23:
My ST motion sensors won’t join although they are in join mode. When I hold reset while connecting the battery, they join, but only show the temperature channel.

[image]

[New Text Document (3).txt](https://github.com/openhab/org.openhab.binding.zigbee/files/1656108/New.Text.Document.3.txt)

cdjackson, 2018-01-23:
This is not a general issue with ST motion sensors since at least mine is working fine.

I checked the logs, and the reason the channel isn't showing is because the binding checks the type of sensor by reading an attribute from the device. In the case here, the device is returning 0.

```
2018-01-23 13:34:33.967 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [IAS Zone: 5241/1 -> 0/1, cluster=0500, TID=2C, records=[ReadAttributeStatusRecord [attributeDataType=ENUMERATION_16_BIT, attributeIdentifier=1, status=SUCCESS, attributeValue=0]]]
```

Maybe there's a setting that disables this or something, but at least that explains what's happening.

AfromD, 2018-01-24:
I have added another one of these, and shows the same behaviour. So it seems to be typical for this device. I don't know about any way to change any settings on the device.

cdjackson, 2018-01-24:
As said, it does not do it with my test device which correctly reports the features of the device. Maybe there is a firmware change or something - I don't know at the moment. 

I will look at it more in the coming week or two.

AfromD, 2018-01-24:
There are different generations:
https://support.smartthings.com/hc/en-us/articles/212204223-Which-SmartThings-motion-sensor-model-do-I-have-
Mine is the "latest" model.

cdjackson, 2018-01-24:
I also have the latest model, but this doesn't mean that there aren't different software versions. It is strange that your device is reporting that it doesn't support any notifications. One thing I want to check is if your device is disabled or something - seems unlikely, but otherwise I think it should really say what type of sensor it is.

cdjackson, 2018-01-24:
Actually I just had another look at the doc. The response of 0 is actually valid - it's just not reporting as a motion sensor - it's a (so called) "system alarm". I will need to add a different handler for this alarm type.

cdjackson, 2018-01-25:
This should be fixed with #115

cdjackson, 2018-01-25:
Please advise if this can be closed with the latest build.

AfromD, 2018-01-26:
Didn't solve it (Version  2.3.0.201801251946)
I'll troubleshoot and post a log on Monday, busy weekend ahead.

cdjackson, 2018-01-26:
I missed to add the converter to the factory - I’ll do an update on this tonight or tomorrow, so when you get a chance please update before doing more testing.

(enjoy your weekend :) ).

AfromD, 2018-01-29:
I've tested with  version 2.3.0.201801282104
ST motion sensor now shows channels 
zigbee:device:3e5f29cf:24fd5b00010603d8:24FD5B00010603D8_1_ias_standard_system
zigbee:device:3e5f29cf:24fd5b00010603d8:24FD5B00010603D8_1_measurement_temperature

Nothing reacts to motion though.
Here's the log of the motion sensor joining the network
[New Text Document (6).txt](https://github.com/openhab/org.openhab.binding.zigbee/files/1672795/New.Text.Document.6.txt)

cdjackson, 2018-01-29:
I see the issue - one line is missing in the converter so I'll add this later. Otherwise it looks like it is working ok - the IAS configuration all looks fine and the zone messages are being received.

cdjackson, 2018-01-29:
Hopefully this should be resolved with the above PR. I'll merge this and create a new binary later this evening.

For my info...
```
ReadAttributesResponse [Basic: 23621/1 -> 0/1, cluster=0000, TID=25, records=[ReadAttributeStatusRecord [attributeDataType=CHARACTER_STRING, attributeIdentifier=5, status=SUCCESS, attributeValue=motionv4]]]
```

AfromD, 2018-01-31:
I am on binding version 2.3.0.201801292203 now. After changing the binding version, the thing would never come online, so I decided to delete it, and join again. Unfortunately, I haven't been able to join the motion sensors on this version. It has always been hit-and-miss, but this time they simply won't pair. 
Here's a log, where I try to pair two different ST motion sensors.
[New Text Document (7).txt](https://github.com/openhab/org.openhab.binding.zigbee/files/1680783/New.Text.Document.7.txt)

cdjackson, 2018-01-31:
There are no nodes showing in the log trying to join the network. Maybe the stick is full already (it allows 16 joins) - I'm not sure how it handles multiple re-pairing of the same device.

I will add a dump of current children to the startup log so we can see if this might be part of the issue at least.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/198

cdjackson, 2018-01-31:
I've added a dump of the children currently connected to the device. I'm not sure this includes routers, or just end devices, but let's see if this shows anything.

AfromD, 2018-01-31:
Here's my log
[New Text Document.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/1681602/New.Text.Document.txt)

cdjackson, 2018-01-31:
Thanks. Child count is showing as 0, so it’s not that then - although this doesn’t seem to include routers that have joined via the coordinator as I do see other devices connected in your network. I’m not sure why else it’s not joining, but if there’s no join, then the binding can’t do anything.

I’ll try and check that this child count actually means something - it will likely take a couple of days to get this confirmation.

cdjackson, 2018-02-17:
Confirmed this issues is resolved.

closed 2018-02-17 by cdjackson