openhab/org.openhab.binding.zigbee #114: Trust Lamp color channel type not identified

AfromD, 2018-01-24:
My Trust branded lamps leave an warn message in the logs because the color type is not known, but assumed. The assumed Hue/Sat seems to work. See attached log.
[New Text Document (4).txt](https://github.com/openhab/org.openhab.binding.zigbee/files/1661518/New.Text.Document.4.txt)

cdjackson, 2018-01-25:
The device doesn't support the attribute request so the binding assumes that hue is supported.

```
2018-01-24 21:03:44.807 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00158D0000AF412C: Node 48977 is added to the network

2018-01-24 21:03:50.941 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: DiscoverAttributesCommand [Color control: 0/0 -> 48977/2, cluster=0300, TID=58, startAttributeIdentifier=0, maximumAttributeIdentifiers=10]
2018-01-24 21:03:55.489 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: DefaultResponse [Color control: 48977/2 -> 0/1, cluster=0300, TID=58, commandIdentifier=12, statusCode=UNSUP_GENERAL_COMMAND]
```

There's not a lot that can be done. I could look to change the logging to debug in this case, but that's about it.

cdjackson, 2018-01-27:
I will update the way the color control cluster works to hopefully detect this differently.

cdjackson, 2018-01-28:
Closed by #121

closed 2018-01-28 by cdjackson