openhab/org.openhab.binding.zigbee #119: On/Off state of level channel is wrong

weakfl, 2018-01-26:
Following up on [your comment](https://github.com/openhab/org.openhab.binding.zigbee/issues/108#issuecomment-360764413), it looks like your assumption was right.

It seems like the BJ relays are intermittently reporting wrong level values:

```
10:18:43.890 [DEBUG] [.converter.ZigBeeConverterSwitchLevel] - D85DEF11A1001973: ZigBee attribute reports ZclAttribute [cluster=LEVEL_CONTROL, id=0, name=CurrentLevel, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=1]
10:18:43.910 [DEBUG] [.converter.ZigBeeBaseChannelConverter] - D85DEF11A1001973: Channel zigbee:device:cb2e62e2:d85def11a1001973:D85DEF11A1001973_18_switch_level updated to 0
```

```
09:27:23.647 [DEBUG] [.converter.ZigBeeConverterSwitchLevel] - D85DEF11A1002826: ZigBee attribute reports ZclAttribute [cluster=LEVEL_CONTROL, id=0, name=CurrentLevel, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=2]
09:27:23.653 [DEBUG] [.converter.ZigBeeBaseChannelConverter] - D85DEF11A1002826: Channel zigbee:device:cb2e62e2:d85def11a1002826:D85DEF11A1002826_18_switch_level updated to 1
```

Maybe you could make the threshold for the Off state configurable?

weakfl, 2018-01-26:
One more question, shouldn't 2/255 still result in a dimmer level of 0%? Why is a value of 2 treated as On?

cdjackson, 2018-01-26:
> It seems like the BJ relays are intermittently reporting wrong level values:

Are you sure it's not really changing? Is this some sort of touch controller, or is there a physical relay?  I'm a bit surprised that it's periodically sending the wrong value.

> Why is a value of 2 treated as On?

I expect any value other than 0 should be treated as ON shouldn't it?

However, the formula is 2 / 254 * 100 + 0.5, so that rounds down to 1%.

weakfl, 2018-01-26:
> Are you sure it's not really changing? Is this some sort of touch controller, or is there a physical relay? I'm a bit surprised that it's periodically sending the wrong value.

100% sure, it's a physical relay. I've noticed it in the past and was surprised as well, must be a bug.

> I expect any value other than 0 should be treated as ON shouldn't it?

Of course.

> However, the formula is 2 / 254 * 100 + 0.5, so that rounds down to 1%.

That looks a bit like a hack imho. A possible value of 101% isn't exactly correct either.
However, rounding 2/254 would be 1% too, so not much to discuss 😉

cdjackson, 2018-01-26:
> 100% sure, it's a physical relay

So it can't be used as a dimmer?

> That looks a bit like a hack imho.

No - this is needed to avoid rounding errors.

> A possible value of 101% isn't exactly correct either.

A value of 101% is not possible with this formulae - this is exactly the point. It ensures that we get the full range - not too high, and not too low. Without the +0.5%, it is not possible to get to 100% in some cases - with this, it is possible to get 100%, but not possible to get more.

weakfl, 2018-01-26:
> So it can't be used as a dimmer?

No. Believe me, I've tried it 😄 

BJ offers two different kinds of inserts:
[6711](http://www.busch-jaeger-catalogue.com/2CKA006710A0006,artikel.html): switchable, but **NOT** dimmable
[6715](http://www.busch-jaeger-catalogue.com/2CKA006710A0007,artikel.html): switchable **AND** dimmable

They must be using a variation of the same firmware, that's the only reason I can think of why the 6711 has a level channel.

cdjackson, 2018-01-26:
Ok, I'm hesitant to add code to put threshold in to fix what I think is an error in the device. What I propose is if you can live with this for a few weeks, then we will look to resolve this with #101 which is planned to add thing definitions into the binding to allow more detailed definition of devices. With this, we can remove the level channel completely and you can just have the on/off channel.

Is that ok? If so, I'd suggest to close this, or to rename it to make it specific to this device and to provide the device properties from HABmin or PaperUI along with the device definition... For this, I need to get the part of the XML for this device that is stored in the userdata/zigbee folder - there's an XML that records basic information about the network devices.

weakfl, 2018-01-26:
> With this, we can remove the level channel completely and you can just have the on/off channel.

Sounds like the proper solution, **thanks!**

I just hope that the dimmable relay doesn't have this bug, removing the level channel wouldn't be an option. But we'll see when I get one for testing. What makes it worse is, that the firmware can only be updated with a special zigbee stick from BJ and access to the firmware, so that's impossible for the average user.

Here are the requested infos:

```xml
<ZigBeeNode>
  <ieeeAddress>D85DEF11A1002826</ieeeAddress>
  <networkAddress>16742</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>89</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>4398</manufacturerCode>
    <logicalType>ROUTER</logicalType>
    <serverCapabilities/>
    <incomingTransferSize>61</incomingTransferSize>
    <outgoingTransferSize>61</outgoingTransferSize>
    <userDescriptorAvailable>true</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities>
      <MacCapabilitiesType>MAINS_POWER</MacCapabilitiesType>
      <MacCapabilitiesType>RECEIVER_ON_WHEN_IDLE</MacCapabilitiesType>
      <MacCapabilitiesType>FULL_FUNCTION_DEVICE</MacCapabilitiesType>
    </macCapabilities>
    <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
    <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
  </nodeDescriptor>
  <powerDescriptor>
    <currentPowerMode>RECEIVER_ON_IDLE</currentPowerMode>
    <availablePowerSources>
      <PowerSourceType>RECHARGABLE_BATTERY</PowerSourceType>
      <PowerSourceType>DISPOSABLE_BATTERY</PowerSourceType>
      <PowerSourceType>MAINS</PowerSourceType>
    </availablePowerSources>
    <currentPowerSource>MAINS</currentPowerSource>
    <powerLevel>FULL</powerLevel>
  </powerDescriptor>
  <endpoints>
    <ZigBeeEndpoint>
      <profileId>49246</profileId>
      <endpointId>18</endpointId>
      <inputClusterIds>
        <int>0</int>
        <int>3</int>
        <int>4</int>
        <int>5</int>
        <int>6</int>
        <int>8</int>
      </inputClusterIds>
      <outputClusterIds/>
    </ZigBeeEndpoint>
    <ZigBeeEndpoint>
      <profileId>49246</profileId>
      <endpointId>10</endpointId>
      <inputClusterIds>
        <int>0</int>
        <int>4096</int>
      </inputClusterIds>
      <outputClusterIds>
        <int>4096</int>
        <int>768</int>
        <int>3</int>
        <int>4</int>
        <int>5</int>
        <int>6</int>
        <int>8</int>
        <int>25</int>
      </outputClusterIds>
    </ZigBeeEndpoint>
  </endpoints>
</ZigBeeNode>
```

```
zigbee_logicaltype  ROUTER
zigbee_powerlevel FULL
modelId RM01
zigbee_networkaddress 16742
zigbee_powersource  MAINS
zigbee_stkversion 1
zigbee_datecode 20161209
zigbee_zclversion 1
zigbee_routes [{"next_hop":"0","destination":"0","state":"ACTIVE"},{"next_hop":"24666","destination":"24666","state":"ACTIVE"},{"next_hop":"17404","destination":"17404","state":"ACTIVE"}]
zigbee_lastupdate 2018-01-26T14:30:41Z
vendor  Busch-Jaeger
zigbee_powermode  RECEIVER_ON_IDLE
zigbee_permitjoining  true
zigbee_powersources [RECHARGABLE_BATTERY, DISPOSABLE_BATTERY, MAINS]
hardwareVersion 0
zigbee_neighbors  [{"joining":"ENABLED","address":"0","depth":"0","lqi":"252","macaddress":"000D6F000D03D29E"},{"joining":"DISABLED","address":"17404","depth":"2","lqi":"252","macaddress":"D85DEF11A1001973"}]
zigbee_devices  []
```

cdjackson, 2018-01-26:
> What makes it worse is, that the firmware can only be updated with a special zigbee stick from BJ and access to the firmware, so that's impossible for the average user.

Says who? The device is reporting that it supports the standard OTA Upgrade cluster, which means it can be upgraded in the standard way. If you can get the firmware files, then it can be upgraded easily (well, reasonably easily).

weakfl, 2018-01-26:
> Says who? The device is reporting that it supports the standard OTA Upgrade cluster, which means it can be upgraded in the standard way. If you can get the firmware files, then it can be upgraded easily (well, reasonably easily).

Right, didn't think about that. The firmware is not publicly available though as far as I know. I'll still report the issue to BJ and see if I can get a recent firmware just in case.

cdjackson, 2018-01-26:
Interested to see what they say :)

weakfl, 2018-01-31:
I've heard back from BJ. They say that the problem should be resolved with the latest firmware and they'll send me an update-stick for free. They've also provided the necessary software to update all BJ zigbee components.

So great customer service, got to give them that.

cdjackson, 2018-01-31:
Great - thanks. So it seems they confirm the problem is with them, so I'd suggest we close this and if the issue doesn't get resolved, we can look to add a fix later?

weakfl, 2018-01-31:
Sounds good 👍

closed 2018-01-31 by weakfl