openhab/org.openhab.binding.zigbee #125: Telegesis Coordinator not correctly initialized

kaikreuzer, 2018-01-29:
This is from my log when adding a Telegesis coordinator:
```
17:35:09.597 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:67d97600' changed from UNINITIALIZED to INITIALIZING
17:35:09.639 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:67d97600' has been updated.
17:35:09.643 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:67d97600' has been updated.
17:35:09.647 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:67d97600' has been updated.
17:35:09.666 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:67d97600' changed from INITIALIZING to UNKNOWN
17:35:09.747 [INFO ] [arthome.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:coordinator_telegesis:67d97600 changed to UNKNOWN.
17:35:13.313 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:67d97600' changed from UNKNOWN to OFFLINE: Failed to startup ZigBee transport layer
17:35:15.273 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:67d97600' changed from OFFLINE: Failed to startup ZigBee transport layer to ONLINE
17:35:15.569 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:67d97600' changed from ONLINE to OFFLINE: Failed to startup ZigBee transport layer
17:35:16.937 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:67d97600' has been updated.
17:35:16.943 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:67d97600' has been updated.
```
As you can see, it briefly goes to ONLINE, but in the end it is set to OFFLINE (although it managed to start up the transport layer). Editing and saving the coordinator Thing made it come ONLINE correctly.
I assume that there are some timing issues during initialization, which might be due to the rather many Thing updates that are done in that process (could those be reduced?).

cdjackson, 2018-01-29:
Can you provide a debug log? This error is normally due to not opening the serial port, or not being able to communicate with the Telegesis dongle.

kaikreuzer, 2018-01-29:
Here you go:
```
19:28:22.286 [DEBUG] [.zigbee.internal.ZigBeeHandlerFactory] - Creating coordinator handler for org.eclipse.smarthome.core.thing.internal.BridgeImpl@a551a8b2
19:28:22.289 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - Creating ZigBee discovery service for zigbee:coordinator_telegesis:1128e2ca
19:28:22.290 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - Activating ZigBee discovery service for zigbee:coordinator_telegesis:1128e2ca
19:28:22.291 [DEBUG] [org.openhab.binding.zigbee           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.discovery.DiscoveryService}={service.id=315, service.bundleid=202, service.scope=singleton} - org.openhab.binding.zigbee
19:28:22.294 [DEBUG] [org.openhab.binding.zigbee           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.core.thing.binding.firmware.FirmwareUpdateHandler}={service.id=316, service.bundleid=202, service.scope=singleton} - org.openhab.binding.zigbee
19:28:22.302 [DEBUG] [ler.ZigBeeCoordinatorTelegesisHandler] - Initializing ZigBee Telegesis serial bridge handler.
19:28:22.303 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Initializing ZigBee network [zigbee:coordinator_telegesis:1128e2ca].
19:28:22.305 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Channel -1
19:28:22.306 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - PANID 0
19:28:22.307 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:1128e2ca' changed from UNINITIALIZED to INITIALIZING
19:28:22.307 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - EPANID 0000000000000000
19:28:22.308 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Key null
19:28:22.309 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Initialising network
19:28:22.310 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Created random ZigBee PAN ID [8CD9].
19:28:22.316 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Created random ZigBee extended PAN ID [507D36CA3B915E95].
19:28:22.318 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:1128e2ca' has been updated.
19:28:22.330 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:1128e2ca' has been updated.
19:28:22.333 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Key String 
19:28:22.335 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:1128e2ca' has been updated.
19:28:22.337 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Key initialised 0534DF5F14FC589F2BED4909CCDB7E73
19:28:22.338 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Key final array 0534DF5F14FC589F2BED4909CCDB7E73
19:28:22.339 [DEBUG] [ler.ZigBeeCoordinatorTelegesisHandler] - ZigBee Coordinator Telegesis opening Port:'/dev/tty.SLAB_USBtoUART' PAN:8cd9, EPAN:507D36CA3B915E95, Channel:-1
19:28:22.341 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Scheduling ZigBee start
19:28:22.342 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:coordinator_telegesis:1128e2ca' changed from INITIALIZING to UNKNOWN
19:28:22.345 [INFO ] [arthome.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:coordinator_telegesis:1128e2ca changed to UNKNOWN.
19:28:23.344 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - ZigBee network starting
19:28:23.346 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Initialising ZigBee coordinator
19:28:23.348 [DEBUG] [gbee.handler.ZigBeeCoordinatorHandler] - Key initialise 0534DF5F14FC589F2BED4909CCDB7E73
19:28:23.350 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis dongle initialize.
19:28:23.351 [DEBUG] [ding.zigbee.internal.ZigBeeSerialPort] - Connecting to serial port [/dev/tty.SLAB_USBtoUART] at 19200 baud, flow control FLOWCONTROL_OUT_NONE.
19:28:23.370 [DEBUG] [ding.zigbee.internal.ZigBeeSerialPort] - Serial port [/dev/tty.SLAB_USBtoUART] is initialized.
19:28:23.372 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TelegesisFrameHandler thread started
19:28:23.372 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
19:28:23.373 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSetPromptEnable1Command [configuration=8804]
19:28:23.374 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS TX: Data 41 54 53 30 45 3D 38 38 30 34 0D 0A
19:28:23.488 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 4F 4B
19:28:23.489 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis command complete: TelegesisSetPromptEnable1Command [configuration=8804, status=SUCCESS]
19:28:23.491 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
19:28:23.492 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
19:28:23.494 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS TX: Data 41 54 53 31 32 34 3D 31 0D 0A
19:28:23.616 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 4F 4B
19:28:23.617 [DEBUG] [egesis.internal.TelegesisFrameHandler] - Telegesis command complete: TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null, status=SUCCESS]
19:28:23.619 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
19:28:23.620 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSetMainFunctionCommand [configuration=350, password=password]
[... cut ...]
19:28:29.484 [DEBUG] [gbee.internal.ZigBeeNetworkDiscoverer] - 0/2: Ending endpoint discovery
19:28:29.485 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:1128e2ca' has been updated.
19:28:29.491 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:coordinator_telegesis:1128e2ca' has been updated.
```
I tried it several times, it is fully reproducible.
Status goes from UNINITIALIZED->INITIALIZING->UNKNOWN->"OFFLINE: zigbee.status.offline_startupfail"->ONLINE->"OFFLINE: zigbee.status.offline_startupfail".

cdjackson, 2018-01-29:
Thanks. Seems to be an error in the initialisation of the dongle under specific conditions (no PanId or Reset Controller set to true).

cdjackson, 2018-01-29:
Ok, so looks simple I think. The dongle driver is waiting for a network left event from the dongle, and it times out after 500ms but at least for your dongle it’s taking 

```
19:28:24.806 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisLeaveNetworkCommand []
19:28:24.807 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS TX: Data 41 54 2B 44 41 53 53 4C 0D 0A
19:28:24.929 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TELEGESIS RX: Data 4F 4B
19:28:25.436 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Error leaving network: No event received

19:28:25.832 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis RX: TelegesisNetworkLeftEvent []
```

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/195

cdjackson, 2018-01-29:
Should be fixed by https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/195 once it's merged into the binding.

kaikreuzer, 2018-01-29:
Thanks @cdjackson!

cdjackson, 2018-03-05:
This should be resolved so closing. If there's further issues, let me know.

closed 2018-03-05 by cdjackson