openhab/org.openhab.binding.zigbee #144: Unit tests not run by maven (as testSourceDirectory incorrectly configured)

hsudbrock, 2018-02-08:
In the `pom.xml`, the source directory for the tests is configured as `<testSourceDirectory>src/test</testSourceDirectory>`, although the unit tests are actually in directory `src/main/test`. In consequence, the maven build will not run any unit tests.

Besides that, would it be sensible to move the unit tests to `/src/test/java`, to align with default maven directory layout?

kaikreuzer, 2018-02-08:
You are right, this is somehow messed up - the tests are actually currently in "src/main/test", while they should be in "src/test/java" instead. And then the "testSourceDirectory" entry could be removed from the pom.xml completely.

hsudbrock, 2018-02-08:
I will create a short PR for this (including a small change for making all tests pass again), so we can discuss in the PR if the changes for this small cleanup are the right ones.

PR #145 (2018-02-08): Make maven run the unit tests
Fixes #144

Moves the unit tests from directory `src/main/test` to directory `src/test/java`, to adhere to the default maven directory structure, and removes `testSourceDirectory` directive from the `pom.xml`. In consequence, the unit tests are executed by the maven surefire plugin (whereas, before this change, surefire ran but could not find the tests).

Moreover, adds a small fix to the `ZigBeeConverterSwitchOnOffTest`, where mockito needs to be told explicitly the type for an 'any' matcher.
commit c75cb935 (2018-02-08): Make maven run the unit tests (#145)

Moves the unit tests from directory src/main/test to directory src/test/java, to adhere to the default maven directory structure, and removes testSourceDirectory from the pom.xml. In consequence, the unit tests are executed by the maven surefire plugin (whereas, before this change, surefire ran but could not find the tests).

Moreover, adds a small fix to the ZigBeeConverterSwitchOnOffTest, where mockito needs to be told explicitly the type for an 'any' matcher.

Fixes #144
closed 2018-02-08 by cdjackson