openhab/org.openhab.binding.zigbee #150: Build broken

tavalin, 2018-02-12:
Hi Chris

Commit https://github.com/openhab/org.openhab.binding.zigbee/commit/ac176d7a6716069c92d630f84496ecdab3f09d10 removed a constant from ZigBeeBindingConstants - not sure if this was deliberate or not - but there are usages of the constant in other classes so the mvn build fails.

cdjackson, 2018-02-12:
Damn. I was working on two different PRs and the removal of the use of the constant seems to be in the other one.

kaikreuzer, 2018-02-12:
@cdjackson I have now added you to the build notifications so that you notice it yourself when a build that you triggered fails :-)

cdjackson, 2018-02-12:
Thanks. I wonder if PR builds are not configured on this repo as there was no failure on the PR itself before merging...

cdjackson, 2018-02-12:
@kaikreuzer I've added CI in travis to verify PRs - I'd not noticed this wasn't configured already...

closed 2018-02-12 by cdjackson

kaikreuzer, 2018-02-12:
Ah, that explains. I was just trying to do the same and was wondering that it was already there, but only 5 times executed. No wonder when you just only had it set up - thanks :-)