openhab/org.openhab.binding.zigbee #160: Exception creating channels

openhab-5iver, 2018-03-02:
OH 2.3.0 snapshot 1218
Zigbee 2.3.0.201802261341 (current version from Cloudbees)

I setup an old GE Link bulb that had been paired a long time ago but was pulled out and sitting on a shelf for a couple months. I was having issues getting to come online, so I deleted the Thing and rediscovered. It is coming online and I've been able to control it, but it become unresponsive. I just sat down to look at the log, and found this after a fresh start of OH...
```
2018-03-02 15:32:34.115 [ERROR] [.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 7CE524000010E370: Exception creating channels 
java.lang.ClassCastException: java.lang.String cannot be cast to java.math.BigDecimal
	at org.openhab.binding.zigbee.internal.converter.config.ZclLevelControlConfig.updateConfiguration(ZclLevelControlConfig.java:119) [16:org.openhab.binding.zigbee:2.3.0.201802261341]
	at org.openhab.binding.zigbee.internal.converter.ZigBeeConverterSwitchLevel.updateConfiguration(ZigBeeConverterSwitchLevel.java:120) [16:org.openhab.binding.zigbee:2.3.0.201802261341]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.doNodeInitialisation(ZigBeeThingHandler.java:234) [16:org.openhab.binding.zigbee:2.3.0.201802261341]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.access$0(ZigBeeThingHandler.java:156) [16:org.openhab.binding.zigbee:2.3.0.201802261341]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:150) [16:org.openhab.binding.zigbee:2.3.0.201802261341]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:1) [16:org.openhab.binding.zigbee:2.3.0.201802261341]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:?]
	at java.lang.Thread.run(Thread.java:748) [?:?]
```

The bulb is not responsive, and there are other oddities in the log. This is repeated a few times...
```
2018-03-02 15:35:11.980 [DEBUG] [.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 7CE524000010E370: Checking endpoint 1 channels
2018-03-02 15:35:11.981 [DEBUG] [e.internal.converter.ZigBeeChannelConverterFactory] - 7CE524000010E370: Removing channel zigbee:switch_onoff in favor of zigbee:switch_level
2018-03-02 15:35:11.981 [DEBUG] [.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 7CE524000010E370: Created 1 channels
2018-03-02 15:35:11.984 [DEBUG] [.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 7CE524000010E370: Initializing channel zigbee:device:9b51c72d:7ce524000010e370:7CE524000010E370_1_switch_level with org.openhab.binding.zigbee.internal.converter.ZigBeeConverterSwitchLevel@36bb8f89
```

And this is repeated a lot...
```
2018-03-02 15:53:21.558 [DEBUG] [gbee.internal.converter.ZigBeeBaseChannelConverter] - 7CE524000010E370: Channel zigbee:device:9b51c72d:7ce524000010e370:7CE524000010E370_1_switch_level updated to 100
2018-03-02 15:53:21.558 [DEBUG] [gbee.internal.converter.ZigBeeConverterSwitchLevel] - 7CE524000010E370: ZigBee attribute reports ZclAttribute [cluster=LEVEL_CONTROL, id=0, name=CurrentLevel, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=254, lastReportTime=Fri Mar 02 15:53:21 EST 2018]
2018-03-02 15:53:21.558 [DEBUG] [.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 7CE524000010E370: Updating ZigBee channel state zigbee:device:9b51c72d:7ce524000010e370:7CE524000010E370_1_switch_level to 100
```

Before restarting OH, this NPE was repeated in the Karaf console many times...
```
Exception in thread "pool-1457-thread-11323" java.lang.IllegalStateException: Could not update state, because callback is missing
        at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.updateState(BaseThingHandler.java:340)
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.setChannelState(ZigBeeThingHandler.java:454)
        at org.openhab.binding.zigbee.internal.converter.ZigBeeBaseChannelConverter.updateChannelState(ZigBeeBaseChannelConverter.java:236)
        at org.openhab.binding.zigbee.internal.converter.ZigBeeConverterSwitchLevel.attributeUpdated(ZigBeeConverterSwitchLevel.java:132)
        at com.zsmartsystems.zigbee.zcl.ZclCluster$4.run(ZclCluster.java:778)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```
[030218 zigbee bulb not responding.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/1776629/030218.zigbee.bulb.not.responding.zip)

cdjackson, 2018-03-02:
I think there are a bunch of things reported here that are not likely linked. The fact the device was not responding probably not linked to the exception.

I don't see the callback missing exception in the log so not sure I can comment on that.

The log looks ok - it looks like it's an integer so it should cast ok. I'll probably update the debug to log the type, but I won't be making any further changes to the binding until #159 is merged.

cdjackson, 2018-03-02:
I'm wondering if there's still a normalisation issue here. Can you take a look in your json database to see if you can find this configuration (zigbee_levelcontrol_transitiontimeonoff) and see what the type is? In your log it looks like a number, but I wonder if it's really a string?

openhab-5iver, 2018-03-03:
It looks like every value for this Thing is a string. Here's the channel...
```
        {
          "acceptedItemType": "Dimmer",
          "kind": "STATE",
          "uid": {
            "segments": [
              "zigbee",
              "device",
              "9b51c72d",
              "7ce524000010e370",
              "7CE524000010E370_1_switch_level"
            ]
          },
          "channelTypeUID": {
            "segments": [
              "zigbee",
              "switch_level"
            ]
          },
          "label": "Dimmer",
          "configuration": {
            "properties": {
              "zigbee_levelcontrol_transitiontimeonoff": "0",
              "zigbee_levelcontrol_transitiontimeoff": "65535",
              "zigbee_levelcontrol_transitiontimedefault": "65535",
              "zigbee_levelcontrol_onlevel": "255",
              "zigbee_levelcontrol_defaultrate": "255",
              "zigbee_levelcontrol_transitiontimeon": "65535"
            }
          },
          "properties": {
            "zigbee_endpoint": "1"
          },
          "defaultTags": []
        }
```
What's really odd is that there is no other reference to transitiontime in the Thing jsondb. I bet if I delete another Zigbee Thing there would be another device having issues...

OK... so I deleted another Zigbee Thing, rediscovered, and that bulb is no longer responding. It now has the transitiontime entries in the Thing jsondb too.

cdjackson, 2018-03-03:
> It looks like every value for this Thing is a string. Here's the channel...

So to me this means the normalisation hasn't worked in this case.  Do you think you ever changed this in the UI, or is it default? I know @sjka changed the normalisation in the rest interface, so if you save, it should work as I tested this. Maybe somewhere there is another case where it's not being normalised.

Try saving the new things in PaperUI. Don't use HABmin in this case as HABmin will only update parameters you change. Normally that's a good thing, but here you want to globally rewrite the configuration so that the normalisation (hopefully!) resolves all the types.

openhab-5iver, 2018-03-03:
[#5018](https://github.com/eclipse/smarthome/pull/5018) was only merged a few days ago, so I grabbed the latest OH snapshot (1221). This version is using ESH 0.10.0.201802191927, so I doubt they changes have made it into the build. I was also having issues getting this build to run properly.

After rolling back, saving the new Things in PaperUI using 1218 had no affect on the bulbs' performance. The changes appeared to save, but the jsondb was not changed. I also noticed that there is only one transition time setting available in these devices, where the others have seven...

[image]

When the jsondb is manually updated to have integer values, the strings come right back.

cdjackson, 2018-03-11:
This is fundamentally the same issue as #36

closed 2018-03-11 by cdjackson