openhab/org.openhab.binding.zigbee #163: Xioami (aqara) devices do not complete discovery

oysteinkrog, 2018-03-07:
### Outline
I've been able to find and add two aqara devices (a temperature and humidity sensor and a door latch sensor), but in Openhab it says that device disovery is not completed.

### Configuration
My setup consists of RPI2, a CC2531 (flashed with CC2531ZNP-Pro-Secure_LinkKeyJoin.hex) and openhabian (with openhab 2.2).
I'm running the latest 2.3 snapshot of the binding.

### Logs
```
2018-03-07 18:34:55.649 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00124B000E896A5E: Node SVC Discovery running

2018-03-07 18:34:55.671 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0/0 -> 65533/0, cluster=0000, TID=58, ieeeAddr=00124B000E896A5E, requestType=0, startIndex=0]

2018-03-07 18:34:55.691 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0000, cluster=0, addressMode=DEVICE, radius=31, sequence=88, payload=00 5E 6A 89 0E 00 4B 12 00 00 00]

2018-03-07 18:34:55.702 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 58 30 1F 0B 00 5E 6A 89 0E 00 4B 12 00 00 00 A4, checksum=A4, error=false) 

2018-03-07 18:34:55.837 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2018-03-07 18:35:05.851 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 0: Node SVC Discovery NetworkAddressRequest returned null

2018-03-07 18:35:05.867 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00124B000E896A5E: Node SVC Discovery request NWK_ADDRESS failed. Retry 270, wait 676494ms before retry.

2018-03-07 18:35:40.572 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:36:40.585 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:36:57.156 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0001AB3773: Node SVC Discovery running

2018-03-07 18:36:57.176 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0/0 -> 65533/0, cluster=0000, TID=59, ieeeAddr=00158D0001AB3773, requestType=0, startIndex=0]

2018-03-07 18:36:57.195 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0000, cluster=0, addressMode=DEVICE, radius=31, sequence=89, payload=00 73 37 AB 01 00 8D 15 00 00 00]

2018-03-07 18:36:57.208 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 59 30 1F 0B 00 73 37 AB 01 00 8D 15 00 00 00 39, checksum=39, error=false) 

2018-03-07 18:36:57.346 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2018-03-07 18:37:07.377 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 29499: Node SVC Discovery NetworkAddressRequest returned null

2018-03-07 18:37:07.394 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0001AB3773: Node SVC Discovery request NWK_ADDRESS failed. Retry 287, wait 348645ms before retry.

2018-03-07 18:37:40.599 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:38:02.628 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0001AB3773: Node SVC Discovery already scheduled or running

2018-03-07 18:38:02.646 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00124B000E896A5E: Node SVC Discovery already scheduled or running

2018-03-07 18:38:02.664 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery already scheduled or running

2018-03-07 18:38:36.834 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery running

2018-03-07 18:38:36.856 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0/0 -> 65533/0, cluster=0000, TID=5A, ieeeAddr=00158D0002273EC3, requestType=0, startIndex=0]

2018-03-07 18:38:36.877 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0000, cluster=0, addressMode=DEVICE, radius=31, sequence=90, payload=00 C3 3E 27 02 00 8D 15 00 00 00]

2018-03-07 18:38:36.889 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 5A 30 1F 0B 00 C3 3E 27 02 00 8D 15 00 00 00 0C, checksum=0C, error=false) 

2018-03-07 18:38:37.030 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2018-03-07 18:38:40.614 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:38:47.044 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 18510: Node SVC Discovery NetworkAddressRequest returned null

2018-03-07 18:38:47.061 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery request NWK_ADDRESS failed. Retry 293, wait 212660ms before retry.

2018-03-07 18:39:40.630 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:40:40.644 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:41:40.688 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job
[... 102 lines cut ...]

2018-03-07 18:52:40.851 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 18:53:02.627 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0001AB3773: Node SVC Discovery already scheduled or running

2018-03-07 18:53:02.646 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00124B000E896A5E: Node SVC Discovery already scheduled or running

2018-03-07 18:53:02.663 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery already scheduled or running

2018-03-07 18:53:05.785 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery running

2018-03-07 18:53:05.804 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0/0 -> 65533/0, cluster=0000, TID=60, ieeeAddr=00158D0002273EC3, requestType=0, startIndex=0]

2018-03-07 18:53:05.824 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0000, cluster=0, addressMode=DEVICE, radius=31, sequence=96, payload=00 C3 3E 27 02 00 8D 15 00 00 00]

2018-03-07 18:53:05.836 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 60 30 1F 0B 00 C3 3E 27 02 00 8D 15 00 00 00 36, checksum=36, error=false) 

2018-03-07 18:53:05.965 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2018-03-07 18:53:15.976 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 18510: Node SVC Discovery NetworkAddressRequest returned null

2018-03-07 18:53:15.992 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery request NWK_ADDRESS failed. Retry 295, wait 414470ms before retry.

2018-03-07 18:53:40.880 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job
```

oysteinkrog, 2018-03-07:
I tried CC2531ZNP-Pro-Secure_Standard.hex as well but it seems to be the same

```

==> /var/log/openhab2/openhab.log <==

2018-03-07 21:14:52.990 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9A 30 1F 0B 00 73 37 AB 01 00 8D 15 00 00 00 FA, checksum=FA, error=false) 

2018-03-07 21:14:54.115 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:14:54.135 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 3]

2018-03-07 21:14:54.147 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9B 30 1F 0B 00 C3 3E 27 02 00 8D 15 00 00 00 CD, checksum=CD, error=false) 

2018-03-07 21:14:55.282 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:14:55.296 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 1]

2018-03-07 21:14:56.318 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9B 30 1F 0B 00 C3 3E 27 02 00 8D 15 00 00 00 CD, checksum=CD, error=false) 

2018-03-07 21:14:57.457 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:14:57.483 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 2]

2018-03-07 21:14:58.507 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9B 30 1F 0B 00 C3 3E 27 02 00 8D 15 00 00 00 CD, checksum=CD, error=false) 

2018-03-07 21:14:59.632 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:14:59.651 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 3]

2018-03-07 21:15:04.156 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 00158D0001AB3773: NWK Discovery node rediscovery request failed. Wait before retry.

2018-03-07 21:15:05.695 [DEBUG] [bee.internal.ZigBeeNetworkDiscoverer] - 00158D0002273EC3: NWK Discovery node rediscovery request failed. Wait before retry.

2018-03-07 21:15:05.688 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0/0 -> 65533/0, cluster=0000, TID=9C, ieeeAddr=00158D0001AB3773, requestType=0, startIndex=0]

2018-03-07 21:15:05.761 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0000, cluster=0, addressMode=DEVICE, radius=31, sequence=156, payload=00 73 37 AB 01 00 8D 15 00 00 00]

2018-03-07 21:15:05.773 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9C 30 1F 0B 00 73 37 AB 01 00 8D 15 00 00 00 FC, checksum=FC, error=false) 

2018-03-07 21:15:06.910 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:15:06.925 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 1]

2018-03-07 21:15:07.225 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0/0 -> 65533/0, cluster=0000, TID=9D, ieeeAddr=00158D0002273EC3, requestType=0, startIndex=0]

2018-03-07 21:15:07.263 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0000, cluster=0, addressMode=DEVICE, radius=31, sequence=157, payload=00 C3 3E 27 02 00 8D 15 00 00 00]

2018-03-07 21:15:08.442 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9C 30 1F 0B 00 73 37 AB 01 00 8D 15 00 00 00 FC, checksum=FC, error=false) 

2018-03-07 21:15:09.567 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:15:09.586 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 2]

2018-03-07 21:15:11.098 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9C 30 1F 0B 00 73 37 AB 01 00 8D 15 00 00 00 FC, checksum=FC, error=false) 

2018-03-07 21:15:12.223 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:15:12.235 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 3]

2018-03-07 21:15:12.260 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=21, apiId=24 01, data=FE 15 24 01 FD FF 00 00 00 00 9D 30 1F 0B 00 C3 3E 27 02 00 8D 15 00 00 00 CB, checksum=CB, error=false) 

2018-03-07 21:15:13.397 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - AF_DATA_REQUEST executed and timed out while waiting for response.

2018-03-07 21:15:13.417 [DEBUG] [.cc2531.network.ZigBeeNetworkManager] - Failed to send AF_DATA_REQUEST [attempt 1]

[... cut ...]
2018-03-07 21:18:39.466 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 18510: Node SVC Discovery NetworkAddressRequest returned null

2018-03-07 21:18:39.522 [DEBUG] [internal.ZigBeeNodeServiceDiscoverer] - 00158D0002273EC3: Node SVC Discovery request NWK_ADDRESS failed. Retry 311, wait 774690ms before retry.

2018-03-07 21:18:43.301 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job

2018-03-07 21:19:43.331 [DEBUG] [core.karaf.internal.FeatureInstaller] - Running scheduled sync job```

cdjackson, 2018-03-07:
The device works fine and a few people are using these devices (I have personally tested this with Telegesis and Ember dongles, and maybe with XBee). You need to keep pressing the button while the discovery is taking place as the device has a habbit of sleeping quicker than it should (as is seen by other controllers such as SmartThings).

closed 2018-03-07 by cdjackson

horgi007, 2018-03-08:
I have the same issue with xiaomi devices.

oysteinkrog, 2018-03-08:
I've tried a lot of button pressing now, but it does not seem to work. Is there a specific timing needed?

cdjackson, 2018-03-08:
Please discuss on the community forum. These devices are not ZigBee devices so are a bit of a  problem, but there are people using them with some constraints, but the forum is the place to discuss your question.

hansaya, 2018-03-13:
I have the same issue and I own few Xioami devices. I would like to follow this discussion. @oysteinkrog let us know when you create a discussion and link to it :)

cdjackson, 2018-03-13:
Again, the Xiaomi devices, at least the temperature sensors, work fine. This has been discussed a bit on the forum. Some other devices do not work (eg buttons) and will not work in future (most likely).

These devices are not ZigBee devices so they don't necessarily work unfortunately...

As this issue is closed, and as this is not a support forum, I'd suggest to start a discussion on the forum - anyone can do it.

cdjackson, 2018-03-27:
Again, the Xiaomi devices, at least the temperature sensors, work fine. This has been discussed a bit on the forum. Some other devices do not work (eg buttons) and will not work in future (most likely). You might need to press the button a few times after association to wake it up so it downloads all the required data.

These devices are not ZigBee devices so they don't necessarily work unfortunately...

As this issue is closed, and as this is not a support forum, I'd suggest to start a discussion on the forum - anyone can do it.