openhab/org.openhab.binding.zigbee #164: Add support for Develco han-interface - power meter readings

coandersen, 2018-03-08:
In Norway there will be installed 700 000+ power meters which gives owners access to real-time power consumption. Develco has developed a Zigbee-device which accesses the HAN-interface of the power meter. 

If possible I would like support for the device to be added to the Zigbee binding. Especially these attributes:

Id# | Name | Type | Man |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | --
0x0305 | TotalReactivePower | Int32 | O |   |   |   |  
0x0304 | TotalActivePower | Int32 | O |   |   |   |  
0x0505 | RMSVoltagePhA | Uint16 | O | L1 | - | Volts | (V)
0x0508 | RMSCurrentPhA | Uint16 | O | L1 | - | Amps | (A)

[emizb-132-technical-manual-emi-norwegian-han.pdf](https://github.com/openhab/org.openhab.binding.zigbee/files/1794391/emizb-132-technical-manual-emi-norwegian-han.pdf)

cdjackson, 2018-06-11:
Any feedback on this?

coandersen, 2018-06-11:
Sorry, did not see this before now. Does this mean my device is supported by the binding? If so I will test it soon.

linuxtrap, 2019-02-20:
Hello
Did you continue work with this cdjackson? Would be nice with some Power readings to :)

PR #646 (2021-04-25): Add metering channels
Adds metering instantaneous demand, sum delivered and sum received channels.

Closes #625 
Closes #164
closed 2021-04-25 by cdjackson