openhab/org.openhab.binding.zigbee #166: Configurable location of xml network state file

tomhoefer, 2018-03-14:
As an user I would like to have a working ZigBee network / installation after a restart of my runtime.

### Context
The ZigBee binding writes the current network state as an XML file into the ESH user data folder. However this folder is not used in my runtime so that after a restart all previous paired devices are "lost".

### Technical proposal
In my runtime the XML file should be written into the configuration directory of the bundle. In order to achieve this we could introduce a read-only configuration parameter. My proposal for the name of parameter is `xmlNetworkStateLocation`. As possible values either only `USER_DATA` or `BUNDLE_CONFIG` should be possible. For consistency reason we can use a corresponding option list.

The default value should be `USER_DATA` so that no changes for openHAB are necessary. In my runtime I can provide `BUNDLE_CONFIG` as value while creating the bridge thing. Then the binding can determine in which location the file should be created. 

Do you think my suggestion makes sense? Or do you have any better idea how the issue can be implemented?

cdjackson, 2018-03-14:
While I'm not against adding a configuration option to the binding to set the user data folder, I wonder if we shouldn't look at the implementation of the ```ConfigConstants.getUserDataFolder()``` method rather than implement another user data configuration option in the binding?

> The default value should be USER_DATA

The binding uses the ```ConfigConstants.getUserDataFolder()``` folder - in ESH, this method is implemented with the following default -:
```
        String progArg = System.getProperty(CONFIG_DIR_PROG_ARGUMENT);
```

I'm not sure if you can work with this on a binding specific way (probably not?), or if you can modify this method in your system? If possible though, it would have the major benefit of working with every binding rather than having to add this configuration into every binding in order to work in your system. 

If it's not possible in your system to override the ```getUserDataFolder()```, maybe an enhancement can be made to the default implementation of this method to read a binding specific configuration option if present rather than using the system default?

If we can manage the issue at this level, then we solve the problem for all bindings, which I think is desirable :) .

tomhoefer, 2018-03-14:
ok - I agree - I will try to make the usage of the user data folder possible

I will re-open it once I have more questions

closed 2018-03-14 by tomhoefer