openhab/org.openhab.binding.zigbee #167: Expected static field org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.scheduler

openhab-5iver, 2018-03-21:
OH 2.3.0 snapshot 1231
Zigbee 2.3.0.201803161151

I believe this error may be related to this ESH PR... eclipse/smarthome#5227 and https://community.openhab.org/t/unit-support-knx-2-karaf-4-1-5-upgrade-and-more/41904/7?u=5iver.

The error occurred when initiating a Zigbee discovery through Habmin or PaperUI.

```
2018-03-17 21:16:20.675 [WARN ] [org.eclipse.jetty.servlet.ServletHandler          ] - 
javax.servlet.ServletException: org.glassfish.jersey.server.ContainerException: java.lang.IncompatibleClassChangeError: Expected static field org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.scheduler
        at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:489) [165:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.WebComponent.service(WebComponent.java:427) [165:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:388) [165:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:341) [165:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:228) [165:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        at com.eclipsesource.jaxrs.publisher.internal.ServletContainerBridge.service(ServletContainerBridge.java:76) [17:com.eclipsesource.jaxrs.publisher:5.3.1.201602281253]
        at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:848) [82:org.eclipse.jetty.servlet:9.3.21.v20170918]
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:584) [82:org.eclipse.jetty.servlet:9.3.21.v20170918]
        at org.ops4j.pax.web.service.jetty.internal.HttpServiceServletHandler.doHandle(HttpServiceServletHandler.java:71) [181:org.ops4j.pax.web.pax-web-jetty:6.0.9]
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548) [79:org.eclipse.jetty.security:9.3.21.v20170918]
        at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:226) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1180) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.ops4j.pax.web.service.jetty.internal.HttpServiceContext.doHandle(HttpServiceContext.java:284) [181:org.ops4j.pax.web.pax-web-jetty:6.0.9]
        at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:512) [82:org.eclipse.jetty.servlet:9.3.21.v20170918]
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1112) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.ops4j.pax.web.service.jetty.internal.JettyServerHandlerCollection.handle(JettyServerHandlerCollection.java:80) [181:org.ops4j.pax.web.pax-web-jetty:6.0.9]
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:134) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.server.Server.handle(Server.java:534) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:333) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:251) [81:org.eclipse.jetty.server:9.3.21.v20170918]
        at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:283) [74:org.eclipse.jetty.io:9.3.21.v20170918]
        at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:108) [74:org.eclipse.jetty.io:9.3.21.v20170918]
        at org.eclipse.jetty.io.SelectChannelEndPoint$2.run(SelectChannelEndPoint.java:93) [74:org.eclipse.jetty.io:9.3.21.v20170918]
        at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.executeProduceConsume(ExecuteProduceConsume.java:303) [85:org.eclipse.jetty.util:9.3.21.v20170918]
        at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceConsume(ExecuteProduceConsume.java:148) [85:org.eclipse.jetty.util:9.3.21.v20170918]
        at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:136) [85:org.eclipse.jetty.util:9.3.21.v20170918]
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671) [85:org.eclipse.jetty.util:9.3.21.v20170918]
        at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589) [85:org.eclipse.jetty.util:9.3.21.v20170918]
        at java.lang.Thread.run(Thread.java:748) [?:?]
Caused by: org.glassfish.jersey.server.ContainerException: java.lang.IncompatibleClassChangeError: Expected static field org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.scheduler
        at org.glassfish.jersey.servlet.internal.ResponseWriter.rethrow(ResponseWriter.java:278) ~[?:?]
        at org.glassfish.jersey.servlet.internal.ResponseWriter.failure(ResponseWriter.java:260) ~[?:?]
        at org.glassfish.jersey.server.ServerRuntime$Responder.process(ServerRuntime.java:509) ~[168:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.ServerRuntime$2.run(ServerRuntime.java:334) ~[168:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.internal.Errors$1.call(Errors.java:271) ~[167:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors$1.call(Errors.java:267) ~[167:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:315) ~[167:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:297) ~[167:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:267) ~[167:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:317) ~[167:org.glassfish.jersey.core.jersey-common:2.22.2]
        at org.glassfish.jersey.server.ServerRuntime.process(ServerRuntime.java:305) ~[168:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.server.ApplicationHandler.handle(ApplicationHandler.java:1154) ~[168:org.glassfish.jersey.core.jersey-server:2.22.2]
        at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:473) ~[165:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
        ... 31 more
Caused by: java.lang.IncompatibleClassChangeError: Expected static field org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.scheduler
        at org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.nodeDiscovered(ZigBeeDiscoveryService.java:202) ~[?:?]
        at org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.startScan(ZigBeeDiscoveryService.java:111) ~[?:?]
        at org.eclipse.smarthome.config.discovery.AbstractDiscoveryService.startScan(AbstractDiscoveryService.java:208) ~[?:?]
        at org.eclipse.smarthome.config.discovery.internal.DiscoveryServiceRegistryImpl.startScan(DiscoveryServiceRegistryImpl.java:427) ~[?:?]
        at org.eclipse.smarthome.config.discovery.internal.DiscoveryServiceRegistryImpl.startScans(DiscoveryServiceRegistryImpl.java:412) ~[?:?]
        at org.eclipse.smarthome.config.discovery.internal.DiscoveryServiceRegistryImpl.startScan(DiscoveryServiceRegistryImpl.java:260) ~[?:?]
        at org.eclipse.smarthome.io.rest.core.internal.discovery.DiscoveryResource.scan(DiscoveryResource.java:97) ~[?:?]
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
[... cut ...]
        at org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.nodeAdded(ZigBeeDiscoveryService.java:208)
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager$6.run(ZigBeeNetworkManager.java:1171)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```
I'm also getting the same error in the log from using a REST API call, which returns 500...
[image]

kaikreuzer, 2018-03-22:
As [mentioned](https://community.openhab.org/t/unit-support-knx-2-karaf-4-1-5-upgrade-and-more/41904/6), this change in ESH is source compatible, so no source code change is required for the Zigbee binding.

The [code is being built with openhab-bundles](https://openhab.ci.cloudbees.com/job/openHAB2-Bundles/lastSuccessfulBuild/artifact/bindings/org.openhab.binding.zigbee/target/), so when using a latest build result, I do not understand how this error could still happen...

openhab-5iver, 2018-03-22:
I'm using a manual install of OH and dropping the Zigbee binding into addons. I'll test a fresh install using this method later today to see if it is reproducible or if it is something with my production setup.

openhab-5iver, 2018-03-22:
I was able to reproduce this error when starting Zigbee discovery on a fresh manual install of 1232 with just the Zigbee binding jar in addons. I'm using an Ember dongle (HUSBZB-1). Other people have reported this in the forum too.

kaikreuzer, 2018-03-22:
> the Zigbee binding jar in addons

Where did you get that jar from? Why don't you use the official one that is part of the distro?

openhab-5iver, 2018-03-22:
https://openhab.ci.cloudbees.com/job/openHAB2-Bundles/lastSuccessfulBuild/org.openhab.binding$org.openhab.binding.zigbee/

I manually install/update the binding because it makes it much easier to swap in a test build of the jar.

cdjackson, 2018-03-27:
I'm not sure there's anything to do here so closing.

closed 2018-03-27 by cdjackson

openhab-5iver, 2018-03-30:
As mentioned in the forum, this error appears to only happen after a Zigbee device has been deleted. By resetting the device and turning it off before it is discovered, eliminates the error. I assume the error would also occur if a new unpaired device was introduced. I reset the two bulbs that were an issue before. Here is a log from the binding, starting with deleting the Thing of a working bulb, and then a discovery. [033018 bulb delete zigbee.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/1864367/033018.bulb.delete.zigbee.zip)

The same exception as above occurred at 2018-03-30 13:46:17.475 and then 2018-03-30 13:46:17.508. As of yet, there are no exceptions in the Karaf console, but I expect there will be.

cdjackson, 2018-03-30:
I'm really not sure what could cause this. There was no change to the binding (last change was well over a month ago), and I'm certainly not able to replicate this. I'm regularly deleting, and adding, and doing pretty much anything else you can think of, and I can't reproduce this.

I'm also not aware of any other reports from any other users? Maybe I'm missing that though, but the only other report I saw on the forum said it was resolved when the user updated their runtime.

I had a quick look at the log, but there's nothing obviously wrong - sorry - not sure what to do with this.

openhab-5iver, 2018-03-30:
I was able to replicate the issue using the 2.3.0 snapshot Debian amd64 [docker image](https://hub.docker.com/r/openhab/openhab/). I had a bulb that I had deleted the Thing for and all I had to do was start a discovery.

weakfl, 2018-04-02:
i've got the same problem. I tried to remove/add a battery powered device and I'm getting the same exception:

```
12:49:18.099 [WARN ] [.eclipse.jetty.servlet.ServletHandler] - 
javax.servlet.ServletException: org.glassfish.jersey.server.ContainerException: java.lang.IncompatibleClassChangeError: Expected static field org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.scheduler
    at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:489) [171:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
    at org.glassfish.jersey.servlet.WebComponent.service(WebComponent.java:427) [171:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
    at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:388) [171:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
    at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:341) [171:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
    at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:228) [171:org.glassfish.jersey.containers.jersey-container-servlet-core:2.22.2]
    at com.eclipsesource.jaxrs.publisher.internal.ServletContainerBridge.service(ServletContainerBridge.java:76) [15:com.eclipsesource.jaxrs.publisher:5.3.1.201602281253]
    at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:848) [88:org.eclipse.jetty.servlet:9.3.21.v20170918]
    at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:584) [88:org.eclipse.jetty.servlet:9.3.21.v20170918]
    at org.ops4j.pax.web.service.jetty.internal.HttpServiceServletHandler.doHandle(HttpServiceServletHandler.java:71) [187:org.ops4j.pax.web.pax-web-jetty:6.0.9]
    at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548) [85:org.eclipse.jetty.security:9.3.21.v20170918]
    at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:226) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1180) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.ops4j.pax.web.service.jetty.internal.HttpServiceContext.doHandle(HttpServiceContext.java:284) [187:org.ops4j.pax.web.pax-web-jetty:6.0.9]
    at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:512) [88:org.eclipse.jetty.servlet:9.3.21.v20170918]
    at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1112) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.ops4j.pax.web.service.jetty.internal.JettyServerHandlerCollection.handle(JettyServerHandlerCollection.java:80) [187:org.ops4j.pax.web.pax-web-jetty:6.0.9]
    at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:134) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.server.Server.handle(Server.java:534) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:333) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:251) [87:org.eclipse.jetty.server:9.3.21.v20170918]
    at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:283) [80:org.eclipse.jetty.io:9.3.21.v20170918]
    at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:108) [80:org.eclipse.jetty.io:9.3.21.v20170918]
    at org.eclipse.jetty.io.SelectChannelEndPoint$2.run(SelectChannelEndPoint.java:93) [80:org.eclipse.jetty.io:9.3.21.v20170918]
    at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.executeProduceConsume(ExecuteProduceConsume.java:303) [91:org.eclipse.jetty.util:9.3.21.v20170918]
    at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceConsume(ExecuteProduceConsume.java:148) [91:org.eclipse.jetty.util:9.3.21.v20170918]
    at org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run(ExecuteProduceConsume.java:136) [91:org.eclipse.jetty.util:9.3.21.v20170918]
    at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:671) [91:org.eclipse.jetty.util:9.3.21.v20170918]
    at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:589) [91:org.eclipse.jetty.util:9.3.21.v20170918]
    at java.lang.Thread.run(Thread.java:748) [?:?]
Caused by: org.glassfish.jersey.server.ContainerException: java.lang.IncompatibleClassChangeError: Expected static field org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService.scheduler
    at org.glassfish.jersey.servlet.internal.ResponseWriter.rethrow(ResponseWriter.java:278) ~[?:?]
    at org.glassfish.jersey.servlet.internal.ResponseWriter.failure(ResponseWriter.java:260) ~[?:?]
    at org.glassfish.jersey.server.ServerRuntime$Responder.process(ServerRuntime.java:509) ~[174:org.glassfish.jersey.core.jersey-server:2.22.2]
[... cut ...]
    at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:512) ~[?:?]
    at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185) ~[?:?]
    at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1112) ~[?:?]
    at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) ~[?:?]
    at org.ops4j.pax.web.service.jetty.internal.JettyServerHandlerCollection.handle(JettyServerHandlerCollection.java:80) ~[?:?]
    ... 13 more
```

cdjackson, 2018-04-02:
Which version of the binding and the runtime are you using? Is it the latest of everything?

weakfl, 2018-04-02:
> Which version of the binding and the runtime are you using? Is it the latest of everything?

Yep, just updated two hours ago to switch to the PaperUI version of another binding (innogy).

cdjackson, 2018-04-02:
Not sure I understand - do you mean you updated PaperUI? Or you updated the binding using PaperUI?

The bottom line is this shouldn’t be an issue (apparently!) if the latest ESH is used.

In any case, there’s not much I can do about this :(

weakfl, 2018-04-02:
I previously had the innogy binding installed manually. The binding is now available via PaperUI, so I updated everything through apt in order to switch from manual install to PaperUI.

Removing/adding the battery powered zigbee device is almost always necessary after an OH update, but this time I couldn't add it and got the exception.

cdjackson, 2018-04-02:
Ok, as per other discussions, the binding hasn’t changed - ESH changed, and the theory was that a recompile should resolve it as the issue is a difference in class definitions. At the moment, I can’t easily do anything on the master branch as I’m working on a heavily refactored version. I’ll try and get this merged in the next week or so and hopefully that will solve the issue as I don’t see the problem here.

weakfl, 2018-04-02:
> At the moment, I can’t easily do anything on the master branch as I’m working on a heavily refactored version.

Yea, I've been following the discussion :)

I also knew that the issue at hand is kind of in limbo atm, but @openhab-5iver was requesting feedback in the community thread, so I thought I'll chime in.

Thanks anyway Chris, I'm sure it will eventually get resolved. As always ;)

cdjackson, 2018-04-02:
It’s useful to know that others have the issue at least - but not really sure why assuming everything has been rebuilt (which appears to be the case). I guess that’s why they call them “breaking changes” :(