openhab/org.openhab.binding.zigbee #175: Client cluster channels not sending commands

weakfl, 2018-04-08:
On my 2-channel BJ control elements, the lower rocker is exposed as `switch_onoff` **client cluster** channel by the binding (second endpoint). When the rocker is pressed, the attached item receives an ON/OFF update, but no **command**.

That makes it impossible to distinguish button presses from update events triggered by reporting.

I've set up two test rules, one for received commands and one for updates, and the `received command` rule never fires:

```
rule "bj relay terrace channel 2 command"
when
    Item BJRelay1Channel2 received command
then
    logWarn("default.rules", "BJRelay1Channel2 received COMMAND is " + receivedCommand)
end
```

*"Primary"* channels are receiving commands, only client cluster channels seem to be problematic.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |   Telegesis                                                                             |
| openHAB version  | 2.3.0~20180408143530-1     |

cdjackson, 2018-04-08:
Sorry, but I'm not sure I understand the issue. It sounds to me like it's working as I would expect, so I guess I don't understand the use case (or I just don't understand ;) ).

> That makes it impossible to distinguish button presses from update events triggered by reporting.

Why do you need to know this? If the item updates - does it matter how it updated - either via a report command, or an onoff command?

> the received command rule never fires:

I don't think this would be expected? The binding doesn't generate commands - only state updates.

weakfl, 2018-04-08:
> Why do you need to know this? If the item updates - does it matter how it updated - either via a report command, or an onoff command?

It does make a difference for rules. If I define a `received command` rule, I can still change the state of the item with `postUpdate()` without triggering the rule. That can be quite useful depending on the use case.

> I don't think this would be expected? The binding doesn't generate commands - only state updates.

I didn't know that. I was expecting that the binding would differentiate between state updates caused by reporting and events triggered by the user.
It's kind of irrelevant for a bulb, but for something like a remote it does make a difference. Let's say I want to use the rocker to increase/decrease the level of multiple dimmer items through a rule. Each reporting update would change the dimmer levels.

cdjackson, 2018-04-08:
> I was expecting that the binding would differentiate between state updates caused by reporting and events triggered by the user.

I'm not sure how it would know this though? To the binding, it's the same - some devices report state updates using the reporting, and some use the commands. Commands are more versatile in that you can use them to trigger other devices directly.

> Let's say I want to use the rocker to increase/decrease the level of multiple dimmer items through a rule. Each reporting update would change the dimmer levels.

I don't know that I understand - if you are changing the rocker on a remote, then aren't you expecting it to change the dimmer levels? Sorry if my questions sound dumb, but I'm obviously missing something here... It sounds a little like you are trying to use the same device for two different functions? I doubt that remotes would support both ways of reporting to allow you to differentiate this? Or do you have a remote that does this?

If I have a motion sensor, it might send a report, or it might send a command when the state changes - in either case, we just want to update the state, and not send a command which can only be processed in a rule.

cdjackson, 2018-04-08:
Maybe once the static thing definitions are added we could look at adding specific support for devices like this. I’m still not sure that I understand your example about pressing the button three times, but let’s park this for now, and maybe a bit later we can look at this again and you can educate me on what I’m missing :)

weakfl, 2018-04-08:
Sounds good, thanks :)