openhab/org.openhab.binding.zigbee #176: Feature request: Allow configuration of end device bindings

AfromD, 2018-04-10:
I would like to propose the next big step for this binding: a functionality where actions on end points can be directly linked to actuators without the need for code to be executed on Openhab. In practical terms: A lamp should react to a switch, motion sensor or dimmer without the need for Openhab giving it instructions.

This would have the following benefits:

1. No need to write rules that pass on simple switch or dimmer status changes to end devices. This will lower the required technical knowledge to use the binding
2. Actions will execute even when Openhab can't be reached
3. Possibly faster interaction between components, less network traffic

Proposed solution:
The end user will see a list of compatible actuators in the configuration of each switchable or dimmable thing. The end user can then mark the appropriate actuator(s) to define a static control relationship between the two. Upon saving the configuration, Openhab would configure the end device to react to one or multiple actuators automatically.

Ideally, this would allow for many-to-many relationships, meaning that one switch or dimmer can control multiple end devices and that end devices can be controlled by multiple switches or dimmers.

This would transform the Openhab Zigbee functionality into more of a management interface and no longer make it the part that has to do all the work, which also is a single point of failure. Zigbee on Openhab would only be crucial for truely "smart" automation, where logical interactions have to be performed with the help of other systems and information outside of the zigbee network.

Of course this will only work independently of OH as long as there are other router capable devices in the Zigbee network.

cdjackson, 2018-04-10:
Fundamentally, you have a request for 2 features - the ability to add bindings, and support for multicast groups. Both of these features are planned, but are not high on the priority list at the moment.

cdjackson, 2021-06-21:
End device bindings should be possible using the console. You should be able to use the `bind` command to set up a binding so a device sends reports to any other device.

This issue also discussed multicast groups - the zigbee library now supports this (for Ember coordinators only), and again this can be configured through the console interface, but there is no way to send a command to a group at this point. Please feel free to raise a separate issue.

Since the end device binding is possible with the console, I'll close this.

closed 2021-06-21 by cdjackson