openhab/org.openhab.binding.zigbee #178: Support discovery of USB dongles

hsudbrock, 2018-04-10:
**Context**
With https://github.com/eclipse/smarthome/pull/5315, Eclipse SmartHome supports the discovery of USB dongles which provide a serial port. To participate in this discovery, bindings can provide an instance of `org.eclipse.smarthome.config.discovery.usbserial.UsbSerialDiscoveryParticipant`. This `UsbSerialDiscoveryParticipant` will be called for newly discovered USB dongles with serial port, getting information about the USB dongle, the serial port, and the USB interface attached to the serial port. The `UsbSerialDiscoveryParticipant` can then provide `DiscoveryResult`s for supported USB dongles.

**Issue**
The ZigBee binding should participate in this discovery, and provide `DiscoveryResult`s for ZigBee USB dongles.

hsudbrock, 2018-04-10:
I would like to provide a first PR for this issue, enabling the discovery of the USB dongles that are available to me:
* bitronVideo ZigBee dongle (Ember chip, "BV 2010/10"); USB vendor ID 10c4, USB product ID 8b34
* Qivicon ZigBee dongle (Telegesis chip); USB vendor ID 10c4, USB product ID 8293
* Qivicon ZigBee dongle (Telegesis chip); USB vendor ID 10c4, USB product ID 89fb

cdjackson, 2018-04-10:
Thanks Henning - I was hoping you were planning to add this :)

closed 2018-04-21 by tomhoefer

puzzle-star, 2018-11-24:
Hi @hsudbrock,

I have seen that the discovery of the USB devices, and naming the different dongles via udev, do not seem to interface very well (at least for me).

The discovery participant is returning the /dev/ttyUSBx (for instance) and using the discovered device name as representationProperty. But this name might change between reboots, so normally, when using different dongles, we symlink to it (i.e. /dev/telegesis-1 -> /dev/ttyUSB0) via udev rules.

With the USB discovery, if for some reason the devname changes between reboots, the dongle becomes unusable. On the other side, if after discovering it I manually change the serial port of the device to the persistent name symlink, it will keep on appearing as new device in the inbox (yes, I can hide it, but it is annoying and maybe misleading for current users).

Changing the representationProperty to the serial number (or maybe vip_pid_serial) might prevent it from appearing again in the inbox (not sure), but would still require to manually change the serial port to the symlink.

Would it make sense to change (this is ESH, not OH):

```
Path devicePath = Paths.get(devDirectory).resolve(serialPortName);`
```

To some thing like (**this is pseudocode**, just to check if this would make sense to you):

```
Path devicePath = resolveDevicePath(devDirectory, serialPortName);
...
Path resolveDevicePath(String devDirectory, String serialPortName) {
    Path devicePath = Paths.get(devDirectory).resolve(serialPortName);
    File devDir = new File(devDirectory);
    for (File dev : devDir.listFiles()) {
       Path devPath = dev.toPath();
       if (Files.isSymbolicLink(devPath) && Files.isSameFile(devPath, devicePath)) return devPath;
    }
    return devicePath;
}
```
I know it is not the most efficient, as it traverses the the full dev directory looking for symlinks to the discovered path, but allows to honour the different dongle naming persistence.

Pedro

cdjackson, 2018-11-24:
@puzzle-star maybe it's better to create a new issue since this is closed a long time now, and it will get lost.

puzzle-star, 2018-11-24:
Done :)