openhab/org.openhab.binding.zigbee #190: Deleted Things are rediscovered, even if powered off

openhab-5iver, 2018-04-21:
HUSBZB-1 (Ember)
manual install of OH 2.3.0 snapshot 1258
 16 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee
 17 │ Active   │  80 │ 2.3.0.201804202151     │ ZigBee Ember Binding
 18 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee.dongle.ember
 19 │ Active   │  80 │ 2.3.0.201804202151     │ ZigBee Binding

I've seen this with GE Link bulbs and Centralite 3400 keypad (which only has a temperature channel, last I looked).

Deleting a Thing does remove it, but rediscovering will find it again, even if the device is powered off. After deleting it from the inbox, it will still be rediscovered. I can provide a log, but I expect this is easily reproduced. I suppose this could be an OH or ESH issue, but I haven't noticed it with any other binding. I've seen this for a while, so this is not something recently introduced.

cdjackson, 2018-04-21:
It will be rediscovered unless it was removed from the network successfully. If it was powered on when it was deleted, then it will be removed from the network - if not, the system doesn’t know if it’s removed, so it retains knowledge of the device.

I could possibly look at adding some sort of force delete to remove it from the network, but what you report is correct functionality at the moment if it’s not removed properly.

openhab-5iver, 2018-04-21:
> If it was powered on when it was deleted, then it will be removed from the network

I'll try this some more. I originally deleted them while they were powered, but maybe I was not quick enough to power them off before the background discovery found them again.

> if not, the system doesn’t know if it’s removed, so it retains knowledge of the device.

So, if the device completely fails, there currently would not be a way to remove it? Force delete makes sense to me.

firasalmanna, 2018-09-05:
Hi @cdjackson 
Any update on the "Force delete" functionality? 
I have some battery powered device, they were discovered by the TI CC2538 ZNP but stayed offline after that. Now I can't even delete them as they reappeared every time I run the discovery.

cdjackson, 2018-09-05:
I will add a function to remove the node from the network cache if the thing is deleted - this should allow you to remove it.

triller-telekom, 2019-05-10:
This issue has been fixed by https://github.com/openhab/org.openhab.binding.zigbee/pull/250

@cdjackson Can you please close this issue?

However, there is still another issue: https://github.com/openhab/org.openhab.binding.zigbee/issues/437

openhab-5iver, 2019-05-10:
Thi is still and issue for me with the latest binding, but I will close it as a #437 is a duplicate or at least very similar to what I was reporting.

closed 2019-05-10 by openhab-5iver