openhab/org.openhab.binding.zigbee #192: Port already in use when restarting org.openhab.binding.zigbee.ember

TomTravaglino, 2018-04-27:
When the bridge (for a ZigBee USB dongle) is ONLINE, and the bundle is restarted a couple of times, the bridge will stay in status OFFLINE. In the log the ZigBeeSerialPort reports that the port is already in use.
The issue is not reliably reproducable. It happens after an undefined number of restarts. It was mostly observed when testing the Ember SPI implementation, but in several occasions also when using the Telegesis dongle.
For the sake of completeness I attached a small snippet of the logs which frankly doesn't contain really useful information.

[test-ember-spi-20180423_port_in_use.log](https://github.com/openhab/org.openhab.binding.zigbee/files/1953901/test-ember-spi-20180423_port_in_use.log)

cdjackson, 2018-04-27:
The issue shown in the log is not the problem - the problem is the closing of the port last time it was open.

I have discussed this a few months back with @tomhoefer . As far as I can tell this is an issue with RXTX - it doesn't seem to return from the close function. There are a lot of comments on the web if you search for this - I tried to implement some of them which often related to stopping the event listener before closing the port. I think this is implemented in the serial handler you are using, but anyway, it didn't help - or at least it didn't completely solve the issue.

I don't know what version of RXTX is used in your system - it seems that maybe newer versions of RXTX might fix the issue from what I've seen.

Can you find the base version of RXTX used in your system? There

cdjackson, 2018-08-01:
I propose to close this as I think the issue is related to your runtime - I've not seen any reports of this with openHAB.

tomhoefer, 2018-08-02:
yes - agree

closed 2018-08-02 by tomhoefer