openhab/org.openhab.binding.zigbee #202: Bridge stays in status ONLINE when the dongle is removed

TomTravaglino, 2018-05-08:
I created a bridge for Telegesis (later also for Ember) dongle and then removed the dongle. After a couple of minutes the bridge was still online. Is the framework/binding able to recognize that the controller is not reachable and then set the bridge status to offline?

TomTravaglino, 2018-06-06:
According to our further investigation and our offline discussion I would like to summarize:
- The SPI driver is able to recognize when the controller stops communicating but not that it has been unplugged. At this point it's good enough to just having the status beeing set to OFFLINE. Detecting that controller has been removed is an improvement which can follow later.
- The drivers for Telegesis and Ember dongle don't yet offer this feature. It needs to be implemented.

TomTravaglino, 2018-06-06:
In the SPI implementation the status of the bridge is passed to ZigBeeCoordinatorHandler.networkStateUpdated(ZigBeeTransportState). At the moment when ZigBeeTransportState is OFFLINE the status of the bridge will be updated as follows:
- status = ThingStatus.OFFLINE
- statusDetail = ThingStatusDetail.NONE
- description = ZigBeeBindingConstants.OFFLINE_STARTUP_FAIL

We would like to do the following changes:
- statusDetail = ThingStatusDetail.COMMUNICATION_ERROR
- remove description

TomTravaglino, 2018-06-07:
Along with this issue I experienced that when the bridge goes offline the things still stay online. I inspected the code and found the potential reason for this.
In _ZigBeeThingHandler.bridgeStatusChanged(ThingStatusInfo)_ there is no special handling for the case that the bridge status has changed to offline. In my opinion we need to call _updateStatus(ThingStatus.OFFLINE, ThingStatusDetail.BRIDGE_OFFLINE)_ in accordance to the implementation in _BaseThingHandler.bridgeStatusChanged(ThingStatusInfo)_

cdjackson, 2018-06-07:
Closed by #216

closed 2018-06-07 by cdjackson