openhab/org.openhab.binding.zigbee #209: Visonic MCT-340 door sensor

hansaya, 2018-05-17:
I bought few Visonic MCT-340 sensors hoping it would work with openhab. Turns out that openhab does detect the device but it comes up as unknown. I'm wondering how can I help to get the device added as a supported device? Is there any good documentation on how could I contribute?

cdjackson, 2018-05-17:
The binding creates an XML file in the {userdata}/zigbee folder - please provide this for starters.

hansaya, 2018-05-18:
Here is the output for the two of them
```
  <ZigBeeNode>
    <ieeeAddress>000D6F000B11E761</ieeeAddress>
    <networkAddress>39348</networkAddress>
    <nodeDescriptor>
      <apsFlags>0</apsFlags>
      <bufferSize>0</bufferSize>
      <complexDescriptorAvailable>false</complexDescriptorAvailable>
      <manufacturerCode>0</manufacturerCode>
      <logicalType>UNKNOWN</logicalType>
      <serverCapabilities/>
      <incomingTransferSize>0</incomingTransferSize>
      <outgoingTransferSize>0</outgoingTransferSize>
      <userDescriptorAvailable>false</userDescriptorAvailable>
      <frequencyBands/>
      <macCapabilities/>
      <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
      <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
      <stackCompliance>0</stackCompliance>
      <R21__BITMASK>65024</R21__BITMASK>
      <R21__BITSHIFT>9</R21__BITSHIFT>
    </nodeDescriptor>
    <powerDescriptor>
      <currentPowerMode>UNKNOWN</currentPowerMode>
      <availablePowerSources/>
      <currentPowerSource>UNKNOWN</currentPowerSource>
      <powerLevel>UNKNOWN</powerLevel>
    </powerDescriptor>
    <endpoints/>
    <bindingTable/>
  </ZigBeeNode>
  <ZigBeeNode>
    <ieeeAddress>000D6F000B3BE746</ieeeAddress>
    <networkAddress>60102</networkAddress>
    <nodeDescriptor>
      <apsFlags>0</apsFlags>
      <bufferSize>82</bufferSize>
      <complexDescriptorAvailable>false</complexDescriptorAvailable>
      <manufacturerCode>4113</manufacturerCode>
      <logicalType>END_DEVICE</logicalType>
      <serverCapabilities/>
      <incomingTransferSize>82</incomingTransferSize>
      <outgoingTransferSize>82</outgoingTransferSize>
      <userDescriptorAvailable>false</userDescriptorAvailable>
      <frequencyBands>
        <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
      </frequencyBands>
      <macCapabilities>
        <MacCapabilitiesType>REDUCED_FUNCTION_DEVICE</MacCapabilitiesType>
      </macCapabilities>
      <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
      <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
      <stackCompliance>0</stackCompliance>
      <R21__BITMASK>65024</R21__BITMASK>
      <R21__BITSHIFT>9</R21__BITSHIFT>
    </nodeDescriptor>
    <powerDescriptor>
      <currentPowerMode>UNKNOWN</currentPowerMode>
      <availablePowerSources/>
      <currentPowerSource>UNKNOWN</currentPowerSource>
      <powerLevel>UNKNOWN</powerLevel>
    </powerDescriptor>
    <endpoints/>
    <bindingTable/>
  </ZigBeeNode>
[zigbee-network--zigbee_coordinator_cc2531_33d2d2e2.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/2015502/zigbee-network--zigbee_coordinator_cc2531_33d2d2e2.zip)

```

cdjackson, 2018-05-18:
Neither of these contain enough information to have completed discovery. If you have a logfile from the discovery I can take a look at that, otherwise it might be a case of trying to get hold of a device for testing.

hansaya, 2018-05-18:
I'll look into the logs today. otherwise I'll buy one and send it to you

hansaya, 2018-05-22:
I deleted the discovery and let it redo the scan.

/var/log/openhab2/openhab.log
```
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2018-05-21 22:55:35.012 [ERROR] [531.network.packet.ZToolPacketStream] - Packet parsing failed due to exception.
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2018-05-21 22:55:38.056 [ERROR] [531.network.packet.ZToolPacketStream] - Packet parsing failed due to exception.
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2018-05-21 22:55:38.191 [ERROR] [531.network.packet.ZToolPacketStream] - Packet parsing failed due to exception.
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2018-05-21 22:55:38.385 [ERROR] [531.network.packet.ZToolPacketStream] - Packet parsing failed due to exception.
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2018-05-21 22:55:48.406 [ERROR] [531.network.packet.ZToolPacketStream] - Packet parsing failed due to exception.
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2018-05-21 22:56:03.876 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 000D6F000B11E761: Starting ZigBee device discovery
2018-05-21 22:56:03.908 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:33d2d2e2:000d6f000b11e761' to inbox.
2018-05-21 22:56:16.690 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 000D6F000B3BE746: Starting ZigBee device discovery
2018-05-21 22:58:03.375 [ERROR] [531.network.packet.ZToolPacketStream] - Packet parsing failed due to exception.
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:140) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107) [200:com.zsmartsystems.zigbee.dongle.cc2531:1.0.11]
        at java.lang.Thread.run(Thread.java:748) [?:?]
```

/var/log/openhab2/events.log

```
2018-05-21 22:55:38.495 [me.event.InboxRemovedEvent] - Discovery Result with UID 'zigbee:device:33d2d2e2:000d6f000b11e761' has been removed.
2018-05-21 22:56:03.913 [home.event.InboxAddedEvent] - Discovery Result with UID 'zigbee:device:33d2d2e2:000d6f000b11e761' has been added.
2018-05-21 22:56:09.582 [me.event.ThingUpdatedEvent] - Thing 'zigbee:device:33d2d2e2:7cb03eaa00a5c187' has been updated.
2018-05-21 22:56:23.830 [me.event.ThingUpdatedEvent] - Thing 'zigbee:device:33d2d2e2:7cb03eaa00a5c187' has been updated.
2018-05-21 22:56:24.241 [hingStatusInfoChangedEvent] - 'zigbee:device:33d2d2e2:7cb03eaa00a5c187' changed from OFFLINE: Node has not completed discovery to ONLINE
2018-05-21 22:56:24.266 [vent.ItemStateChangedEvent] - couch_color_temperature changed from NULL to 16
[... cut ...]
2018-05-21 22:56:24.290 [nt.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:33d2d2e2:7cb03eaa00a5c187 changed to UNKNOWN.
2018-05-21 22:58:28.363 [nt.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:33d2d2e2:00158d0001ab2cb6 changed to UNKNOWN.
2018-05-21 22:58:28.378 [nt.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:33d2d2e2:00158d0001fa5bdc changed to UNKNOWN.
2018-05-21 22:58:28.386 [nt.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:33d2d2e2:00158d000182e50d changed to UNKNOWN
```

cdjackson, 2018-05-22:
The checksum failures are a bit suspicious, but other than that I can't really say much about the log. I assume the CS failures are real, which might indicate some sort of communications error (not uncommon if your running in a VM or a slow computer).

hansaya, 2018-05-25:
@cdjackson I can send you a unit to play with. If you send me your address? hansaka.perera@gmail.com

cdjackson, 2018-06-16:
Using the Ember coordinator, it fundamentally seems to work ok (see images below). Removing/restoring the magnet updates the state immediately. The only issue I see at the moment is the battery voltage doesn't work - the device is returning a value that means the voltage is unknown, so this may just not work with this device.

I will see if I can test with a TI dongle shortly.

Inbox...
[image]

Things list...
[image]

Thing properties...
[image]

Control page...
[image]

cdjackson, 2018-06-16:
@hansaya The TI dongle isn't passing the ZDO MatchDescriptor requests to the stack. This prevents IAS devices joining the network and is the likely reason this device isn't working for you.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/326

hansaya, 2018-06-18:
Thank you for looking into this. I saw multiple discussion around ember devices and sounds like a solid solution for zigbee. I was thinking of getting https://www.digikey.com/product-detail/en/cel/ZM3588S-USB-LR/ZM3588S-USB-LR-ND/5864203 

Unless you have any other good recommendations for a dongle.

closed 2018-06-21 by hansaya

wtf911, 2018-11-29:
@cdjackson I was under the impression that this pull request would make the MCT-340 work with a CC2531:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/410

Was I wrong to think that?

I'm using Windows, a CC2531, and trying to connect my MCT-340 which isn't showing up in the inbox. I downloaded a fresh 2.4.0 snapshot build and clicked install on the zigbee binding.

cdjackson, 2018-11-29:
This isn't included in the binding yet. I will look at doing an update in the coming days.