openhab/org.openhab.binding.zigbee #211: GE Link bulbs not responding

openhab-5iver, 2018-05-20:
OH 2.3.0 snapshot 1281
233 │ Active   │  80 │ 2.3.0.201805172001     │ org.openhab.binding.zigbee
234 │ Active   │  80 │ 2.3.0.201805172001     │ org.openhab.binding.zigbee.cc2531
235 │ Active   │  80 │ 2.3.0.201805172001     │ org.openhab.binding.zigbee.ember
236 │ Active   │  80 │ 2.3.0.201805172001     │ org.openhab.binding.zigbee.telegesis
237 │ Active   │  80 │ 2.3.0.201805172001     │ org.openhab.binding.zigbee.xbee
198 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee
199 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee.dongle.cc2531
200 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee.dongle.ember
201 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee.dongle.telegesis
202 │ Active   │  80 │ 1.0.8                  │ com.zsmartsystems.zigbee.dongle.xbee
HUSBZB-1 (Ember)

I've had issues upgrading OH to the recent snapshots, so it's been a while since upgrading (previously on 1272). Build 1281 worked, but Zigbee is failing. The Ember coordinator comes online in Habmin, as does a Centralite Outlet (which is controllable), but my GE Link light bulbs stay offline or they come online but are not responsive, and the log seems to stall out. 
[image]

I also noticed that the Ember coordinator has a Flow Control configuration that is set to Hardware (RTS/CTS). Is this setting suitable for the HUSBZB-1, or should I be experimenting with it? This log has a number of restarts and trials in it.
[zigbee.log](https://github.com/openhab/org.openhab.binding.zigbee/files/2020312/zigbee.log)

Also, when stopping OH, I have to wait a LONG time for each device to go through this...
```
2018-05-20 07:20:26.276 [WARN ] [eclipse.smarthome.core.thing.internal.ThingManager] - Disposing handler for thing 'zigbee:device:9b51c72d:7ce524000013f335' takes more than 5000ms.
```
This is happening to all devices using the Zigbee, wifiled, zoneminder, and network bindings. If I restart the Zigbee binding (just the OH one), then Zwave devices do this too and the Zigbee coordinator never comes back online. I realize this may not be related to the binding and the bulbs not responding, but thought I'd mention it.

openhab-5iver, 2018-05-20:
I had included a debug log in previous post. BUT... after upgrading to... 

251 │ Active   │  80 │ 1.0.11                 │ com.zsmartsystems.zigbee.dongle.ember
252 │ Active   │  80 │ 1.0.11                 │ com.zsmartsystems.zigbee
253 │ Active   │  80 │ 2.3.0.201805172001     │ ZigBee Ember Binding
254 │ Active   │  80 │ 2.3.0.201805172001     │ ZigBee Binding

... and power cycling the bulbs, they are slowly coming up and responding!

I did see one ugly bit though...

```
2018-05-20 10:37:31.251 [DEBUG] [zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP dongle initialize done: Initialised false
2018-05-20 10:37:31.252 [DEBUG] [g.zigbee.internal.ZigBeeNetworkStateSerializerImpl] - Loading ZigBee network state: Start.
2018-05-20 10:37:31.255 [DEBUG] [s.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=7, notRdy=false]
2018-05-20 10:37:31.315 [DEBUG] [g.zigbee.internal.ZigBeeNetworkStateSerializerImpl] - Error loading ZigBee state 
com.thoughtworks.xstream.converters.ConversionException: Could not set field class com.zsmartsystems.zigbee.zdo.field.NodeDescriptor.R21_BITMASK : null : Could not set field class com.zsmartsystems.zigbee.zdo.field.NodeDescriptor.R21_BITMASK : null
---- Debugging information ----
message             : Could not set field class com.zsmartsystems.zigbee.zdo.field.NodeDescriptor.R21_BITMASK : null
cause-exception     : com.thoughtworks.xstream.converters.reflection.ObjectAccessException
cause-message       : Could not set field class com.zsmartsystems.zigbee.zdo.field.NodeDescriptor.R21_BITMASK : null
class               : com.zsmartsystems.zigbee.zdo.field.NodeDescriptor
required-type       : com.zsmartsystems.zigbee.zdo.field.NodeDescriptor
converter-type      : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
path                : /list/ZigBeeNode/nodeDescriptor/R21_BITMASK
line number         : 26
class[1]            : com.zsmartsystems.zigbee.dao.ZigBeeNodeDao
class[2]            : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
version             : 1.4.7
-------------------------------
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:79) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:474) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:406) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:257) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:134) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32) ~[?:?]
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1185) ~[?:?]
	at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1169) ~[?:?]
	at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1040) ~[?:?]
	at org.openhab.binding.zigbee.internal.ZigBeeNetworkStateSerializerImpl.deserialize(ZigBeeNetworkStateSerializerImpl.java:155) ~[?:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:278) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:345) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.lambda$1(ZigBeeCoordinatorHandler.java:424) ~[?:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:?]
[... cut ...]
	at com.thoughtworks.xstream.converters.reflection.SunUnsafeReflectionProvider.writeField(SunUnsafeReflectionProvider.java:56) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:437) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:257) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	... 34 more
```

cdjackson, 2018-05-20:
> I had included a debug log in previous post.

Ok, I saw that, but as it was in relation to the point about flow control, I didn't think it was related - sorry.

> I did see one ugly bit though...

Interesting. I'm not really sure why these two fields are serialised as they are constants (ie final) and this is probably what is causing the exception. I'll see if I can find a way to avoid this.

openhab-5iver, 2018-05-20:
I have only seen the exception once. All my bulbs are behaving now, so it looks like 1.0.11 resolved whatever the issue was!

closed 2018-05-20 by openhab-5iver