openhab/org.openhab.binding.zigbee #215: Devices leave network shortly after joining

tavalin, 2018-06-04:
Hi @cdjackson 

As per your comment in the OH forum I've raised an issue.

Devices: Philiips Hue Dimmer switch, Xiaomi Body Sensor (V2 motion sensor)

The issue is that the devices work for a period of time but then for some reason leave the network.  The log file attached shows the Philips Hue Dimmer leaving within half an hour of joining.

Unfortunately I don't have the required hardware setup to give you any sniffed packet logs.

[zigbee.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/2070193/zigbee.txt)

cdjackson, 2018-06-05:
Please you state what this log shows? There are a few devices - what is the Hue device and when does it leave? I don't see any problem in the log - everything looks fine from what I can tell and devices are responding to requests for the most part, so it would be good to know what the log is showing in case I missed it.

tavalin, 2018-06-05:
This section shows the Hue dimmer receiving a command and then leaving the network.
```
==> /var/log/openhab2/openhab.log <==

2018-06-04 23:19:18.464 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 52 58 3A 41 35 30 45 2C 30 31 30 34 2C 30 31 2C 30 31 2C 30 30 30 36 2C 30 33 3A 01 18 01 2C 2D 33 37 2C 46 46

2018-06-04 23:19:18.468 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=42254, profileId=260, destinationEp=1, sourceEp=1, clusterId=6, messageData=01 18 01, rssi=-55, lqi=255]

2018-06-04 23:19:18.472 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=42254/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, sequence=0, payload=01 18 01]

2018-06-04 23:19:18.475 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=24, commandId=1]

2018-06-04 23:19:18.479 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: OnCommand [On/Off: 42254/1 -> 0/1, cluster=0006, TID=18]

2018-06-04 23:19:18.484 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 0017880103E47DAF: ZigBee command receiveds OnCommand [On/Off: 42254/1 -> 0/1, cluster=0006, TID=18]

2018-06-04 23:19:18.489 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4E 4F 44 45 4C 45 46 54 3A 41 35 30 45 2C 30 30 31 37 38 38 30 31 30 33 45 34 37 44 41 46

2018-06-04 23:19:18.494 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDeviceLeftNetworkEvent [networkAddress=42254, ieeeAddress=0017880103E47DAF]

2018-06-04 23:19:18.493 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 0017880103E47DAF: Channel zigbee:device:04000BE3:0017880103e47daf:0017880103E47DAF_1_switch_onoff updated to ON

2018-06-04 23:19:18.497 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 0017880103E47DAF: nodeStatusUpdate - node status is DEVICE_LEFT, network address is 42254.

2018-06-04 23:19:18.501 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 0017880103E47DAF: Node 42254 is removed from the network

2018-06-04 23:19:18.504 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 0017880103E47DAF: Updating ZigBee channel state zigbee:device:04000BE3:0017880103e47daf:0017880103E47DAF_1_switch_onoff to ON

2018-06-04 23:19:18.573 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: Start.

==> /var/log/openhab2/events.log <==

2018-06-04 23:19:18.578 [me.event.ThingUpdatedEvent] - Thing 'zigbee:device:04000BE3:0017880103e47daf' has been updated.

2018-06-04 23:19:18.583 [hingStatusInfoChangedEvent] - 'zigbee:device:04000BE3:0017880103e47daf' changed from ONLINE to OFFLINE
```

cdjackson, 2018-06-05:
Ok, I’m not sure what I can do about this without knowing why the trust centre sent this notification. The binding has not sent anything to tell the device to leave - it has decided to leave by itself, right?

tavalin, 2018-06-05:
I don't know the ins and outs of the communication.  Ignorantly, from a user perspective, I pressed the ON button and the result is that the device left.  I don't know what/how that would be instigated.  I've also got a rather large log (can upload when I get home) for the Xiaomi sensor where it also leaves, but that doesn't seem to be related to any user input.

cdjackson, 2018-06-05:
What you see in the log, is what the binding does. Here, it did nothing. It didn’t tell the device to leave - it decided by itself for some reason. I don’t know why that is - I’m open to ideas if you have any, but at the moment I don’t see why the device would have decided to leave the network. Maybe this is a Hue specific issue - eg maybe Hue devices need a certain request within some time period or they leave the network. I’m not sure - if you have any ideas, then I’d welcome them as I think you can see here that the binding has not told the device to leave the network - it decided this itself.

nairolf86, 2019-01-28:
Hi I'm new here and first I want to say thank you for this great product. I really like it!
I think I have the exact same issue as tavalin reports and I also did some debug logging, perhaps it helps.

I too have a Philips Hue Dimmer Switch and at timecode 18:46:57.457 I pressed the "ON" Button. Then you can see the device (0017880104e4be39) leave the network. And at timecode 18:47:07.914 you can see the device again with status online, but it doesn't work any more. When I press the buttons nothing happens.

If I can do anything to help please tell me. Here attached the corresponding snippet from my openhab.log:

```
2019-01-28 18:46:56.699 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,17,08,19F5,220A4FBE58CB18EE
2019-01-28 18:46:56.699 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2019-01-28 18:46:56.700 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=17, power=8, panId=6645, epanId=220A4FBE58CB18EE, status=SUCCESS]
2019-01-28 18:46:57.455 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:RX:8F80,0104,01,01,0006,03:8,-75,FF
2019-01-28 18:46:57.456 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=36736, profileId=260, destinationEp=1, sourceEp=1, clusterId=6, messageData=01 38 01, rssi=-117, lqi=255]
2019-01-28 18:46:57.457 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 0017880104E4BE39: ZigBee command receiveds OnCommand [On/Off: 36736/1 -> 0/1, cluster=0006, TID=38]
2019-01-28 18:46:57.457 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 0017880104E4BE39: Channel zigbee:device:0400052C:0017880104e4be39:0017880104E4BE39_1_switch updated to ON
2019-01-28 18:46:57.458 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 0017880104E4BE39: Updating ZigBee channel state zigbee:device:0400052C:0017880104e4be39:0017880104E4BE39_1_switch to ON
2019-01-28 18:46:57.461 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:NODELEFT:8F80,0017880104E4BE39
2019-01-28 18:46:57.461 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDeviceLeftNetworkEvent [networkAddress=36736, ieeeAddress=0017880104E4BE39]
2019-01-28 18:46:57.470 [INFO ] [del.script.PhilipsSwitchKueche.rules] - Status: ON
2019-01-28 18:46:57.483 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 0017880104E4BE39: Node updated - ZigBeeNode [state=OFFLINE, IEEE=0017880104E4BE39, NWK=8F80, Type=END_DEVICE]
2019-01-28 18:46:57.486 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: Start.
2019-01-28 18:46:57.549 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: Done.
2019-01-28 18:46:57.550 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 0017880104E4BE39: Polling stopped
2019-01-28 18:46:57.551 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 0017880104E4BE39: Polling initialised at 1879415ms
2019-01-28 18:46:58.551 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-01-28 18:46:58.551 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
2019-01-28 18:46:58.552 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2019-01-28 18:46:58.683 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,17,08,19F5,220A4FBE58CB18EE
2019-01-28 18:46:58.683 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2019-01-28 18:46:58.684 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=17, power=8, panId=6645, epanId=220A4FBE58CB18EE, status=SUCCESS]
2019-01-28 18:46:59.684 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-01-28 18:46:59.684 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
2019-01-28 18:46:59.685 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2019-01-28 18:46:59.816 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,17,08,19F5,220A4FBE58CB18EE
2019-01-28 18:46:59.816 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2019-01-28 18:46:59.816 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=17, power=8, panId=6645, epanId=220A4FBE58CB18EE, status=SUCCESS]
2019-01-28 18:47:00.817 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-01-28 18:47:00.817 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
2019-01-28 18:47:00.818 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2019-01-28 18:47:00.948 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,17,08,19F5,220A4FBE58CB18EE
2019-01-28 18:47:00.949 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2019-01-28 18:47:00.949 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=17, power=8, panId=6645, epanId=220A4FBE58CB18EE, status=SUCCESS]
[... cut ...]
2019-01-28 18:47:09.621 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis send: TelegesisSendUnicastCommand [address=0, sourceEp=0, destEp=0, profileId=0, clusterId=49, messageData=00 00, messageId=null]
2019-01-28 18:47:09.624 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
2019-01-28 18:47:09.624 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=0, sourceEp=0, destEp=0, profileId=0, clusterId=49, messageData=00 00, messageId=null]
2019-01-28 18:47:09.625 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+SENDUCASTB:02,0000,00,00,0000,0031
```

cdjackson, 2019-08-11:
I will close this as there have been a lot of changes to the underlying framework in the 14 months since this was opened and the 7 months of inactivity indicates that this may no longer be an issue.

If there are still problems, please open a new issue.

closed 2019-08-11 by cdjackson