openhab/org.openhab.binding.zigbee #219: Make ZigBee console commands available in the binding

hsudbrock, 2018-06-11:
The ZigBee framework used by this binding (cf. https://github.com/zsmartsystems/com.zsmartsystems.zigbee) features a console application. This console application permits to access a ZigBee dongle via simple console commands (like, e.g., enabling/disabling join mode, listing devices in the network, getting infos about a device's endpoints, and so forth).

I think that it would be helpful (e.g., for diagnostics and for development) to make the console commands of the core framework available in the binding, so that console commands can be used to control the ZigBee dongle that is represented by a ZigBee binding bridge.

Technically, Eclipse SmartHome permits to add console commands by adding a component implementing the interface `org.eclipse.smarthome.io.console.extensions.ConsoleCommandExtension`. These console commands are then available, e.g., in the openhab Karaf console. Assume for now that there is only one ZigBee bridge thing available. One could then implement such a console command extension by simply forwarding commands to an instance of the core ZigBee framework console, where that instance is configured to use the ZigBee dongle that is also used by the bridge handler. (If there are multiple bridge things available, the command could be prepended with the bridge ID to disambiguate between bridges.)

The `ZigBeeConsole` in the core framework can be configured by providing an instance of the `ZigBeeNetworkManager` and an instance of the `ZigBeeTransportTransmit`, which are then used to access the dongle when executing console commands. To use the console from the ZigBee binding, the console could hence be configured by using the instances of these classes that are already used by the `ZigBeeCoordinatorHandler`. To make this possible, the `ZigBeeCoordinatorHandler` could be extended with getter methods for these two objects to be used when configuring the core framework console instance.

My suggestion would be to add an additional bundle for such a console extension in the ZigBee binding, and to add an additional Karaf feature for that console extension to openhab (containing this additional ZigBee binding bundle, and the bundle(s) containing the core framework console implementation). By doing so, openhab users would be able to install the ZigBee binding either without console, or with console.

This is only a rough outline of such a console command extension, with the goal to get first feedback before starting work on a PR (which might require a PR with some rather small changes in the ZigBee core framework as well, for simplifying the integration).

I would be happy about any feedback!

cdjackson, 2018-06-12:
Hi Henning.
I don't think you want to be looking at the```ZigBeeConsole``` class - this is an implementation class, and it was never intended to be used in the way you propose.

The console commands are each individual classes that implement ```ZigBeeConsoleCommand ```. They take a simple parameter array, and a reference to the ```ZigBeeNetworkManager``` that is applicable to the network. 

The core commands are in the bundle ```com.zsmartsystems.zigbee.console```. Extensions for dongle specific commands are each in their own bundle.

My expectation for re-use of these console commands is that there would be a simple command processor class in ESH that handled the ESH commands, including the network selection if there are multiple dongles. This would select the appropriate implementation command, and pass the parameters to that command - effectively the same as is done in the console [around this line](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.console.main/src/main/java/com/zsmartsystems/zigbee/console/main/ZigBeeConsole.java#L296). Most of this I believe is already performed by the command processor in ESH - so all you need to do is to take arg[0] to get the command name, then pass the complete args list into the class.

Does that make sense?  Please don't consider using the bundle ```com.zsmartsystems.zigbee.console.main``` - it's a mess and needs a rewrite. The commands are what you need.

As mentioned above, there is a core set of commands that relate to the network, but then other commands that are dongle specific (mostly only for Ember, but a few for Telegesis). This perform dongle specific functions and they are split out into separate bundles in the same way as the dongle drivers - this allows the user to include only the parts required for their dongle, or in fact to not have any console.

If you want to discuss on the phone, please don't hesitate to call me (Thomas has my number if you don't find it) - it might be easier than trying to describe this all on emails here ;)

hsudbrock, 2018-06-12:
This makes sense, thanks Chris!

There are still some private command classes in the `ZigBeeConsole` implementing the private `ConsoleCommand` interface - I guess that the goal is to replace those with implementations of `ZigBeeConsoleCommand` in the future, right?

This leaves one issue for me (before starting implementing and finding further issues then :) ): How to find the set of all available commands. I see several options here:

1. An OSGi way would be to make those command classes into `Component`s, which would be injected into the simple command processor class in ESH (using compile-time annotations in the command classes for generating the OSGI-INF folder at build time).
2. Use reflection to find all implementations of `ZigBeeConsoleCommand`.
3. Define constant sets in the core framework that contain all commands (one constant for the general commands, one for Ember commands, one for Telegesis commands, ...).

Options 1 and 2 have the disadvantage that one does not know if a command is only applicable for a particular dongle type (but that info could be added by extending `ZigBeeConsoleCommand` with a method providing that info).

Options 2 and 3 have the disadvantage that the constant sets must be maintained, and that it is not possible to dynamically add console commands in other bundles.

I, personally, would prefer option 1 or 3, with a slight preference for option 1 (as, at least from ESH perspective, it feels more natural to implement this in an easily extensible way). Do you have a preference here, or a suggestion for a better solution?

cdjackson, 2018-06-12:
> I guess that the goal is to replace those with implementations of ZigBeeConsoleCommand in the future, right?

Yes - that's correct. Please feel free to convert any that you want - I've not systematically gone through and done this, but when I go to modify something, I try and convert to the new system at that time.

> An OSGi way would be to make those command classes into Components, 

I am not familiar with the OSGi component - what are the annotations? So long as it doesn't drive too many OSGi dependencies, I don't mind, but since we have people using the framework who are not using OSGi, I would prefer not to end up with any "non-standard" dependency.

> Use reflection to find all implementations of ZigBeeConsoleCommand.
The issue that the applicability of a command to a dongle could in solved by only searching for the interface in appropriate bundles. Since the package is defined for each dongle, this is known. You'd just need a link from package to dongle name. Why do you have to maintain the constant sets with reflection - I would expect we can find all instances of an interface within a package - no maintenance required right? (probably I miss something here? ;) ).

> Define constant sets in the core framework that contain all commands

I agree this is probably least 'nice' - it's what's done in the current console for simplicity. Maintaining the list is pretty simple of course so it's probably not a major downside - it has one advantage in that you could decide not to provide all commands to ESH, so there is an element of configurability with this option (I'm not sure it's needed - probably it's fine to just provide all commands, but I'm not 100% sure).

hsudbrock, 2018-06-12:
> I am not familiar with the OSGi component - what are the annotations? So long as it doesn't drive too many OSGi dependencies, I don't mind, but since we have people using the framework who are not using OSGi, I would prefer not to end up with any "non-standard" dependency.

These are `@Component(...)` annotations (like used, e.g., [here](https://github.com/openhab/org.openhab.binding.zigbee/blob/master/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/ZigBeeHandlerFactory.java) in the ZigBee binding). They are used at build-time to generate the XML component descriptors in the OSGI-INF folder used by OSGi. I.e., the annotations will not be needed as dependency by people using the framework without OSGi; the dependency is only needed at build-time of the framework.

> The issue that the applicability of a command to a dongle could in solved by only searching for the interface in appropriate bundles. Since the package is defined for each dongle, this is known. [...]

I see your point that this could be done with package names; I am always somewhat wary by hardcoding package names in the application, but I think it should be fine for this purpose here. There's no maintenance required with this reflection-based approach, we probably just misunderstood each other here.

>  it has one advantage in that you could decide not to provide all commands to ESH, so there is an element of configurability with this option (I'm not sure it's needed - probably it's fine to just provide all commands, but I'm not 100% sure).

I agree that this configurability is probably not needed, and would, hence, no longer consider this "constant set" option.

Would it be ok for you if I start with the OSGi-variant? Or would you prefer the reflection-based variant (which would also be ok for me). We could then continue discussion based on an implementation in a PR (I write this assuming that it would be OK for you if I create a PR - if not please object :) ).

cdjackson, 2018-06-12:
> These are @Component(...) annotations 

My bad - I should have recognised that!

> Would it be ok for you if I start with the OSGi-variant? Or would you prefer the reflection-based variant (which would also be ok for me).

I think I would prefer the reflection option since these are changes that would otherwise need to be made to the framework, and this might impose some dependencies that might be a problem (eg some people use this for Android).

> I write this assuming that it would be OK for you if I create a PR - if not please object 

Of course - very welcome, thanks.

Also, just to note that I'm not against changing the ```ZigBeeConsoleCommand``` interface if you have suggestions for any improvement. It's something that I added after our meeting in Darmstadt, but if in the course of you implementing an ESH console you have suggestions for improvement, then please feel free to suggest them.

hsudbrock, 2018-06-13:
> I think I would prefer the reflection option since these are changes that would otherwise need to be made to the framework, and this might impose some dependencies that might be a problem (eg some people use this for Android).

OK, I will use the reflection-based variant then. (If this gets totally ugly - which I do not expect - we can still discuss changing this.)

cdjackson, 2018-06-13:
Sure thing - I'm open to looking at other options if this doesn't look so nice - just shout :)

hsudbrock, 2018-06-19:
While implementing this feature, it turned out that using reflection is actually not as easy as I thought. The reason for this appears to be that one cannot simply ask a class loader to list all classes within a package. 

There are libraries that permit to do this (like, e.g., https://github.com/lukehutch/fast-classpath-scanner or https://github.com/ronmamo/reflections), but I'd rather not add an additional library just for that. For scanning the class path oneself I found solutions based on iterating over all resources provided by a class loader, and manually finding class files among them - but these solutions depend on the class loader implementation, and I, hence, do not consider them suitable for an OSGi environment.

I will, hence, finish a first implementation that uses constant sets enumerating the commands, and we can then use the PR for discussing whether and how this should be improved.

comletMB, 2018-06-29:
I work my way into the library via the console and I have two questions:

1. How to use the command "endpoint ENDPOINT" correctly or what is "ENDPOINT"?

I tried integer and hex values, but I only get "Error executing command: Endpoint '3' is not found".

"nodes" prints the following:
> Network_____Addr_____IEEE Address_____________Device Type________EP___Profile
> 0_____________0000_____000D6F000FA85944_____COORDINATOR  
> 13090________3322_____7CB03EAA00AC26B2____ROUTER____________3____ZIGBEE_LIGHT_LINK

2. Is there documentation that describes the use of the console in more detail?

cdjackson, 2018-06-29:
Endpoint in the address/endpoint (so 1234/1 for example).

There is more information here -:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee <https://github.com/zsmartsystems/com.zsmartsystems.zigbee>

It’s a debug tool - don’t expect loads of documentation ;)

PR #224 (2018-08-01): Integrate ZigBee console into the ZigBee binding.
This PR integrates the console commands from the ZigBee core framework into the binding. Resolves #219.

The integration of the core framework console into the binding consists of three new bundles: One general console bundle, and two bundles containing Ember-specific and Telegesis-specific console commands, respectively. Multiple bundles are used so that the general console bundle can be installed even if the binding bundles providing Ember and Telegesis support, respectively, are not available.

The general console bundle contains a console c [... cut]
closed 2018-08-01 by cdjackson