openhab/org.openhab.binding.zigbee #222: Network state file not deleted when bridge is removed

TomTravaglino, 2018-06-18:
When you delete the bridge the network status file is not removed. When you then create a new bridge it will get a different thing uid and as a consequence a new file will be created. So the previous file will be kept forever. I order to keep the directory clean we would like to introduce a file deletion when the bridge is removed.

The first idea was
- In the drivers (zsmartsystems) we add a method 'delete()' to the interface ZigBeeNetworkStateSerializer.
- In ZigBeeCoordinatorHandler.dispose() after shutting down the network we check if the thing status is "REMOVED" and then delete the file.

Unfortunately when dispose() is called the bridge status is not "REMOVED" but "UNINITIALIZED". So this doesn't seem to be realizable.

The second approach
- In the drivers (zsmartsystems) we add a method 'delete()' to the interface ZigBeeNetworkStateSerializer.
- In ZigBeeCoordinatorHandler we override BaseThingHandler.handleRemoval(). Here we set a flag that a bridge removal is in progress and call super.handleRemoval().
- In ZigBeeCoordinatorHandler.dispose() after shutting down the network we will use the flag and then delete the file  

Is there a better way to delete the file when you remove the bridge?

cdjackson, 2018-06-18:
Why do you need to introduce a method in the library? The writing of the files is completely handled by the binding. So you add a method that will then need to call back to the binding which doesn't really make any sense.

Just deleting the file should be all that is needed shouldn't it?

TomTravaglino, 2018-06-18:
For clarification: in the library there is the interface com.zsmartsystems.zigbee.ZigBeeNetworkStateSerializer. The change there would be just about adding a new method to it. No implementation is planned there.

The interface could look like this:
```
public interface ZigBeeNetworkStateSerializer {
    public void serialize(final ZigBeeNetworkManager networkManager);
    public void deserialize(final ZigBeeNetworkManager networkManager);
    public void delete();
}
```

The implementation will then happen in the binding in org.openhab.binding.zigbee.internal.ZigBeeNetworkStateSerializerImpl.

cdjackson, 2018-06-18:
I know the interface, but my question remains. I'm not sure why you need to add this - it will not be called from the library so there's no need to add this to the public interface is there?

TomTravaglino, 2018-06-18:
Of course we could simply add it to org.openhab.binding.zigbee.internal.ZigBeeNetworkStateSerializerImpl and use the concrete class  to access the new method. This is not a problem but it makes the code somehow "ugly". That was the main motivation to also have the method in the interface.

cdjackson, 2018-06-18:
I'm not sure it's ugly is it? It's quite normal for implementations to implement more methods. I would think that unless the framework is going to use the method then there's no need to define this - otherwise we are forcing implementations to add methods that will never be called which seems strange to me.

TomTravaglino, 2018-06-18:
OK, then we can use the concrete class.

What do you think about the approaches to handle the removal?

cdjackson, 2018-06-18:
I don't see any issue with the proposal and agree that it's a useful addition.

PR #223 (2018-06-21): Delete network state file on bridge removal
fixes #222
closed 2018-06-21 by hsudbrock