openhab/org.openhab.binding.zigbee #226: Zipato BULB2 ZigBee Edition support

vifino, 2018-07-21:
Hi!

After getting a QIVICON dongle to work on OpenHABian running snapshots (`openHAB 2.4.0~20180716173031-1`) -
(A matter of adding this to `rc.local`:
```sh
# Bind QIVICON stick.
sudo modprobe cp210x
echo 10c4 89fb > /sys/bus/usb-serial/drivers/cp210x/new_id
```
) - I continued trying to get the ZigBee-edition Zipato BULB2 RGBW bulbs to work.

Autodiscovery does discover it, but as an unknown device with only a Dimmer channel, which doesn't seem to work.

There is no error in the log, it does connect fine, but commands to set the brightness just don't seem to work. It goes online, I send commands, no effect.

I checked whether it'd be the ZigBee connection/dongle, but other devices, such as OSRAM Smart+ Plugs work beautifully.

Let me know if you need anything from me.

cdjackson, 2018-07-21:
According to the Zipato website, these operate at the lower ZigBee frequency. I guess maybe Zipato have a special dongle for this as I'm not familiar with any dongle providing support in this band.

[image]

vifino, 2018-07-21:
Huh, interesting. That brings up even more questions. Why does it do the initial setup fine if it operates on an entirely different band?

cdjackson, 2018-07-21:
Hmmm - maybe the data sheet is wrong of course! I missed the fact it was joining the network ok and may have been blind sided by the data sheet.

Can you provide a full debug log of the initialisation so I can see what it's showing. Alternatively, or additionally, there's an xml file saved in {userdata}/zigbee that should have all the device information in it - can you paste that here please?

vifino, 2018-07-21:
I uploaded the one file in the zigbee folder, with one OSRAM Smart+ Plug and the BULB2 registered.

http://0x0.st/sVBS.xml

I'll try to get the full debug log if that's not enough, but the xml was pretty easy to get.

cdjackson, 2018-07-21:
Something is wrong with the file - it appears to be an invalid http address...

If you can correct this, then don't worry about the log for now - I'll take a look at the XML first.

vifino, 2018-07-21:
I wrote down the address wrong, it was an 8, not a B. http://0x0.st/sV8S.xml

cdjackson, 2018-07-21:
Thanks - that's fine. One last question - what is the address (either long or short) of the device?

This one? -:

```
    <ieeeAddress>00158D0001783BE2</ieeeAddress>
    <networkAddress>26735</networkAddress>
```

vifino, 2018-07-21:
Yes, that's the one!

Thank you for taking a look at this, I appreciate it!

vifino, 2018-07-21:
I've been using OpenHAB for all of two days, could you guide me how to extract the logs you want properly? :)

cdjackson, 2018-07-22:
You can find general logging info here (https://www.openhab.org/docs/administration/logging.html#logging-in-openhab <https://www.openhab.org/docs/administration/logging.html#logging-in-openhab>) and specific ZigBee info here (https://www.openhab.org/addons/bindings/zigbee/#when-things-don-t-appear-to-be-working <https://www.openhab.org/addons/bindings/zigbee/#when-things-don-t-appear-to-be-working>).

cdjackson, 2018-12-01:
It's been a few months since any activity on this - if it's still an issue, then please reopen and provide the logs as above.

closed 2018-12-01 by cdjackson