openhab/org.openhab.binding.zigbee #233: ZigBee bridge configuration change results in non-working things

hsudbrock, 2018-08-09:
*Context*
After certain configuration updates of a coordinator handler, the handler is reinitialized (update of baud rate, flow control, or port, see `ZigBeeCoordinatorHandler#handleConfigurationUpdate`). Reinitialization is done via a call to `dispose` followed by a call to `initialize`.

*Issue*
After the reinitialization, previously paired things are no longer usable, in the sense that commands sent to a device via the channel converters do not reach the device.

*First analysis*
The `dispose` call to the coordinator handler will do a shutdown of the `ZigBeeNetworkManager`; the `initialize` call will then create a new `ZigBeeNetworkManager`.

The converters, however, still use classes using the old network manager (e.g., a `ZclOnOffCluster` instance obtained from the old network manager). This results in exceptions like the following when using the converters:
```
00124B00028A1B7D: Exception sending command to channel zigbee:device:0138143B:00124b00028a1b7d:00124B00028A1B7D_1_switch_onoff
	at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2047)
	at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:823)
	at java.util.concurrent.ScheduledThreadPoolExecutor.delayedExecute(ScheduledThreadPoolExecutor.java:326)
	at java.util.concurrent.ScheduledThreadPoolExecutor.schedule(ScheduledThreadPoolExecutor.java:533)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.scheduleTask(ZigBeeNetworkManager.java:499)
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.sendTransaction(ZigBeeTransaction.java:69)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.unicast(ZigBeeNetworkManager.java:902)
	at com.zsmartsystems.zigbee.zcl.ZclCluster.send(ZclCluster.java:159)
	at com.zsmartsystems.zigbee.zcl.clusters.ZclOnOffCluster.offCommand(ZclOnOffCluster.java:344)
	at org.openhab.binding.zigbee.internal.converter.ZigBeeConverterSwitchOnoff.handleCommand(ZigBeeConverterSwitchOnoff.java:143)
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$3.run(ZigBeeThingHandler.java:434)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:748)
```

cdjackson, 2018-10-08:
From my work over the weekend, I think this is actually quite a big issue as currently the binding will restart the coordinator link whenever anything is changed, which presumably leads to things stopping (although I have never seen such exceptions).

#271 will help this as it will now only restart the bridge when something necessary for the dongle configuration changes, rather than every time the configuration is updated.

@hsudbrock were you planning to look at this issue?

hsudbrock, 2018-10-08:
@cdjackson I agree that this issue is bad (at least in situations where config properties are changed that currently cause a coordinator restart). I don't have any concrete plans for looking at this issue within the next couple of weeks, as there are currently other things higher on my list. I might get to this at a later point (but feel free to have a look if you want to).

hsudbrock, 2018-10-08:
Thanks for the referenced PR - I will review soon.

cdjackson, 2019-08-10:
Closed by #273

closed 2019-08-10 by cdjackson