openhab/org.openhab.binding.zigbee #234: Remote Switches with 4 Buttons are not detected correctly. BitronHome Remote and OsramLighty4x Switch

JochenPfeifferHD, 2018-08-10:
### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |     Qivicon Zigbee Stick registered as  Telegesis Coordinator                   |
| openHAB version  |  2.3.0  |
| Hardware         |     Raspberry Zero                                                                     |
| Memory           |                                                                                |
| Java version     |                                                                                |
| Devices          | BitronHome Remote - 00124B0008B75A25  |
|                        |  OSRAMSwitch4xEULIGHTIFY  -    000D6F000FE48EDF                          |

Hi
I have a generic issue with remotes it seems. I have two remotes - both have 4 buttons and for both only 2 buttons works.
For the osram switch the Button 1 sends on the button 3 send off, openhab detect is as 6 switch. (It offers 8 functions based on short and long pressed buttons)
For the Bitron Remote Button 2 sends on, button 3 sends off, openhab detect is a 1 switch.

I goggled it already, and it seems that other libaries have the same issue. (https://github.com/Koenkk/zigbee2mqtt/issues/159)

I had a look into the console and enabled the logging. It logs only for the working buttons. No logs appears for the other buttons.

I would really like to used both buttons as I have already a couple of functionality in my home.

If I can help you be setting up some debug environment feel free to contact my.

cdjackson, 2018-08-10:
This looks like a duplicate of #37 . At the moment this is not being worked on, but if you have any information that can help, please add it there.

If it's not a duplicate, then please feel free to reopen, but please add more information on the specifics.

closed 2018-08-10 by cdjackson

JochenPfeifferHD, 2018-08-10:
I think the remotes are basicly working - but not all Buttons are working correctly. So it is more an Interpretation Issue if this is a bug or an improvement.
Do you know if any work ins planned in this area?

cdjackson, 2018-08-10:
Yes, it is planned, just not too high on my list at the moment.  I have a couple of remotes - it’s “just” about time.

Flunkyball, 2018-10-23:
Is there anyway to support you ? I have a tradfri remote over here and I could have a look at the code if you point me to the right direction....

hsudbrock, 2018-10-23:
@Flunkyball Trying to streamline discussion on ZigBee remotes integration, I suggest to use https://github.com/openhab/org.openhab.binding.zigbee/issues/37, as @cdjackson closed this  issue as duplicate of the other issue.