openhab/org.openhab.binding.zigbee #236: ST "Multi" (door) sensors don't report contact state

electric-monk, 2018-08-20:
This issue looks pretty much the same as #110, however I've tried the latest snapshot in openhabian and it still appears to be occurring.

Basically, the SmartThings "Multipurpose Sensor", which is just a door/window sensor with their usual temperature/battery reporting, is only doing the temperature/battery bits. A screenshot like the above:

[image]

I will note that this new SmartThing contact sensors do not look like my other ones (which are currently at another property), it appears they've changed their hardware, which I imagine is why the firmware has also mysteriously changed. The unit says it's made by Samjin Co. Ltd, and has quite a different shape.

cdjackson, 2018-08-20:
Please can you provide a link to the device you have, and also provide the XML file that is in your {userdata}/zigbee folder.

electric-monk, 2018-08-20:
This is a link to SmartThing's site: https://www.smartthings.com/products/smartthings-multipurpose-sensor

Note that it doesn't look like the "old style" ([still pictured on Amazon](https://user-images.githubusercontent.com/1221667/44352164-744bc480-a458-11e8-8b81-174b198e3b4a.jpg))

Here's the XML file:
[zigbee-network--zigbee_coordinator_xbee_44e0284c.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/2303084/zigbee-network--zigbee_coordinator_xbee_44e0284c.txt)

cdjackson, 2018-08-20:
Interestingly, this thing is reporting as a water sensor!

The two pictures you linked look the same to me, but it probably doesn’t matter. I was just thinking about buying one of the new ones for debugging…

For my info…

Manufacturer: Samjin
Model: multi

ZoneType: 42

cdjackson, 2018-08-21:
Water sensor support has now been added, so hopefully this device will now work with the latest binding (I’ve just triggered a build).

electric-monk, 2018-08-23:
Apologies I haven't been able to try this yet, when I install the latest snapshot the Zigbee binding disappears completely from my things! Can't reconfigure an existing instance or add a new one (though it shows as installed). Tried a clean re-install of openhabian and it still occurs. I'll retry reproducing this issue later.

electric-monk, 2018-08-30:
Confirmed that other than the oddness of having to pick "water sensor" for the door reed relay, it works perfectly :)

closed 2018-08-30 by electric-monk