openhab/org.openhab.binding.zigbee #243: Feature Request: Radio Tx Power - HUSBZB-1 range is terrible

jkl1337, 2018-08-27:
I was using a HUSBZB-1 with another software program and had serviceable results but noticed after forming a network with openhab that the range deteriorated to a ridiculous amount (can't hit a node 20 feet away). I found that if I formed the network on the stick with the alternate program before adding it to openhab it worked better. Turns out that the other software was forming the network with a configured txpower of 8 (dBm) and the openhab stack just leaves it at 0.

I can of course initialize the stick outside of openhab, but it would be nice if I could at least configure the txpower. Obviously it would be nicer if the stack picked a model specific reasonable default.

cdjackson, 2018-08-27:
Once the next update of the libraries is merged I’ll provide an option to configure to boost mode.

jkl1337, 2018-08-27:
Thanks. Just to clarify, I was specifically referring to the TX power that is set as a network parameter during network initialization (formation). There is also an the configuration option TX_POWER_MODE that allows setting a "boost" mode.  It sounds like both need to be set to get optimal TX and RX performance (https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2012/06/28/how_do_i_enable_boos-EP6A), since according to this boost mode configuration only raises the network tx power by 0.5dBm and is mainly a tweak on receiver sensitivity. The whole thing is still a bit unclear to me.

cdjackson, 2018-08-28:
Thanks for the reference. I don’t think it needs to be set during initialisation - there is a specific function to set the power (setRadioPower). I think I’ll just provide the ability to set the boost mode (defaulting to on) and provide a list of power settings people can try. There doesn’t seem to be a way to get the maximum power that is supported by the device, but it will return an error if you try and set it too high…..

cdjackson, 2018-08-28:
Setting power to 8 works ok, anything higher returns an error - even on the LRS device I have that should work up to 20dBm.

PR #252 (2018-09-17): Provide option in Ember to use boost mode (and make default) and set tx power
Provides two options in Ember -:

* Boost most, which improves the receive performance at the expense of power consumption. Given the coordinator will nearly always be plugged into a mains powered system, this is the default.
* TX Power - allows the transmit power to be set up to 8dBm to improve transmit performance.

Closes #243
closed 2018-09-17 by cdjackson