openhab/org.openhab.binding.zigbee #245: CentraLite 3200-de smartplug 4 vs 2 channels on thing

triller-telekom, 2018-08-31:
I have a strange behaviour with the CentraLite 3200-de smartplug. I have it paired to my OH installation and on the initial pairing it had 4 channels (see below).

I turned off my laptop running OH and started OH on the next day and I only had 2 channels on the thing (see below). I thought this was a one time issue, but this morning I had the very same problem, so I recorded some log information to file this bug.

Only 2 channels:

```
Total Active Power
zigbee:device:01381368:000d6f000bbd2b5b:000D6F000BBD2B5B_1_activepower
Number
```

```
Switch
zigbee:device:01381368:000d6f000bbd2b5b:000D6F000BBD2B5B_1_switch
Switch
```

```
smarthome zigbee node 16880

IEEE Address     : 000D6F000BBD2B5B
Network Address  : 16880
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=82, complexDescriptorAvailable=false, manufacturerCode=4174, logicalType=ROUTER, serverCapabilities=[], incomingTransferSize=82, outgoingTransferSize=82, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[MAINS_POWER, FULL_FUNCTION_DEVICE, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=0]
Power Descriptor : RECEIVER_ON_IDLE, [MAINS], MAINS, FULL
Associations     : []
Endpoints        : 
            1    : Profile 0104 ZIGBEE_HOME_AUTOMATION
                   -> BASIC
                   -> IDENTIFY
                   -> GROUPS
                   -> ELECTRICAL_MEASUREMENT
                   -> SCENES
                   -> DIAGNOSTICS
                   -> ON_OFF
                   <- OTA_UPGRADE
Neighbors:
NeighborTable [extendedPanId=B8D540A8597FB61C, extendedAddress=000D6F000DCFA28A, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=RX_ON, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=0, lqi=255]
Routes:
RoutingTable [destinationAddress=23081, status=DISCOVERY_UNDERWAY, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0]
RoutingTable [destinationAddress=0, status=ACTIVE, memoryConstrained=true, manyToOne=true, routeRecordRequired=false, nextHopAddress=0]
RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0]
```

```
smarthome zigbee endpoint 16880/1

IEEE Address     : 000D6F000BBD2B5B
Network Address  : 16880
Endpoint         : 1
Device Profile   : ZIGBEE_HOME_AUTOMATION (0x0104)
Device Version   : 0
Input Clusters   : (Server)
   0000 Basic
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCLVersion                               Wed Aug 29 14:38:00 CEST 2018 1
        S       1 r-- UNSIGNED_8_BIT_INTEGER    ApplicationVersion                        
        S       2 r-- UNSIGNED_8_BIT_INTEGER    StackVersion                             Wed Aug 29 14:38:00 CEST 2018 2
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HWVersion                                Wed Aug 29 14:37:59 CEST 2018 3
        S       4 r-- CHARACTER_STRING          ManufacturerName                         Wed Aug 29 14:37:59 CEST 2018 CentraLite
        S       5 r-- CHARACTER_STRING          ModelIdentifier                          Wed Aug 29 14:37:59 CEST 2018 3200-de
        S       6 r-- CHARACTER_STRING          DateCode                                 Wed Aug 29 14:38:00 CEST 2018 ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
        S       7 r-- ENUMERATION_8_BIT         PowerSource                               
        S      16 rw- CHARACTER_STRING          LocationDescription                       
        S      17 rw- ENUMERATION_8_BIT         PhysicalEnvironment                       
        S      18 rw- BOOLEAN                   DeviceEnabled                             
        S      19 rw- BITMAP_8_BIT              AlarmMask                                 
        S      20 rw- BITMAP_8_BIT              DisableLocalConfig                        
        S   16384 r-- CHARACTER_STRING          SWBuildID                                 
   0003 Identify
        S       0 rw- UNSIGNED_16_BIT_INTEGER   IdentifyTime                              
[... 197 lines cut ...]
        S     275 r-- UNSIGNED_16_BIT_INTEGER   APSFCFailure                              
        S     276 r-- UNSIGNED_16_BIT_INTEGER   APSUnauthorizedKey                        
        S     277 r-- UNSIGNED_16_BIT_INTEGER   NWKDecryptFailures                        
        S     278 r-- UNSIGNED_16_BIT_INTEGER   APSDecryptFailures                        
        S     279 r-- UNSIGNED_16_BIT_INTEGER   PacketBufferAllocateFailures              
        S     280 r-- UNSIGNED_16_BIT_INTEGER   RelayedUcast                              
        S     281 r-- UNSIGNED_16_BIT_INTEGER   PhytoMACqueuelimitreached                 
        S     282 r-- UNSIGNED_16_BIT_INTEGER   PacketValidatedropcount                   
        S     283 r-- UNSIGNED_16_BIT_INTEGER   AverageMACRetryPerAPSMessageSent          
        S     284 r-- UNSIGNED_8_BIT_INTEGER    LastMessageLQI                            
        S     285 r-- SIGNED_8_BIT_INTEGER      LastMessageRSSI                           
Output Clusters  : (Client)
   0019 OTA Upgrade
        S       0 r-- IEEE_ADDRESS              UpgradeServerID                           
        S       1 r-- UNSIGNED_32_BIT_INTEGER   FileOffset                                
        S       2 r-- UNSIGNED_32_BIT_INTEGER   CurrentFileVersion                       Wed Aug 29 14:38:00 CEST 2018 805589776
        S       3 r-- UNSIGNED_16_BIT_INTEGER   CurrentZigBeeStackVersion                 
        S       4 r-- UNSIGNED_32_BIT_INTEGER   DownloadedFileVersion                     
        S       5 r-- UNSIGNED_16_BIT_INTEGER   DownloadedZigBeeStackVersion              
        S       6 r-- UNSIGNED_8_BIT_INTEGER    ImageUpgradeStatus                        
        S       7 r-- UNSIGNED_16_BIT_INTEGER   ManufacturerID                            
        S       8 r-- UNSIGNED_16_BIT_INTEGER   ImageTypeID                               
        S       9 r-- UNSIGNED_16_BIT_INTEGER   MinimumBlockRequestDelay                  
        S      10 r-- UNSIGNED_32_BIT_INTEGER   ImageStamp
```

cdjackson, 2018-08-31:
Do you have a debug log file available showing the binding starting up and the channels being generated?

triller-telekom, 2018-08-31:
2 channels:

```
2018-08-31 09:40:16.034 [DEBUG] [b.z.h.ZigBeeCoordinatorHandler:432  ] - ZigBee network starting
2018-08-31 09:40:16.034 [DEBUG] [b.z.h.ZigBeeCoordinatorHandler:330  ] - Initialising ZigBee coordinator
2018-08-31 09:40:16.063 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'lang_en_GB.sitemap'
2018-08-31 09:40:16.069 [DEBUG] [b.z.h.ZigBeeCoordinatorHandler:346  ] - Key initialise 786E03CC0A368F5615CB3C534DF586C0
2018-08-31 09:40:16.072 [DEBUG] [o.b.z.handler.ZigBeeSerialPort:122  ] - Connecting to serial port [/dev/ttyUSB0] at 57600 baud, flow control FLOWCONTROL_OUT_XONOFF.
2018-08-31 09:40:16.103 [DEBUG] [o.b.z.handler.ZigBeeSerialPort:149  ] - Serial port [/dev/ttyUSB0] is initialized.
2018-08-31 09:40:16.109 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'demo.sitemap'
2018-08-31 09:40:16.142 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'demo.things'
2018-08-31 09:40:16.386 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'rrd4j.persist'
2018-08-31 09:40:16.415 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'demo.items'
2018-08-31 09:40:16.417 [DEBUG] [.e.s.m.i.i.GenericItemProvider:188  ] - Read items from model 'demo.items'
2018-08-31 09:40:16.417 [DEBUG] [.e.s.m.i.i.GenericItemProvider:210  ] - Processing binding configs for items from model 'demo.items'
2018-08-31 09:40:16.426 [DEBUG] [model.script.ScriptServiceUtil:66   ] - ScriptServiceUtil started
2018-08-31 09:40:16.911 [DEBUG] [e.s.m.s.r.i.e.ScriptEngineImpl:66   ] - Registered 'script' configuration parser
2018-08-31 09:40:16.955 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 841826000001077E: Channel zigbee:device:01381368:841826000001077e:841826000001077E_3_colortemperature linked - polling started.
2018-08-31 09:40:16.961 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 000D6F000BBD2B5B: Channel zigbee:device:01381368:000d6f000bbd2b5b:000D6F000BBD2B5B_1_current linked - polling started.
2018-08-31 09:40:16.962 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 841826000001077E: Channel zigbee:device:01381368:841826000001077e:841826000001077E_3_color linked - polling started.
2018-08-31 09:40:16.964 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 841826000001077E: Channel zigbee:device:01381368:841826000001077e:841826000001077E_3_switch linked - polling started.
2018-08-31 09:40:16.968 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 841826000001077E: Channel zigbee:device:01381368:841826000001077e:841826000001077E_3_dimmer linked - polling started.
2018-08-31 09:40:16.970 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 000D6F000BBD2B5B: Channel zigbee:device:01381368:000d6f000bbd2b5b:000D6F000BBD2B5B_1_switch linked - polling started.
2018-08-31 09:40:16.970 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 000D6F000BBD2B5B: Channel zigbee:device:01381368:000d6f000bbd2b5b:000D6F000BBD2B5B_1_voltage linked - polling started.
2018-08-31 09:40:16.971 [DEBUG] [b.z.handler.ZigBeeThingHandler:378  ] - 000D6F000BBD2B5B: Channel zigbee:device:01381368:000d6f000bbd2b5b:000D6F000BBD2B5B_1_activepower linked - polling started.
2018-08-31 09:40:17.033 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'demo.script'
2018-08-31 09:40:17.121 [DEBUG] [s.m.r.r.i.RuleRuntimeActivator:39   ] - Registered 'rule' configuration parser
2018-08-31 09:40:17.155 [INFO ] [.e.s.m.c.i.ModelRepositoryImpl:113  ] - Loading model 'demo.rules'
2018-08-31 09:40:17.161 [DEBUG] [.core.common.ThreadPoolManager:130  ] - Created scheduled thread pool 'RuleEngine' of size 5
2018-08-31 09:40:17.163 [INFO ] [rg.quartz.core.QuartzScheduler:2311 ] - JobFactory set to: org.eclipse.smarthome.model.rule.runtime.internal.engine.GuiceAwareJobFactory@6d8a04bf
2018-08-31 09:40:17.164 [INFO ] [rg.quartz.core.QuartzScheduler:613  ] - Scheduler openHAB-job-scheduler_$_NON_CLUSTERED paused.
2018-08-31 09:40:17.164 [DEBUG] [.m.r.r.i.engine.RuleEngineImpl:107  ] - Started rule engine
2018-08-31 09:40:17.164 [WARN ] [.e.s.m.c.i.ModelRepositoryImpl:75   ] - Configuration model 'demo.rules' is either empty or cannot be parsed correctly!
2018-08-31 09:40:17.180 [DEBUG] [.u.i.proxy.ProxyServletService:168  ] - Starting up 'Proxy (async)' servlet  at /proxy
2018-08-31 09:40:17.207 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.openhab.binding.zigbee.cc2531.internal.CC2531HandlerFactory@5e7ad9a4
2018-08-31 09:40:17.208 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.openhab.binding.zigbee.internal.ZigBeeHandlerFactory@18794b5f
2018-08-31 09:40:17.209 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.openhab.binding.zigbee.ember.internal.EmberHandlerFactory@48a58659
2018-08-31 09:40:17.209 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.openhab.binding.zigbee.xbee.internal.XBeeHandlerFactory@1dcd12bf
2018-08-31 09:40:17.209 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.eclipse.smarthome.binding.hue.internal.HueThingHandlerFactory@3911ffcb
2018-08-31 09:40:17.209 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.eclipse.smarthome.binding.yahooweather.internal.YahooWeatherHandlerFactory@312eb6b4
2018-08-31 09:40:17.210 [DEBUG] [e.s.m.t.i.GenericThingProvider:860  ] - ThingHandlerFactory added org.eclipse.smarthome.binding.ntp.internal.NtpHandlerFactory@adc355f
[... cut ...]
2018-08-31 10:29:19.983 [DEBUG] [b.z.handler.ZigBeeThingHandler:339  ] - 841826000001077E: Polling zigbee:device:01381368:841826000001077e:841826000001077E_3_colortemperature
2018-08-31 10:29:27.985 [DEBUG] [b.z.handler.ZigBeeThingHandler:339  ] - 841826000001077E: Polling zigbee:device:01381368:841826000001077e:841826000001077E_3_switch
2018-08-31 10:29:35.986 [DEBUG] [b.z.handler.ZigBeeThingHandler:339  ] - 841826000001077E: Polling zigbee:device:01381368:841826000001077e:841826000001077E_3_dimmer
2018-08-31 10:29:51.988 [DEBUG] [b.z.handler.ZigBeeThingHandler:339  ] - 841826000001077E: Polling zigbee:device:01381368:841826000001077e:841826000001077E_3_color
```

cdjackson, 2018-09-05:
I've found that the two missing channels had a minor, but significant bug that may be the cause of this. When the converter starts, it searches for attributes, and in this check it should not need to re-read the values - it should instead be able to use the last value in order to detect if the attribute is supported. Instead, these two converters requested the value again, and if the device didn't respond, they would have been dropped.

This is simply not so tolerant to timeouts and will be updated to be inline with the other converters.

cdjackson, 2018-09-06:
The PR was already provided - see #251.

PR #251 (2018-10-04): Add trace level logging into channel detection
This adds a standard set of trace level logging into the channel detection method in all converters. This should allow the channel detection to be debugged in the event that a device doesn't provide a channel.
Also fixes #245
closed 2018-10-04 by hsudbrock