openhab/org.openhab.binding.zigbee #259: Support Default Tags

firasalmanna, 2018-09-12:
Hello,

I was trying to modify the binding to add default tags to items so that I can use it with Alexa. I use PaperUI to add things in simple mode, so I can't manually add tags to items.

I added the `<tags>` tag to some channels in `channels.xml` and recompile the jar. Then I added some devices but the tags didn't appear. I dug a little into the code and found that the binding manually creates the channels in the corresponding converter, and it doesn't include `withDefaultTags`. 

I can't say for sure if adding `withDefaultTags` to each converter is the best way to do it or there is a better way, I thought asking before is better.

BR.

firasalmanna, 2018-09-13:
If I understand the implementation correctly, the channels are being built in the corresponding converter class using the `getChannel(ThingUID thingUID, ZigBeeEndpoint endpoint)` method. 

e.g. `ZigBeeConverterSwitchOnoff` for `switch_onoff` channel-type. 

But inside the method, I see no tags are being passed, and the `create` method from the `ChannelBuilder` will pass empty HashSet for the tags, so `withDefaultTags` need to be used to pass the tags. I am still not sure how to get them from `channels.xml`, or maybe I need to hardcode them in `ZigBeeBindingConstants`?

cdjackson, 2018-09-13:
Channels are built in the converter, but channel-types are defined in the XML. I would expect that we should be able to define this information in the XML, and this should be the default. When building channels, it is possible to override the information that is defined in the channel-type for each channel, and if this is not done (as is the case for the tags) then it should use the value defined in the channel-type.

This is the same as we do for other parts of the channel definition - eg we also don't define the state label - this comes from the XML from the channel type.

Maybe I'm wrong, but this is how it works for other parts of the system - we define the information for the channel-type, and this should be used in the channel unless it is overridden. Maybe it's different when done in this way, but if that is the case, then why do we specify channel-type at all?

firasalmanna, 2018-09-13:
IMHO I guess other add-ons have the thing linked to its channels in the thing-type XML file, but here the thing has no linked channels and what the binding is doing is that it links them dynamically based on Zigbee clusters. 
So the way this binding creates things and channels is different than other bindings. Maybe the method they use actually reads the channels details and adds them on the fly?

Out of curiosity I manually added the tags in `ZigBeeConverterSwitchOnoff` and now they are appearing in the items automatically as expected.

I still think there should be a better way to do it as they are already there in the XML file and it doesn't make sense to rewrite them. Maybe @kaikreuzer can help me here?

cdjackson, 2018-09-13:
Yes - maybe if you build channels, then the XML definition of the channel type is not used. However, I think that if the channel types are not listed in the XML file, then the the channel can not be instantiated. It seems a little inconsistent though - as I said, some information does seem to be used from the XML (such as the states), and if some is used, then I would expect everything, but maybe Kai can comment further.

closed 2022-06-25 by cdjackson