openhab/org.openhab.binding.zigbee #261: Support restoring state of bulbs on reconnect

choller, 2018-09-17:
Currently, when a bulb gets disconnected and then reconnects, it will go into its default state. For most bulbs, this is a 100% white light. This behavior is undesirable for two major reasons:

1. Some people use hardware switches in combination with soft-switching (for saving power, accidentally or simply because no soft switches are mounted/available).

2. On power outage, all lamps that are soft-switched off will turn on.

Instead, we should have some option to restore the lamp state to what it was before the lamp lost power (e.g. powered off, dimmed, different color, etc). Some gateways offer this already, e.g. Homee and Phoscon.

Of course it would be best if the lamps themselves would support this, but due to the vast majority of lamps out there and Hue only starting now to consider doing this, I don't think that will be a viable option.

cdjackson, 2018-09-17:
There are two ways I can see to implement this -:

* The binding can do this internally. It can detect when the device comes online through a special message that is sent when a device powers on. The color and level converters already maintains the state of the device for other reasons, so it can restore this quite easily I think.
* We could add a new trigger channel that allows a rule to be written to do this outside of the binding.

Having thought about this some more, my feeling is that this should be done in the binding. My rational is that if some systems perform this (eg some bulbs, or some hubs) then we end up with an inconsistent implementation between different implementations (eg some Hue bulbs might require something to do this while others don't.

Eg @kaikreuzer stated that "LIFX does (provide this) and therefore there’s a configuration parameter on the channel through which the user can define the desired behavior upon an “ON” command.". So why not make this consistent (from a user perspective) and provide the same configuration parameter to restore the state, but in this case it's handled by the binding?

I welcome other views and opinions...

kaikreuzer, 2018-09-18:
It seems my statement was in a different context (maybe I missed the full context when making it). What I said is that the LIFX binding offers a channel configuration on how an "ON" command is to be interpreted - whether it sends 100% brightness, some other value or the last known one before it was switched off.

Here, the suggestion is to even capture commands that come from anywhere else than the binding itself (wall switches, remotes, 3rd party apps) and try to do a corrective action.
I am not so convinced that this is in the responsibility of the binding to do. After all, there bulb WILL turn on to 100%, if this is what it's firmware says - the binding would then only send another command to dim it to some other value afterwards; probably not really a nice user experience.

cdjackson, 2018-09-18:
> What I said is that the LIFX binding offers a channel configuration on how an "ON" command is to be interpreted - whether it sends 100% brightness, some other value or the last known one before it was switched off.

Ok, I misunderstood. I think this is a standard ZigBee feature anyway.

However, some bulbs do have the ability to restore their state on power on to the condition they were in when power was removed. I'm not sure about ZigBee, but certainly many ZWave bulbs do provide this feature.

> Here, the suggestion is to even capture commands that come from anywhere else than the binding itself (wall switches, remotes, 3rd party apps) and try to do a corrective action.

No (or at least that was not my intention). My proposal here is just to restore the last value that the binding set if the device resets. 

> the binding would then only send another command to dim it to some other value afterwards; probably not really a nice user experience.

I think it's accepted that this isn't be best user experience, but that it's a lot better than the alternative of the lights coming on at 100%, and then the user having to manually adjust them back to where they were.

What do you think about the alternative idea with the trigger? I'm not such a big fan of this, but it's the only way I can otherwise think of to provide this function if it's not allowed in the binding.

kaikreuzer, 2018-09-20:
> My proposal here is just to restore the last value that the binding set if the device resets.

What exactly do you mean by "device resets"? Reconnects to the network? So it would be a kind of initialisation, I see. But that would be indeed a very different use case than the one I mentioned for LIFX.

So the use case here is that someone uses the physical power switch (accidentially?) and after switching off and on again, the bulb should be the same as before (even "software off"?), right? This indeed would not sound like a binding specific behavior.

> What do you think about the alternative idea with the trigger? I'm not such a big fan of this, but it's the only way I can otherwise think of to provide this function if it's not allowed in the binding.

Doesn't this exist already? We have a Thing ONLINE/OFFLINE trigger for rules. So one could write a rule, which stores the last current state when a Thing goes OFFLINE and sends it as a command once it goes ONLINE again.

cdjackson, 2018-09-20:
> What exactly do you mean by "device resets"? Reconnects to the network? 

Yes - when a device has been powered off (even for a second) it will reconnect, and it then sends out a announcement that it is available again.

> So it would be a kind of initialisation, I see.

I wouldn't probably call it initialisation - at least to me that is something that is mostly done when the device is first connected to the network. There is nothing really required to be done here to use the device - only that in the case of these bulbs, they have lost their "current" state.

> that would be indeed a very different use case than the one I mentioned for LIFX.

Agreed - it's not the same.

>  the bulb should be the same as before (even "software off"?), right?

Yes - this is the use case.

> This indeed would not sound like a binding specific behavior.

My initial thought was the same, but if we want to "normalise" all color bulbs (from the user perspective), and some provide this function in hardware, then _maybe_ the binding should also support such a function so that all bulbs can be considered equal?

> We have a Thing ONLINE/OFFLINE trigger for rules.

Except that the bulb will not be OFFLINE. In this use case, we will not detect an offline condition - it's only off for a short time, so there is no way to detect this. We could send an OFFLINE/ONLINE when the bulb sends it's announcement, but I don't really like to say it's offline in this case.

cdjackson, 2018-12-23:
This is now implemented in #345

PR #345 (2018-12-24): Refresh devices when they announce themselves
This adds a ```ZigBeeAnnounceListener``` to detect if the device has potentially been reset, and will refresh the thing state through restarting the polling. This means that the binding will update a bulb state if it was powered off/on again, as requested in #261.

Closes #261
closed 2018-12-24 by cdjackson