openhab/org.openhab.binding.zigbee #267: Color temperature converter mixes up max and min

jkl1337, 2018-09-22:
When controlling Osram Lightify bulbs, 100% is high temp and the lower bound (which is unfortunately not 0) is low temp. With Hue bulbs the scale is reversed (100% is 2000K), and the Hue bulb scale direction is probably the one that wrong

The bug is that the converter code queries ColorTemperatureMin cluster attribute for the lower kelvin bound, but this is actually the minimum in mireds (reports as 153 (~6500K) for Hue bulbs). Since the Hue bulbs support the queries the scale gets inverted.

The Osram bulbs do not support querying the Min and Max temp so the hardcoded default of 2000 and 6500 is used.  Unfortunate side effect is that nothing below 16% is supported as a command because the bulbs actually only support down to 2700K.

On a side note, I sometimes wish that the color temperature was exposed as a number rather than dimmer either in mireds or K.

jkl1337, 2018-09-22:
It just occurred to me that the Alexa binding seems to be setting Hue bulbs correctly - so is the official convention 100% is low temp? If that is the case then the problem is that the hardcoded default values are flipped.

hsudbrock, 2018-10-31:
I agree, that the computation of the min/max values in Kelvin from the min/max values in Mired needs to be turned around (as low Mired values correspond to high Kelvin values and vice versa).

I have created a PR that, I believe, should fix this, and also tries to make a little more explicit where conversion between Mired, Kelvin, and ESH's percent type is performed.

Would you mind to take a quick look, @jkl1337 ?