openhab/org.openhab.binding.zigbee #268: Innr sp120 switch not working with telegesis coordinator

arins, 2018-09-23:
### Outline

I have an issue with a zigbee device called innr sp120 switch.
The thing has been detected and added but switching does not work. Look in the logs

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
|Coordinator |    Telegesis                                                                            |
| Openhab version  |  2.4.0.M3 
| Hardware         |    x86 pc  intel core 2 duo 2.4 ghz                                                                          |
| Memory           |     8 Gb                                                                           |
| Java version     |          java -version java version "1.8.0_181" Java(TM) SE Runtime Environment (build 1.8.0_181-b13) Java HotSpot(TM) 64-Bit Server VM (build 25.181-b13, mixed mode)                                                                      |
| Devices          |  ikea trådfri bulb (works) zigbee:device:04000A6E:90fd9ffffe0853cb innr sp120 switch (does not work) zigbee:device:04000A6E:00158d000217bcf9 telegesis coordinator zigbee:coordinator_telegesis:04000A6E  xiaomi door sensor (works) zigbee:device:04000A6E:00158d0001ab151f | 

### Logs
```
10:00:57.706 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
10:00:57.707 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
10:00:57.708 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis Data: 41 54 2B 4E 3F 0D 0A
10:00:57.838 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 2B 4E 3D 43 4F 4F 2C 32 32 2C 2D 31 37 2C 43 39 36 34 2C 32 31 46 35 43 41 39 38 41 37 39 32 35 45 33 34
10:00:57.839 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4F 4B
10:00:57.840 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=22, power=-17, panId=51556, epanId=21F5CA98A7925E34, status=SUCCESS]
10:00:57.861 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00158D000217BCF9: Command for channel zigbee:device:04000A6E:00158d000217bcf9:00158D000217BCF9_1_dimmer --> 89
10:00:57.862 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: MoveToLevelWithOnOffCommand [Level Control: 0/0 -> 47499/1, cluster=0008, TID=E7, level=226, transitionTime=10]
10:00:57.863 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=231, commandId=4]
10:00:57.863 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=47499/1, profile=0104, cluster=8, addressMode=DEVICE, radius=31, apsCounter=231, payload=01 E7 04 E2 0A 00]
10:00:57.864 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis send: TelegesisSendUnicastCommand [address=47499, sourceEp=0, destEp=1, profileId=260, clusterId=8, messageData=01 E7 04 E2 0A 00, messageId=null]
10:00:57.864 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
10:00:57.864 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=47499, sourceEp=0, destEp=1, profileId=260, clusterId=8, messageData=01 E7 04 E2 0A 00, messageId=null]
10:00:57.865 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis Data: 41 54 2B 53 45 4E 44 55 43 41 53 54 42 3A 30 36 2C 42 39 38 42 2C 30 30 2C 30 31 2C 30 31 30 34 2C 30 30 30 38 0D 01 E7 04 E2 0A 00 0D 0A
10:00:58.009 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 53 45 51 3A 33 38
10:00:58.011 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4F 4B
10:00:58.011 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisSendUnicastCommand [address=47499, sourceEp=0, destEp=1, profileId=260, clusterId=8, messageData=01 E7 04 E2 0A 00, messageId=56, status=SUCCESS]
10:00:58.012 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 41 43 4B 3A 33 38
10:00:58.013 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisAckMessageEvent [messageId=56]
10:00:58.014 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisAckMessageEvent [messageId=56]
10:00:59.014 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
10:00:59.015 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
10:00:59.016 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis Data: 41 54 2B 4E 3F 0D 0A
10:00:59.147 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 2B 4E 3D 43 4F 4F 2C 32 32 2C 2D 31 37 2C 43 39 36 34 2C 32 31 46 35 43 41 39 38 41 37 39 32 35 45 33 34
10:00:59.148 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4F 4B
10:00:59.149 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=22, power=-17, panId=51556, epanId=21F5CA98A7925E34, status=SUCCESS]
10:01:00.149 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
10:01:00.151 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]
10:01:00.152 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis Data: 41 54 2B 4E 3F 0D 0A
10:01:00.283 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 2B 4E 3D 43 4F 4F 2C 32 32 2C 2D 31 37 2C 43 39 36 34 2C 32 31 46 35 43 41 39 38 41 37 39 32 35 45 33 34
10:01:00.284 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4F 4B
10:01:00.284 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=22, power=-17, panId=51556, epanId=21F5CA98A7925E34, status=SUCCESS]
10:01:00.994 [DEBUG] [.zigbee.transaction.ZigBeeTransaction] - Transaction timeout: MoveToLevelWithOnOffCommand [Level Control: 0/0 -> 47499/1, cluster=0008, TID=E6, level=135, transitionTime=10]
10:01:01.168 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00158D000217BCF9: Command for channel zigbee:device:04000A6E:00158d000217bcf9:00158D000217BCF9_1_dimmer --> 99
10:01:01.170 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: MoveToLevelWithOnOffCommand [Level Control: 0/0 -> 47499/1, cluster=0008, TID=E8, level=251, transitionTime=10]
10:01:01.172 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=232, commandId=4]
10:01:01.173 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=47499/1, profile=0104, cluster=8, addressMode=DEVICE, radius=31, apsCounter=232, payload=01 E8 04 FB 0A 00]
10:01:01.174 [DEBUG] [ongle.telegesis.ZigBeeDongleTelegesis] - Telegesis send: TelegesisSendUnicastCommand [address=47499, sourceEp=0, destEp=1, profileId=260, clusterId=8, messageData=01 E8 04 FB 0A 00, messageId=null]
10:01:01.174 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
10:01:01.177 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=47499, sourceEp=0, destEp=1, profileId=260, clusterId=8, messageData=01 E8 04 FB 0A 00, messageId=null]
10:01:01.179 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis Data: 41 54 2B 53 45 4E 44 55 43 41 53 54 42 3A 30 36 2C 42 39 38 42 2C 30 30 2C 30 31 2C 30 31 30 34 2C 30 30 30 38 0D 01 E8 04 FB 0A 00 0D 0A
10:01:01.285 [DEBUG] [egesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
10:01:01.334 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 53 45 51 3A 33 39
10:01:01.335 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4F 4B
[... cut ...]
10:01:11.347 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 2B 4E 3D 43 4F 4F 2C 32 32 2C 2D 31 37 2C 43 39 36 34 2C 32 31 46 35 43 41 39 38 41 37 39 32 35 45 33 34
10:01:11.349 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis Data: 4F 4B
10:01:11.349 [DEBUG] [egesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=22, power=-17, panId=51556, epanId=21F5CA98A7925E34, status=SUCCESS]
Paste logs here (surounded by code fences to ensure formatting) or upload larger files to your favourite sharing site and link from the issue.
```

cdjackson, 2018-09-23:
It would help if you can describe what you are doing and what I’m looking at in the logs?  I can see you sending some commands from the UK, and these commands get sent to the device. What are you expecting to happen?

I know some Innr devices don’t work with the Telegesis dongle (not sure why, but I know some bulbs at least will not work).

Diagnosing further may be difficult without the device.

arins, 2018-09-25:
Sorry for not describing the problem good enough. The switch is not reacting to the commands from openhab. I didnt know telegesis didn't work with innr. Witch dongle should I use for innr devices? Or can I buy you a innr sp120 device so you can try to find out what is wrong?

cdjackson, 2018-09-29:
As this is a power outlet, I would have expected it would work ok with the Telegesis dongle - where I've seen problems is with the Innr bulbs.

Can you post the xml file from the {userdata}/zigbee folder? This shows the services supported by the device and may be useful. Otherwise getting hold of the device should help.

cdjackson, 2018-12-17:
Unfortunately the device``` 00158d000217bcf9```  does not seem to be included in this file.

arins, 2018-12-17:
whats this?  

```xml
<bindingTable>
      <IeeeAddress>
        <srcAddr>00158D000217BCF9</srcAddr>
        <srcEndpoint>1</srcEndpoint>
        <clusterId>6</clusterId>
        <dstAddrMode>3</dstAddrMode>
        <dstGroupAddr>0</dstGroupAddr>
        <dstAddr>000D6F000D03D356</dstAddr>
        <dstNodeEndpoint>1</dstNodeEndpoint>
      </IeeeAddress>
      <IeeeAddress>
        <srcAddr>00158D000217BCF9</srcAddr>
        <srcEndpoint>1</srcEndpoint>
        <clusterId>8</clusterId>
        <dstAddrMode>3</dstAddrMode>
        <dstGroupAddr>0</dstGroupAddr>
        <dstAddr>000D6F000D03D356</dstAddr>
        <dstNodeEndpoint>1</dstNodeEndpoint>
      </IeeeAddress>
    </bindingTable>

cdjackson, 2018-12-17:
That's the binding table.

However, it is in the file - my search seemed to be case sensitive. The device supports the standard server clusters (on/off and level) so it should work ok. Without seeing a log during the initialisation though, it's kind of hard to know why it would not detect these channels.

cdjackson, 2019-08-10:
Closing this due to inactivity. If the problem persists, please reopen and provide further information as above.

closed 2019-08-10 by cdjackson