openhab/org.openhab.binding.zigbee #282: Illegal State transition on unplugging a zigbee stick

triller-telekom, 2018-10-23:
I just unplugged my zigbee stick from a running openHAB instance and the bridge thing stayed ONLINE. A view in the logs showed:

```
Illegal status transition from REMOVING to OFFLINE, only REMOVED would have been allowed.
        at org.eclipse.smarthome.core.thing.internal.ThingManager$1.statusUpdated(ThingManager.java:177)
        at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.updateStatus(BaseThingHandler.java:399)
        at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.updateStatus(BaseThingHandler.java:423)
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.nodeRemoved(ZigBeeThingHandler.java:556)
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager$5.run(ZigBeeNetworkManager.java:1182)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```
So I guess in https://github.com/openhab/org.openhab.binding.zigbee/blob/master/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeThingHandler.java#L557

the state should be set to `ThingStatus.REMOVED`. Is that correct?

I can provide a PR to fix this once someone confirms it.

cdjackson, 2018-10-23:
Probably you are right in that the state should be REMOVED if the thing is deleted, but I'm a bit confused by what triggered this. If you remove the stick, then I'm not sure why the things linked to the stick are removed? Or is this a new feature? Or maybe I just misunderstand what the issue is?

In general, I would have expected the stick to stay ONLINE if you remove it as other than for the Telegesis stick, there is no detection of stick removal. However, I don't know what stick you removed.

triller-telekom, 2018-10-23:
It was an ember USB stick which I pulled out of my system.

But I am also a bit confused about what happened. Because I would have expected that either:

1. the bridge goes OFFLINE with reason "Device unplugged"
2. bridge goes into REMOVED state

I am also expecting all connected things of that bridge to go OFFLINE in both cases.

cdjackson, 2018-10-23:
Hmmm - I didn't think I had added this detection to the ZigBee library - I'm pretty sure it was only added for Telegesis, and even then, I don't know where the "Device unplugged" comes from. Maybe @hsudbrock can comment as I wonder if it is somehow related to the USB detection?

triller-telekom, 2018-10-23:
Actually I was wrong with what I said before: The stacktrace was old and I just caught it again.

It happend while removing a OSRAM rgb bulb thing through paperUI!

So I guess in that case it should go from REMOVING to REMOVED, right?

hsudbrock, 2018-10-23:
I take from @triller-telekom's comment above that this issue is related to a thing, and not to the bridge, and that hence it should be unrelated to USB detection. 

I am wondering anyway regarding the bridge: The USB detection will remove the `DiscoveryResult` when the USB dongle is unplugged - would removing the discovery result remove the corresponding thing, @triller-telekom?

hsudbrock, 2018-10-31:
For completeness, here's the info on the side track opened above regarding bridge removal and USB detection: As expected, the bridge is **not** removed when the USB dongle is plugged out (only a not yet approved discovery result would be removed, but not the thing generated from the discovery result).

hsudbrock, 2018-10-31:
I will have a look on how to fix this hopefully small issue.

PR #286 (2018-10-31): Do not set thing status to OFFLINE on node removal during thing removal
Setting the status to `OFFLINE` for a thing in status `REMOVING` leads to an invalid thing status transition, as one can only go to `REMOVED` from `REMOVING` state.

Fixes #282
closed 2018-10-31 by hsudbrock